jurobystricky/Netgear-A6210 #2: Edup EP-AC1601

veaa, 2015-12-18:
Could this driver be used with Edup EP-AC1601 with the same chipset (MT7612U) somehow? 

The Dmesg shows following:

[   14.591373] usb 3-2: New USB device found, idVendor=203a, idProduct=fffa
[   14.591377] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   14.591380] usb 3-2: Product: Virtual Printer (Print to PDF (Mac Desktop))
[   14.591381] usb 3-2: Manufacturer: Parallels
[   14.591382] usb 3-2: SerialNumber: TAG11d87aca0
[   14.598633] usblp 3-2:1.0: usblp0: USB Unidirectional printer dev 3 if 0 alt 0 proto 1 vid 0x203A pid 0xFFFA
[   14.598650] usbcore: registered new interface driver usblp
[   14.647250] #
[   14.727236] #
[   14.807267] #
[   14.887236] #
[   14.967208] #
[   15.047334] #
[   15.127406] #
[   15.207286] #
[   15.212339] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x9a8,pAd->Flags=0x0
[   15.212342] JB: RTUSB_VendorRequest failed!
[   15.287237] #
[   15.367254] #
[   15.447661] #
[   15.531172] #
[   15.611400] #
[   15.695210] #
[   15.775263] #
[   15.855617] #
[   15.939488] #
[   16.019231] #
[   16.024211] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x6, Idx=0x9a8,pAd->Flags=0x0
[   16.024213] JB: RTUSB_VendorRequest failed!
[   16.024214] VendrCmdMultiWrite_nBytes failed!
[   16.103235] #
[   16.183294] #
[   16.263354] #
[   16.343353] #
[   16.427458] #
[   16.507447] #
[   16.587317] #
[   16.667452] #
[   16.747372] #
[   16.827800] #
[   16.832786] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x42, Idx=0x230,pAd->Flags=0x0
[   16.832790] JB: RTUSB_VendorRequest failed!
[   16.832791] set fce dma descriptor fail
[   16.832797] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[   16.832946] receive cmd msg fail(-2)
[   16.833019] tx_kickout_fail_count = 0
[   16.833021] tx_timeout_fail_count = 0
[   16.833021] rx_receive_fail_count = 0
[   16.833022] alloc_cmd_msg = 1
[   16.833022] free_cmd_msg = 1
[   16.833023] !!! rt28xx init fail !!!
[   16.919300] #
[   16.999641] #
[   17.080368] #
[   17.159641] #
[   17.239490] #
[   17.319772] #
[   17.400227] #
[   17.479286] #
[   17.559773] #
[   17.639936] #
[   17.644929] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x1000,pAd->Flags=0x0
[   17.644931] JB: RTUSB_VendorRequest failed!
[   17.719426] #
[   17.799261] #
[   17.879321] #
[   17.959536] #
[   18.039552] #
[   23.127295] xhci_hcd 0000:00:1d.6: xHCI host not responding to stop endpoint command.
[   23.127303] xhci_hcd 0000:00:1d.6: Assuming host is dying, halting host.
[   23.127592] xhci_hcd 0000:00:1d.6: HC died; cleaning up
[   23.127643] #
[   23.132637] #
[   23.132640] RTUSB_VendorRequest failed(-19),TxFlags=0x0, ReqType=IN, Req=0x47, Idx=0x148,pAd->Flags=0x100
[   23.132642] JB: RTUSB_VendorRequest failed!
[   23.132643] write reg fail
[   23.132644] write reg fail
[   23.132654] write reg fail
[   23.132665] write reg fail
[   23.132665] write reg fail
[   23.132666] write reg fail
[   23.132667] write reg fail
[   23.132667] write reg fail
[   23.132668] write reg fail
[   23.132678] write reg fail
[   23.132689] write reg fail
[   23.132690] write reg fail
[   23.132691] write reg fail
[   23.132692] write reg fail
[   23.132692] write reg fail
[   23.132694] write reg fail
[   23.132819] write reg fail
[   23.132869] write reg fail
[   23.132870] VendrCmdMultiWrite_nBytes failed!
[   23.132871] write reg fail
[   23.132882] write reg fail
[   23.132892] write reg fail
[   23.132893] write reg fail
[   23.132894] write reg fail
[   23.132895] write reg fail
[   23.132896] write reg fail
[   23.132897] write reg fail
[   23.133022] write reg fail
[   23.133073] write reg fail
[   23.133073] VendrCmdMultiWrite_nBytes failed!
[   23.133076] !!! rt28xx init fail !!!
[   23.133270] usb 2-1: USB disconnect, device number 2
[   23.133371] rtusb_disconnect: unregister usbnet usb-0000:00:1d.6-1
[   23.133376] RtmpOSNetDevDetach(): RtmpOSNetDeviceDetach(), dev->name=wlan0!
[   23.175208] 80211> unregister/free wireless device
[   23.195200] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0xffffffff
[   23.195203] VendrCmdMultiWrite_nBytes failed!
[   23.195232] ---> RTMPFreeTxRxRingMemory
[   23.195248] <--- RTMPFreeTxRxRingMemory
[   23.195288]  RTUSB disconnect successfully

jurobystricky, 2015-12-28:
Did you enter your USB VendorID/DeviceID into rtusb_dev_id.c?

veaa, 2016-01-03:
Hi

I know you have more important job to do but I'm still getting errors. I am running Kali Linux 2 now. Could you help me? Output of DMESG follows.

Regards,
Vesa

[   11.844855] rtusb init rt2870 --->
[   11.845039] 
=== pAd = ffffc90003d71000, size = 1292976 ===

[   11.845041] driver version: 3.0.0.1 (Jan  3 2016 15:40:55) .
[   11.845924] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20b
[   11.865523] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[   11.865525] RtmpEepromGetDefault::e2p_dafault=1
[   11.865985] NVM is EFUSE mode
[   11.865988] Endpoint(8) is for In-band Command
[   11.865989] Endpoint(4) is for WMM0 AC0
[   11.865989] Endpoint(5) is for WMM0 AC1
[   11.865990] Endpoint(6) is for WMM0 AC2
[   11.865990] Endpoint(7) is for WMM0 AC3
[   11.865991] Endpoint(9) is for WMM1 AC0
[   11.865991] Endpoint(84) is for Data-In
[   11.865992] Endpoint(85) is for Command Rsp
[   11.866226] 80211> RFICType= 3, NumOfChan= 58
[   11.866228] 80211> Number of rate = 12
[   11.866229] 80211> CurTxPower = 20 dBm
[   11.866230] ====> Radar Channel 52
[   11.866231] ====> Radar Channel 54
[   11.866231] ====> Radar Channel 56
[   11.866232] ====> Radar Channel 60
[   11.866232] ====> Radar Channel 62
[   11.866233] ====> Radar Channel 64
[   11.866233] ====> Radar Channel 100
[   11.866234] ====> Radar Channel 104
[   11.866235] 80211> TxStream = 0
[   11.866482] usbcore: registered new interface driver rt2870
[   11.878370] -->RTUSBVenderReset
[   11.878604] <--RTUSBVenderReset
[   11.888227] .....
[   11.938187] random: nonblocking pool is initialized
[   12.006772] #
[   12.086575] #
[   12.170547] #
[   12.251547] #
[   12.321012] usblp0: removed
[   12.325337] usblp 2-1:1.0: usblp0: USB Unidirectional printer dev 2 if 0 alt 0 proto 1 vid 0x203A pid 0xFFFA
[   12.330569] #
[   12.410548] #
[   12.498653] #
[   12.578590] #
[   12.658659] #
[   12.742551] #
[   12.747529] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x9a8,pAd->Flags=0x0
[   12.747531] JB: RTUSB_VendorRequest failed!
[   12.822564] #
[   12.902636] #
[   12.982640] #
[   13.062545] #
[   13.142580] #
[   13.222584] #
[   13.302662] #
[   13.382623] #
[   13.462556] #
[   13.542596] #
[   13.547580] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x6, Idx=0x9a8,pAd->Flags=0x0
[   13.547582] JB: RTUSB_VendorRequest failed!
[   13.547583] VendrCmdMultiWrite_nBytes failed!
[   13.638592] #
[   13.706556] floppy0: no floppy controllers found
[... 182 lines cut ...]
[   25.082801] #
[   25.162678] #
[   25.243483] #
[   25.322753] #
[   25.403716] #
[   25.482788] #
[   25.563377] #
[   25.643544] #
[   25.723590] #
[   25.728593] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x46, Idx=0x1204,pAd->Flags=0x0
[   25.728595] JB: RTUSB_VendorRequest failed!
[   25.728597] write reg fail
[   25.803103] #
[   25.882740] #
[   25.963486] #
[   26.043677] #
[   26.122746] #
[   26.203870] #
[   26.283758] #
[   26.363466] #
[   26.442961] #
[   26.523339] #
[   26.528328] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x47, Idx=0x80,pAd->Flags=0x0
[   26.528331] JB: RTUSB_VendorRequest failed!
[   26.602726] #

jurobystricky, 2016-01-04:
Seems there is no communication with USB (timeouts).
Please provide more information about your USB device (i.e. output of 'lsusb') and OS (Distro, kernel version).
Make sure your device is listed in the file rtusb_dev_id.c.
Finally, edit rt_linux.c so the driver prints out more detailed debug messages:

ULONG RTDebugLevel = RT_DEBUG_TRACE;

Rebuild/reinstall the driver and reboot and post the new dmesg output.
This will hopefully shed some light on the problem.

veaa, 2016-01-11:
I,m sorry about the delay. Thank you for your help.

This the original rtusb_dev_id.c and I already see my device in there:
{USB_DEVICE(0x0E8D, 0x7612), .driver_info = RLT_MAC_BASE},

 Here follows the output form uname, lsusb and dimes. I understood the debugging was set on by you.

root@kali:~# uname -r
4.0.0-kali1-amd64

root@kali:~# lsusb
Bus 001 Device 003: ID 203a:fff9  
Bus 001 Device 002: ID 203a:fffa  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0e8d:7612 MediaTek Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

[   11.127266] snd_intel8x0 0000:00:1f.4: clocking to 46713
[   11.170085] cfg80211: Calling CRDA to update world regulatory domain
[   11.186677] cfg80211: World regulatory domain updated:
[   11.186680] cfg80211:  DFS Master region: unset
[   11.186681] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   11.186683] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   11.186684] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   11.186684] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   11.186685] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   11.186686] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   11.186687] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   11.186688] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   11.186689] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   11.208830] Parallels Linux shared folders filesystem driver 1.2.1 loaded
[   11.319344] Bluetooth: Core ver 2.20
[   11.319355] NET: Registered protocol family 31
[   11.319356] Bluetooth: HCI device and connection manager initialized
[   11.319358] Bluetooth: HCI socket layer initialized
[   11.319363] Bluetooth: L2CAP socket layer initialized
[   11.319774] Bluetooth: SCO socket layer initialized
[   11.331630] Netfilter messages via NETLINK v0.30.
[   11.570170] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input5
[   11.654857] random: nonblocking pool is initialized
[   11.780951] rtusb init rt2870 --->
[   11.781098] 
=== pAd = ffffc90003d7b000, size = 1292976 ===

[   11.781100] driver version: 3.0.0.1 (Jan 11 2016 08:57:15) .
[   11.781786] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20b
[   11.794944] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[   11.794947] RtmpEepromGetDefault::e2p_dafault=1
[   11.795287] NVM is EFUSE mode
[   11.795288] Endpoint(8) is for In-band Command
[   11.795289] Endpoint(4) is for WMM0 AC0
[   11.795290] Endpoint(5) is for WMM0 AC1
[   11.795290] Endpoint(6) is for WMM0 AC2
[   11.795291] Endpoint(7) is for WMM0 AC3
[   11.795291] Endpoint(9) is for WMM1 AC0
[   11.795292] Endpoint(84) is for Data-In
[   11.795292] Endpoint(85) is for Command Rsp
[   11.795511] 80211> RFICType= 3, NumOfChan= 58
[   11.795513] 80211> Number of rate = 12
[   11.795514] 80211> CurTxPower = 20 dBm
[   11.795515] ====> Radar Channel 52
[   11.795516] ====> Radar Channel 54
[   11.795516] ====> Radar Channel 56
[   11.795517] ====> Radar Channel 60
[   11.795517] ====> Radar Channel 62
[   11.795517] ====> Radar Channel 64
[   11.795518] ====> Radar Channel 100
[   11.795518] ====> Radar Channel 104
[... 1201 lines cut ...]
[   27.255894] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x46, Idx=0x80,pAd->Flags=0x0
[   27.255896] JB: RTUSB_VendorRequest failed!
[   27.255897] write reg fail
[   27.331181] #
[   27.411140] #
[   27.490119] #
[   27.571276] #
[   27.650432] #
[   27.731099] #
[   27.810967] #
[   27.890133] #
[   27.971015] #
[   28.051271] #
[   28.056245] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x47, Idx=0x64,pAd->Flags=0x0
[   28.056247] JB: RTUSB_VendorRequest failed!
[   28.131317] #
[   28.210012] #
[   28.290582] #
[   28.370645] #
[   28.451123] #
[   28.530773] #
[   28.611236] #
[   28.690389] #
[   28.770481] #
root@kali:~#

jurobystricky, 2016-01-11:
There seems to be a bug in RTUSB_VendorRequest (file rtusb_io.c). The code uses somewhat obfuscated call RTUSB_CONTROL_MSG which has a weirdly defined timeout: CONTROL_TIMEOUT_JIFFIES.
But deep inside the code that is actually called (usb_control_msg) requires timeout in ms, not in jiffies.
So please try replacing  CONTROL_TIMEOUT_JIFFIES with some sufficiently large number, i.e. 500 or more. 
The error you see (-110) is due to USB timeout, so this should help. 
I will fix this in the git code regardless as well, as it is a bug. (I will  fix the rather stupid debug messages printing # for each unsuccessful retry as well)

veaa, 2016-01-12:
Hi,

I replaced CONTROL_TIMEOUT_JIFFIES with 500 in file rtusb_io.c and complied and installed. Seems like the problem persists.

If I connect the stick to a running system the linux freezes in few seconds and I can't get any output. When I boot up the system with stick connected I got following:

   11.423356] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   11.579987] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input5
[   11.757828] random: nonblocking pool is initialized
[   11.783702] rtusb init rt2870 --->
[   11.783838] 
=== pAd = ffffc90003d79000, size = 1292976 ===

[   11.783839] driver version: 3.0.0.1 (Jan 12 2016 07:04:15) .
[   11.784459] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20b
[   11.796884] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[   11.796886] RtmpEepromGetDefault::e2p_dafault=1
[   11.797247] NVM is EFUSE mode
[   11.797248] Endpoint(8) is for In-band Command
[   11.797249] Endpoint(4) is for WMM0 AC0
[   11.797250] Endpoint(5) is for WMM0 AC1
[   11.797250] Endpoint(6) is for WMM0 AC2
[   11.797251] Endpoint(7) is for WMM0 AC3
[   11.797252] Endpoint(9) is for WMM1 AC0
[   11.797252] Endpoint(84) is for Data-In
[   11.797253] Endpoint(85) is for Command Rsp
[   11.797429] 80211> RFICType= 3, NumOfChan= 58
[   11.797430] 80211> Number of rate = 12
[   11.797431] 80211> CurTxPower = 20 dBm
[   11.797432] ====> Radar Channel 52
[   11.797432] ====> Radar Channel 54
[   11.797433] ====> Radar Channel 56
[   11.797433] ====> Radar Channel 60
[   11.797434] ====> Radar Channel 62
[   11.797434] ====> Radar Channel 64
[   11.797435] ====> Radar Channel 100
[   11.797435] ====> Radar Channel 104
[   11.797436] 80211> TxStream = 0
[   11.797853] usbcore: registered new interface driver rt2870
[   11.803169] -->RTUSBVenderReset
[   11.803350] <--RTUSBVenderReset
[   11.810779] .....
[   12.067051] usblp0: removed
[   12.068208] usblp 4-1:1.0: usblp0: USB Unidirectional printer dev 2 if 0 alt 0 proto 1 vid 0x203A pid 0xFFFA
[   12.336878] #
[   12.840010] #
[   13.343694] #
[   13.691770] floppy0: no floppy controllers found
[   13.691787] work still pending
[   13.847879] #
[   14.351684] #
[   14.855798] #
[   15.359740] #
[   15.863768] #
[   16.368682] #
[   16.871956] #
[   16.876940] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x9a8,pAd->Flags=0x0
[   16.876941] JB: RTUSB_VendorRequest failed!
[   17.376543] #
[   17.880705] #
[   18.379942] #
[   18.887979] #
[   19.392753] #
[   19.895762] #
[   20.400532] #
[   20.900722] #
[   21.407967] #
[   21.912483] #
[   21.917452] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x6, Idx=0x9a8,pAd->Flags=0x0
[   21.917453] JB: RTUSB_VendorRequest failed!
[   21.917454] VendrCmdMultiWrite_nBytes failed!
[   22.420267] #
[   22.924421] #
[   23.428079] #
[   23.931749] #
[   24.435832] #
[   24.936248] #
[   25.439812] #
[   25.944552] #
[   26.448117] #
[   26.951894] #
[   26.956870] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x42, Idx=0x230,pAd->Flags=0x0
[   26.956871] JB: RTUSB_VendorRequest failed!
[   26.956872] set fce dma descriptor fail
[   26.956875] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[   26.956956] receive cmd msg fail(-2)
[   26.957003] tx_kickout_fail_count = 0
[   26.957005] tx_timeout_fail_count = 0
[   26.957006] rx_receive_fail_count = 0
[   26.957006] alloc_cmd_msg = 1
[   26.957007] free_cmd_msg = 1
[   26.957008] !!! rt28xx init fail !!!
root@kali:~#

jurobystricky, 2016-01-12:
From the timestamps I  see that the timeouts are now correct. One suspicious observation: The USB communications stops working after VendorReset ( call to RTUSBVenderReset). The VendorReset itself uses RTUSB_VendorRequest code, so it seems to work up to that time. 

I just pushed a bunch of changes that may help, mostly dealing with spinlocks and related bugfixes. (Also the changes to USB timeouts are included.) Please try and see if there is any difference.

veaa, 2016-01-13:
I'll test it asap. I'm aboard now and will return on Sunday. I"ll let you know.

veaa, 2016-01-20:
Hi,

The driver loads and it is not freezing Kali anymore. However, the stick cannot find any network. DMESG follows.

[   11.911930] rtusb init rt2870 --->
[   11.912069] 
=== pAd = ffffc90003d50000, size = 1292952 ===

[   11.912071] driver version: 3.0.0.1 (Jan 20 2016 11:14:17) .
[   11.912737] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20b
[   11.925757] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[   11.925758] RtmpEepromGetDefault::e2p_dafault=1
[   11.926084] NVM is EFUSE mode
[   11.926086] Endpoint(8) is for In-band Command
[   11.926086] Endpoint(4) is for WMM0 AC0
[   11.926087] Endpoint(5) is for WMM0 AC1
[   11.926087] Endpoint(6) is for WMM0 AC2
[   11.926088] Endpoint(7) is for WMM0 AC3
[   11.926089] Endpoint(9) is for WMM1 AC0
[   11.926089] Endpoint(84) is for Data-In
[   11.926090] Endpoint(85) is for Command Rsp
[   11.926309] 80211> RFICType= 3, NumOfChan= 58
[   11.926311] 80211> Number of rate = 12
[   11.926312] 80211> CurTxPower = 20 dBm
[   11.926313] ====> Radar Channel 52
[   11.926314] ====> Radar Channel 54
[   11.926314] ====> Radar Channel 56
[   11.926315] ====> Radar Channel 60
[   11.926315] ====> Radar Channel 62
[   11.926316] ====> Radar Channel 64
[   11.926316] ====> Radar Channel 100
[   11.926317] ====> Radar Channel 104
[   11.926317] 80211> TxStream = 0
[   11.926552] usbcore: registered new interface driver rt2870
[   13.810653] floppy0: no floppy controllers found
[   13.898834] systemd-journald[220]: Received request to flush runtime journal from PID 1
[   14.301684] Parallels Linux shared folders filesystem driver 1.2.1 loaded
[   14.339535] -->RTUSBVenderReset
[   14.343812] <--RTUSBVenderReset
[   14.810303] random: nonblocking pool is initialized
[   15.279457] usblp0: removed
[   15.281637] usblp 2-1:1.0: usblp0: USB Unidirectional printer dev 2 if 0 alt 0 proto 1 vid 0x203A pid 0xFFFA
[   16.353652] andes_usb_load_rom_patch: upload fw timeout
[   16.353657] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[   16.353698] receive cmd msg fail(-2)
[   16.353709] tx_kickout_fail_count = 0
[   16.353709] tx_timeout_fail_count = 0
[   16.353711] rx_receive_fail_count = 0
[   16.353712] alloc_cmd_msg = 1
[   16.353712] free_cmd_msg = 1
[   16.353713] !!! rt28xx init fail !!!
[   16.408669] -->RTUSBVenderReset
[   16.410544] <--RTUSBVenderReset
[   18.429530] andes_usb_load_rom_patch: upload fw timeout
[   18.429536] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[   18.429592] receive cmd msg fail(-2)
[   18.429729] tx_kickout_fail_count = 0
[   18.429731] tx_timeout_fail_count = 0
[   18.429731] rx_receive_fail_count = 0
[   18.429732] alloc_cmd_msg = 1
[   18.429732] free_cmd_msg = 1
[   18.429733] !!! rt28xx init fail !!!
[   18.431122] -->RTUSBVenderReset
[   18.431305] <--RTUSBVenderReset
[   20.437596] andes_usb_load_rom_patch: upload fw timeout
[   20.437603] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[   20.437662] receive cmd msg fail(-2)
[   20.437677] tx_kickout_fail_count = 0
[   20.437678] tx_timeout_fail_count = 0
[   20.437679] rx_receive_fail_count = 0
[... 598 lines cut ...]
[  118.083547] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
[  118.112530] andes_usb_erasefw
[  118.112651] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
[  118.113119] receive cmd msg fail(-2)
[  118.113130] tx_kickout_fail_count = 0
[  118.113131] tx_timeout_fail_count = 0
[  118.113131] rx_receive_fail_count = 0
[  118.113132] alloc_cmd_msg = 10
[  118.113134] free_cmd_msg = 10
[  118.165083] RTMP_TimerListRelease: release timer obj ffffc900017f14b0!
[  118.165085] RTMP_TimerListRelease: release timer obj ffffc900017f1528!
[  118.165086] RTMP_TimerListRelease: release timer obj ffffc900017f15a0!
[  118.165087] RTMP_TimerListRelease: release timer obj ffffc900017f1438!
[  118.165088] RTMP_TimerListRelease: release timer obj ffffc900017f12d0!
[  118.165088] RTMP_TimerListRelease: release timer obj ffffc900017f1348!
[  118.165089] RTMP_TimerListRelease: release timer obj ffffc90001782fa0!
[  118.165090] RTMP_TimerListRelease: release timer obj ffffc90001771d78!
[  118.165091] RTMP_TimerListRelease: release timer obj ffffc90001771df8!
[  118.165092] RTMP_TimerListRelease: release timer obj ffffc90001783128!
[  118.165092] RTMP_TimerListRelease: release timer obj ffffc90001782eb0!
[  118.165093] RTMP_TimerListRelease: release timer obj ffffc900017830b0!
[  118.177272] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 0, pAd->WlanFunCtrl:0x208, Reg-WlanFunCtrl=0x209
[  118.178635] -->RTUSBVenderReset
[  118.178778] <--RTUSBVenderReset
root@kali:~#

jurobystricky, 2016-01-24:
I pushed some changes that could/should affect your problem.
Based on your dmesg output, it seems you re-plugged the USB dongle.
If you do that, you should also restart the network manager, typically like this:

$ sudo service network-manager restart

Also, if the problem persists  you can increase the debug message level:
(The driver needs to be already loaded though)

$ sudo -i
# cd /sys/kernel/debug/rt2870/
# echo 4 > RTDebugLevel 
#exit

jurobystricky, 2016-01-24:
Well, I should have used "Preview" before commenting...
(Automatic formatting)

veaa, 2016-01-25:
Hi Juro,

The problem is still there. I'm sorry causing you this much trouble.

The first log is normal DMESG from the very beginning to end. Possibly, if you can see some underlying problems in my system.

The second log is with increased debug level.

Regards,
Vesa
 

root@kali:~# dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.0.0-kali1-amd64 (debian-kernel@lists.debian.org) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP Debian 4.0.4-1+kali2 (2015-06-03)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.0.0-kali1-amd64 root=UUID=d81030a6-6116-4a12-9f92-225d6a5681c2 ro initrd=/install/initrd.gz quiet
[    0.000000] Disabled fast string operations
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffecfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffed000-0x000000007ffedfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ffee000-0x000000007fffafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007fffb000-0x000000007fffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fd7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Parallels Software International Inc. Parallels Virtual Platform/Parallels Virtual Platform, BIOS 11.1.0 (32202) 11/09/2015
[    0.000000] Hypervisor detected: KVM
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] AGP: No AGP bridge found
[    0.000000] e820: last_pfn = 0x7ffed max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-C7FFF write-protect
[    0.000000]   C8000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] PAT configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.000000] found SMP MP-table at [mem 0x0009fc60-0x0009fc6f] mapped at [ffff88000009fc60]
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 24576
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000]  [mem 0x00000000-0x000fffff] page 4k
[    0.000000] BRK [0x01b12000, 0x01b12fff] PGTABLE
[    0.000000] BRK [0x01b13000, 0x01b13fff] PGTABLE
[    0.000000] BRK [0x01b14000, 0x01b14fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x7fc00000-0x7fdfffff]
[    0.000000]  [mem 0x7fc00000-0x7fdfffff] page 2M
[    0.000000] BRK [0x01b15000, 0x01b15fff] PGTABLE
[    0.000000] init_memory_mapping: [mem 0x60000000-0x7fbfffff]
[    0.000000]  [mem 0x60000000-0x7fbfffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x40000000-0x5fffffff]
[... 1128 lines cut ...]
[  368.565265] frame_type = d0, req = 1 , (2)
[  368.565266] [1] TYPE pAd->Cfg80211RegisterActionFrame=1[13]
[  368.565272] 80211> CFG80211_OpsMgmtFrameRegister ==>
[  368.565273] frame_type = d0, req = 1 , (2)
[  368.565274] [1] TYPE pAd->Cfg80211RegisterActionFrame=1[14]
[  368.565294] CFG80211NetdevNotifierEvent ndev:ffff880036596000  state:13
[  368.565296] CFG80211NetdevNotifierEvent default: [13]
[  368.566119] MAC[Ver:Rev=0x76623000 : 0x76120044]
[  368.568555] -->RTUSBVendorReset
[  368.588685] <--RTUSBVendorReset
[  368.588687] build time = 
[  368.588688] 20140408060640a
[  368.588691] rom patch for E3 IC
[  368.588691] 
platform = 
[  368.588692] ALPS
hw/sw version = 
[  368.588693] 0x8a 0x10 0x8a 0x10 
patch version = 
[  368.588694] 
[  368.590410] loading rom patchpatch_len = 20656
[  368.590414] cur_len = 0
[  368.590415] sent_len = 2040
[  368.590417] rom_patch_offset = 90000
[  368.591392] andes_usb_load_rom_patch: submit urb, sent_len = 2040, patch_ilm = 20656, cur_len = 2040

Litros, 2016-01-29:
Hi an awesome job here trying to run this devices under linux. I see that you get it, with the Netgear-A6210; I was reading A LOT in google before getting here... and you are the first to achieve this. 
By the way, I have the same problem as Veaa. Im running Ubuntu 15.1 with kernel 4.1 / with Edup EP-AC1601 with the same chipset (MT7612U). Trying to get it work with no good results... Hope you can help us with this in the future.
Very Thankful for everything

jurobystricky, 2016-01-29:
I'll try to order Edup EP-AC1601 so I can test it on my machine to see what's going on there.
BTW,  are there any LEDs ?

veaa, 2016-01-31:
Honestly I'm not sure. On the front side there is a led looking stripe but I have not seen it operating ever. Of course I could install the driver in my windows machine and see if it is working... I'll let you khow.

veaa, 2016-02-01:
Installed the driver and stick to Windows 7 64bit (Parallels VM) and the blue led light operates. As well the network itself is operating flawlessly.

Litros, 2016-02-08:
The same as Veaa again... Tried with windows 7 and XP as well; there it works great!... but nothing with ubuntu :(...
jurobystricky, I really appreciate your help; I'll always be looking here for some possible update of your usefull work!

jurobystricky, 2016-02-11:
OK, so I just ordered Edup EP-AC1601, should arrive in a week, so stay tuned...

veaa, 2016-02-11:
That's good to hear. I will wait patiently...

madmalkav, 2016-02-23:
I just compiled the module for this EDUP device and it is working ok.

jurobystricky, 2016-02-23:
Good to know you got it running. Which dongle exactly are you using?  LED is not enabled. Waiting (still) for a dongle with LED on it (Edup EP-AC1601), coming on a slow boat from China.. I'll try to get it functional then.
Performance-wise: On a 5GHz band I get over 10-12MB/sec (10-12 megabytes/second), both upload/download. Not too scientific, just reading/writing some huge files via Google drive.

madmalkav, 2016-02-23:
EDUP-AC1601. I'm using 2.4Ghz, 40Mhz width, 802.11n. My WAN is just 30Mb/sec, so I test on LAN, where I need to squeeze every single bit of speed I can :)

paolomeraviglia, 2016-02-24:
@madmalkav can you please write your kernel version? I have debian 4.3.0, a edup ep-ac1605 that has the same chip as your but after compile if I attach or detach the usb dongle, the system stuck in kernel panic... if you are interested the problem is reported here: https://github.com/jurobystricky/Netgear-A6210/issues/6

Did you modified the something?

madmalkav, 2016-02-24:
4.5rc5. I haven't got time to test much, but I got some disconnects. No
kernel panic on unplug but I have to restart network manager service before
replugging.

paolomeraviglia, 2016-02-24:
@madmalkav thanks. what distro do you use? I tried with some different versions of debian / ubuntu from kernel 3.16. to kernel 4.4.0 and I got the same KP lock problem...

madmalkav, 2016-02-24:
Manjaro Linux

madmalkav, 2016-03-05:
Today I had some time to play with the EDUP dongle again. Here you have two dmesg dumps with thte latest code on the repo:

[10362.976001] ------------[ cut here ]------------
[10362.976041] WARNING: CPU: 0 PID: 536 at /home/madmalkav/github/Netgear-A6210/os/linux/../../phy/phy.c:55 NICInitBBP+0x9e/0xd0 [mt7662u_sta]()
[10362.976044] Modules linked in: mt7662u_sta(O) uas usb_storage fuse btrfs xor raid6_pq ufs hfsplus hfs minix ntfs vfat msdos fat jfs xfs libcrc32c crc32c_generic dm_mod nvram video msr cfg80211 rfkill r8712u(C) snd_usb_audio snd_usbmidi_lib uvcvideo snd_rawmidi snd_seq_device videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev joydev media mousedev sch_fq_codel nfs lockd grace sunrpc fscache snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel snd_hda_codec fglrx(O) snd_hda_core snd_hwdep snd_pcm snd_timer snd soundcore kvm_amd kvm irqbypass crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper mxm_wmi cryptd nuvoton_cir edac_mce_amd tg3 ptp pps_core libphy evdev input_leds led_class
[10362.976115]  mac_hid serio_raw fjes wmi rc_core edac_core shpchp k10temp sp5100_tco i2c_piix4 fam15h_power acpi_cpufreq tpm_tis tpm processor button amd_iommu_v2 nct6775 hwmon_vid sg ip_tables x_tables ext4 crc16 mbcache jbd2 sd_mod ata_generic pata_acpi hid_roccat_koneplus hid_roccat hid_roccat_common hid_generic usbhid hid atkbd libps2 pata_atiixp ohci_pci ahci xhci_pci libahci ohci_hcd ehci_pci xhci_hcd ehci_hcd libata scsi_mod usbcore usb_common i8042 serio
[10362.976227] CPU: 0 PID: 536 Comm: NetworkManager Tainted: G        WC O    4.5.0-1-MANJARO #1
[10362.976231] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./990FX Extreme3, BIOS P1.90 08/04/2015
[10362.976236]  0000000000000286 00000000bb66a085 ffff88023463f5d0 ffffffff812e43d1
[10362.976242]  0000000000000000 ffffffffa176eb58 ffff88023463f608 ffffffff81079a02
[10362.976247]  0000000000000000 ffffc900062ea000 0000000000000001 0000000000000000
[10362.976251] Call Trace:
[10362.976261]  [<ffffffff812e43d1>] dump_stack+0x63/0x82
[10362.976268]  [<ffffffff81079a02>] warn_slowpath_common+0x82/0xc0
[10362.976274]  [<ffffffff81079b4a>] warn_slowpath_null+0x1a/0x20
[10362.976294]  [<ffffffffa1714b3e>] NICInitBBP+0x9e/0xd0 [mt7662u_sta]
[10362.976317]  [<ffffffffa16ec31d>] NICInitializeAdapter+0x17d/0x4f0 [mt7662u_sta]
[10362.976342]  [<ffffffffa16ef166>] rt28xx_init+0x326/0x600 [mt7662u_sta]
[10362.976369]  [<ffffffffa1727817>] rt28xx_open+0xc7/0x150 [mt7662u_sta]
[10362.976395]  [<ffffffffa16ff197>] RTMP_COM_IoctlHandle+0x6d7/0x790 [mt7662u_sta]
[10362.976420]  [<ffffffffa1727a91>] MainVirtualIF_open+0x51/0xe0 [mt7662u_sta]
[10362.976449]  [<ffffffffa1727750>] ? RT28xx_get_ether_stats+0x140/0x140 [mt7662u_sta]
[10362.976472]  [<ffffffffa17279c0>] ? rt28xx_ioctl+0x80/0x80 [mt7662u_sta]
[10362.976479]  [<ffffffff814b2668>] __dev_open+0xc8/0x140
[10362.976488]  [<ffffffff814b297d>] __dev_change_flags+0x9d/0x160
[10362.976492]  [<ffffffff814b2a69>] dev_change_flags+0x29/0x70
[10362.976497]  [<ffffffff814c1f0d>] do_setlink+0x32d/0xb90
[10362.976502]  [<ffffffff81171b5d>] ? __alloc_pages_nodemask+0x14d/0xbf0
[10362.976508]  [<ffffffff815b6180>] ? _raw_read_unlock_bh+0x20/0x30
[10362.976513]  [<ffffffff8130fbc1>] ? __nla_reserve+0x41/0x50
[10362.976517]  [<ffffffff8130fbc1>] ? __nla_reserve+0x41/0x50
[10362.976521]  [<ffffffff8130f9b2>] ? nla_parse+0x32/0x100
[10362.976526]  [<ffffffff814c2e5a>] rtnl_newlink+0x5aa/0x8a0
[10362.976535]  [<ffffffff81293508>] ? security_capable+0x48/0x60
[10362.976539]  [<ffffffff81081f59>] ? ns_capable+0x29/0x50
[10362.976544]  [<ffffffff814c1326>] rtnetlink_rcv_msg+0xe6/0x220
[10362.976549]  [<ffffffff811ce705>] ? __kmalloc_node_track_caller+0x35/0x320
[10362.976553]  [<ffffffff8149bd47>] ? __alloc_skb+0x87/0x1f0
[10362.976557]  [<ffffffff814c1240>] ? rtnetlink_rcv+0x30/0x30
[10362.976562]  [<ffffffff814e3104>] netlink_rcv_skb+0xa4/0xc0
[10362.976567]  [<ffffffff814c1238>] rtnetlink_rcv+0x28/0x30
[10362.976575]  [<ffffffff814e27bd>] netlink_unicast+0x11d/0x1b0
[10362.976580]  [<ffffffff814e2e18>] netlink_sendmsg+0x4f8/0x610
[10362.976584]  [<ffffffff81492977>] sock_sendmsg+0x17/0x30
[10362.976588]  [<ffffffff81493092>] ___sys_sendmsg+0x292/0x2a0
[10362.976593]  [<ffffffff81555786>] ? unix_dgram_sendmsg+0x336/0x6f0
[10362.976598]  [<ffffffff81492977>] ? sock_sendmsg+0x17/0x30
[10362.976602]  [<ffffffff81493d7d>] ? SyS_sendto+0x15d/0x190
[10362.976607]  [<ffffffff8120af67>] ? __fget+0x77/0xb0
[10362.976611]  [<ffffffff8120b005>] ? __fget_light+0x25/0x60
[10362.976616]  [<ffffffff814941e1>] __sys_sendmsg+0x51/0x90
[10362.976622]  [<ffffffff81494232>] SyS_sendmsg+0x12/0x20
[10362.976627]  [<ffffffff815b666e>] entry_SYSCALL_64_fastpath+0x12/0x6d
[10362.976631] ---[ end trace c1157aa24405cebb ]---

[12997.461599] ------------[ cut here ]------------
[12997.461638] WARNING: CPU: 3 PID: 536 at /home/madmalkav/github/Netgear-A6210/os/linux/../../phy/phy.c:55 NICInitBBP+0x9e/0xd0 [mt7662u_sta]()
[12997.461642] Modules linked in: mt7662u_sta(O) uas usb_storage fuse btrfs xor raid6_pq ufs hfsplus hfs minix ntfs vfat msdos fat jfs xfs libcrc32c crc32c_generic dm_mod nvram video msr cfg80211 rfkill r8712u(C) snd_usb_audio snd_usbmidi_lib uvcvideo snd_rawmidi snd_seq_device videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev joydev media mousedev sch_fq_codel nfs lockd grace sunrpc fscac [... cut ...]
[12997.462102]  [<ffffffff81555786>] ? unix_dgram_sendmsg+0x336/0x6f0
[12997.462107]  [<ffffffff81492977>] ? sock_sendmsg+0x17/0x30
[12997.462111]  [<ffffffff81493d7d>] ? SyS_sendto+0x15d/0x190
[12997.462116]  [<ffffffff8120af67>] ? __fget+0x77/0xb0
[12997.462120]  [<ffffffff8120b005>] ? __fget_light+0x25/0x60
[12997.462125]  [<ffffffff814941e1>] __sys_sendmsg+0x51/0x90
[12997.462129]  [<ffffffff81494232>] SyS_sendmsg+0x12/0x20
[12997.462134]  [<ffffffff815b666e>] entry_SYSCALL_64_fastpath+0x12/0x6d
[12997.462137] ---[ end trace c1157aa24405cebc ]---

jurobystricky, 2016-03-05:
Thanks, I put that warning in the code a while ago. Basically, there is a hardware  initialization  condition that fails, yet the routine NICInitBBP may report success in spite of that. The warning is there to notify us some assumptions about HW initialization are wrong.

madmalkav, 2016-03-05:
I always get one of those on every connection lost.

veaa, 2016-03-09:
I got a compilation error:

/root/Desktop/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsStaDel’:
/root/Desktop/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2112:6: error: ‘pMacAddr’ undeclared (first use in this function)
  if (pMacAddr == NULL) {
      ^
/root/Desktop/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2112:6: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/linux-headers-4.0.0-kali1-common/scripts/Makefile.build:263: recipe for target '/root/Desktop/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o' failed
make[4]: *** [/root/Desktop/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
/usr/src/linux-headers-4.0.0-kali1-common/Makefile:1407: recipe for target '_module_/root/Desktop/Netgear-A6210/os/linux' failed
make[3]: *** [_module_/root/Desktop/Netgear-A6210/os/linux] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.0.0-kali1-amd64'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

jurobystricky, 2016-03-09:
Bummer, please change the line 2106 from
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,2,0))
to
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,19,0))

I'll push the changes soon

veaa, 2016-03-10:
Now I got it compiled but still something:

[  126.756468] andes_usb_load_rom_patch: upload fw timeout
[  126.756475] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[  126.756596] receive cmd msg fail(-2)
[  126.756619] tx_kickout_fail_count = 0
[  126.756620] tx_timeout_fail_count = 0
[  126.756621] rx_receive_fail_count = 0
[  126.756622] alloc_cmd_msg = 1
[  126.756623] free_cmd_msg = 1
[  126.756624] !!! rt28xx init fail !!!
[  126.758450] -->RTUSBVendorReset
[  126.778598] <--RTUSBVendorReset
[  128.780588] andes_usb_load_rom_patch: upload fw timeout
[  128.780595] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[  128.780682] receive cmd msg fail(-2)
[  128.780700] tx_kickout_fail_count = 0
[  128.780701] tx_timeout_fail_count = 0
[  128.780702] rx_receive_fail_count = 0
[  128.780703] alloc_cmd_msg = 1
[  128.780704] free_cmd_msg = 1
[  128.780705] !!! rt28xx init fail !!!

veaa, 2016-07-27:
My case is still the same. Any ideas anybody?

artimtb, 2016-08-01:
Hi. i have same issue with edup and until now i have no idea how to make it work. my distro is ubuntu 16.04. waiting for any solutions too....

witi83, 2016-08-02:
My workaround is to remove the stick, insert it again and to restart the connection using NetworkManager, netctl, whatever.

artimtb, 2016-08-03:
Now i can use it but with no signal. Already try to remove then insert it again and restart the connection. Still no luck

jurobystricky, 2016-08-04:
Maybe a log output could help....

artimtb, 2016-08-06:
`[  691.367846] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  691.367849] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  691.367852] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[  691.367855] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2400 mBm), (0 s)
[  691.367857] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2400 mBm), (0 s)
[  691.367859] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[  691.749026] Rcv Wcid(1) AddBAReq
[  691.749033] Start Seq = 00000000
[  696.139658] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  700.199991] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  702.213016] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  704.103801] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  709.331999] RTMP_TimerListRelease: release timer obj ffffc900039bc028!
[  709.381351] cfg80211: World regulatory domain updated:
[  709.381358] cfg80211:  DFS Master region: unset
[  709.381360] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  709.381363] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  709.381366] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  709.381368] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  709.381371] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  709.381374] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  709.381376] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  709.381379] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  709.381381] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  715.836559] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  715.837568] 80211> Connect bssid 94:db:da:0c:e4:30
[  715.881466] PeerBeaconAtJoinAction(): HT-CtrlChannel=7, CentralChannel=>9
[  715.881472] PeerBeaconAtJoinAction(): Set CentralChannel=9
[  715.881476] AdjustChannelRelatedValue(): Input BW=1, rf_channel=9, vht_bw=1, Channel=7, vht_cent_ch=9!
[  715.939444] mt76x2_reinit_agc_gain:original agc_vga0 = 0x48, agc_vga1 = 0x48
[  715.939450] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x48, agc_vga1 = 0x48
[  715.939944] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x33, hi_lna1 = 0x33
[  715.939946] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x33, hi_lna1 = 0x33
[  715.940580] original vga value(chain0) = 48
[  715.940699] original vga value(chain1) = 48
[  715.941257] mt76x2_calibration(channel = 9)
[  715.970891] cfg80211: Regulatory domain changed to country: SG
[  715.970898] cfg80211:  DFS Master region: FCC
[  715.970901] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  715.970904] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  715.970907] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[  715.970910] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2400 mBm), (0 s)
[  715.970912] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2400 mBm), (0 s)
[  715.970914] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[  717.202240] Rcv Wcid(1) AddBAReq
[  717.202247] Start Seq = 00000000
[  717.207610] !!! ba_enqueue_reordering_packet: (0) Can't allocate reordering mpdu blk
[  721.971205] RTMP_TimerListRelease: release timer obj ffffc900039bc028!
[  722.022905] cfg80211: World regulatory domain updated:
[  722.022912] cfg80211:  DFS Master region: unset
[  722.022914] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  722.022917] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  722.022920] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  722.022922] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  722.022924] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  722.022927] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  722.022930] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  722.022932] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  722.022934] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  728.478687] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[  728.479663] 80211> Connect bssid 94:db:da:0c:e4:30
[  728.524781] PeerBeaconAtJoinAction(): HT-CtrlChannel=7, CentralChannel=>9
[  728.524787] PeerBeaconAtJoinAction(): Set CentralChannel=9
[  728.524791] AdjustChannelRelatedValue(): Input BW=1, rf_channel=9, vht_bw=1, Channel=7, vht_cent_ch=9!
[  728.582700] mt76x2_reinit_agc_gain:original agc_vga0 = 0x48, agc_vga1 = 0x48
[  728.582707] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x48, agc_vga1 = 0x48
[  728.583391] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x33, hi_lna1 = 0x33
[... cut ...]
[  993.188956] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  993.188958] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  993.188960] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  993.188963] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

gsc@MT-01:/etc$ uname -a
Linux MT-01 4.4.0-21-lowlatency #37-Ubuntu SMP PREEMPT Mon Apr 18 20:20:26 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
`

artimtb, 2016-08-06:
above are the log. all freq N/A

dave562, 2016-10-03:
I am running into the same issue.

NetGear A6210

Linux kali 4.3.0-kali1-amd64 #1 SMP Debian 4.3.3-7kali2 (2016-01-27) x86_64 GNU/Linux

[   25.322620] usb 2-1: new high-speed USB device number 2 using ehci-pci
[   25.649704] usb 2-1: New USB device found, idVendor=0846, idProduct=9053
[   25.649706] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   25.649707] usb 2-1: Product: A6210
[   25.649708] usb 2-1: Manufacturer: NETGEAR
[   25.649708] usb 2-1: SerialNumber: 100
[   26.821781] cfg80211: World regulatory domain updated:
[   26.821784] cfg80211:  DFS Master region: unset
[   26.821784] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   26.821786] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   26.821786] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   26.821787] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   26.821788] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   26.821789] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   26.821789] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   26.821790] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   26.821791] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   27.424960] rtusb init rt2870 --->
[   27.425208] 
               === pAd = ffffc90001902000, size = 1292952 ===

[   27.425210] driver version: 3.0.0.1 (Oct  3 2016 14:33:04) .
[   27.428139] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20f
[   27.532128] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[   27.532130] RtmpEepromGetDefault::e2p_dafault=1
[   27.534660] NVM is EFUSE mode
[   27.534661] Endpoint(8) is for In-band Command
[   27.534662] Endpoint(4) is for WMM0 AC0
[   27.534663] Endpoint(5) is for WMM0 AC1
[   27.534663] Endpoint(6) is for WMM0 AC2
[   27.534664] Endpoint(7) is for WMM0 AC3
[   27.534664] Endpoint(9) is for WMM1 AC0
[   27.534664] Endpoint(84) is for Data-In
[   27.534665] Endpoint(85) is for Command Rsp
[   27.536115] 80211> CurTxPower = 20 dBm
[   27.536116] ====> Radar Channel 52
[   27.536117] ====> Radar Channel 54
[   27.536117] ====> Radar Channel 56
[   27.536117] ====> Radar Channel 60
[   27.536118] ====> Radar Channel 62
[   27.536118] ====> Radar Channel 64
[   27.536119] ====> Radar Channel 100
[   27.536119] ====> Radar Channel 104
[   27.536120] 80211> TxStream = 0
[   27.544908] usbcore: registered new interface driver rt2870
[   27.568034] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   27.574119] -->RTUSBVendorReset
[   27.595340] <--RTUSBVendorReset
[   27.608648] .....RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x9a8,pAd->Flags=0x0

mail2onur, 2016-11-16:
I recently bought this EDUP EP-AC1601 wifi dongle and noticed that it doesn't work in Linux Mint 18, Kernel v4.4. This is such a shame because it performs so good in Win10 with excellent speed and quite high signal coverage.

I tried the driver from EDUP's website, Mediatek website and i have failed to compile these drivers as you can guess and then i found this github page while searching on Google. This driver is the only one which gets complied and installed without any errors but unfortunately i couldn't make the wifi dongle work no matter whet i tried. This is so strange because i didn't see any error while installing this driver. 

I tried resetting network manager via terminal, remove&insert the usb dongle, restarting pc but none helps. I don't see any activity in network manager.  I can see the device getting removed and inserted via lsusb and dpmesg commands but it doesnt call the necessary firmware from kernel to make it work.

Can anyone help me to get this work in Linux Mint 18 because i don't wanna return to Win10 just to be able to connect to the internet!