jurobystricky/Netgear-A6210 #3: Wifi activity LED not working

MrEngman, 2016-01-16:
Hi, 

I realise the A6210 does not appear to use an activity LED but I have compiled the driver with the ASUS USB-N53_B1 USB id added for the Raspberry Pi and the LED does not work on that one, otherwise it appears to be working just fine. Good job you've done.

Can you advise me on how I can get the LED working?

MrEngman

jurobystricky, 2016-01-17:
Yes, there is no LED on the Netgear USB dongle, I wish there was. Had there been one, implementing a proper LED indication would be very high on my priority list, so I understand your point.
 In principle, implementing LED indicators should be not too difficult. The code for it is in various other Ralink drivers. I would/could do it, but I don't have any means to test it. I'll try to get my hands on one of those ASUS USB-N53_B1 USB dongles.
The driver actually did contain the function "andes_led_op" (file mcu_and.c), but I removed it, as it was not used. You can try googling "RTMPSetLEDStatus" or "CMDTHREAD_SET_LED_STATUS", and probably reuse the LED related code. In any case, please keep me posted.

MrEngman, 2016-01-17:
Thanks for the pointer. I am not a programmer so whether I can do anything I don't know. If you want though I can test your code if you update it.

I let you know if I get it working.

edsnor, 2016-03-07:
I got the "Netgear WNDA3100v3 N600 WiFi USB Adapter—802.11n Dual Band", chip MediaTek MT7632U, working in Ubuntu 15.10. Only the status LED's are not working. I added the line
{USB_DEVICE(0x0846, 0x9014), .driver_info = RLT_MAC_BASE}, /* MT7632U, Netgear WNDA3100v3 <====== */
to the list in rtusb_dev_id.c and had to create directory /etc/Wireless

jurobystricky, 2016-03-07:
Thanks, that is good news.
I happened to merge the entry for Netgear WNDA3100v3 just a few day ago, commit aed9db95dc8d7bd4b711e83bc67893bca22c14e3
(patch by Joni Palosaari). 
I will try to get the LED done at some point. I will look into /etc/Wireless, it should be created automatically by "sudo make install", there should be no need to create it manually.

paolomeraviglia, 2016-04-01:
Hi, excuse me: do you have any news?