jurobystricky/Netgear-A6210 #4: Seems to break after sleep mode

cory2067, 2016-01-22:
After turning the computer back on from sleep mode, the driver doesn't work right and won't connect to networks.

Furthermore, when trying to play with the network settings to resolve it, the system hangs and needs to be forcefully restarted.

(Linux Mint 17.3)

jurobystricky, 2016-01-22:
Cannot say I am surprised. If possible, please describe the steps to reproduce the problem.
I test on Ubuntu 15, turning off wireless does not cause problems there. At present I am working on hot unplugging/re-plugging. There are some problems there. I think once I fix that all kinds of other problems will go away as well as a by-product.

cory2067, 2016-01-22:
All I tried was:
1. Boot linux
2. Turn wireless on via GUI
3. Attempt to turn it off, system hangs

andy5995, 2016-02-23:
Sounds like what may have happened to me when I disabled and re-enabled. I'm running Debian Jessie 8 x64. The module built great! Is this information helpful? (The RIP message is at the bottom)

[code]Feb 12 09:33:28 debian64aa kernel: [15584.320237] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Feb 12 11:00:29 debian64aa kernel: [20805.288433] usb 1-1.4: new high-speed USB device number 6 using ehci-pci
Feb 12 11:00:29 debian64aa kernel: [20805.382147] usb 1-1.4: New USB device found, idVendor=0846, idProduct=9053
Feb 12 11:00:29 debian64aa kernel: [20805.382153] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 12 11:00:29 debian64aa kernel: [20805.382156] usb 1-1.4: Product: A6210
Feb 12 11:00:29 debian64aa kernel: [20805.382160] usb 1-1.4: Manufacturer: NETGEAR
Feb 12 11:00:29 debian64aa kernel: [20805.382163] usb 1-1.4: SerialNumber: 100
Feb 12 11:00:30 debian64aa kernel: [20806.428892] rtusb init rt2870 --->
Feb 12 11:00:30 debian64aa kernel: [20806.429054] 
Feb 12 11:00:30 debian64aa kernel: [20806.429054] === pAd = ffffc900272da000, size = 1292952 ===
Feb 12 11:00:30 debian64aa kernel: [20806.429054] 
Feb 12 11:00:30 debian64aa kernel: [20806.429056] driver version: 3.0.0.1 (Feb 12 2016 10:59:23) .
Feb 12 11:00:30 debian64aa kernel: [20806.429825] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
Feb 12 11:00:30 debian64aa kernel: [20806.445041] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
Feb 12 11:00:30 debian64aa kernel: [20806.445046] RtmpEepromGetDefault::e2p_dafault=1
Feb 12 11:00:30 debian64aa kernel: [20806.445411] NVM is EFUSE mode
Feb 12 11:00:30 debian64aa kernel: [20806.445413] Endpoint(8) is for In-band Command
Feb 12 11:00:30 debian64aa kernel: [20806.445415] Endpoint(4) is for WMM0 AC0
Feb 12 11:00:30 debian64aa kernel: [20806.445417] Endpoint(5) is for WMM0 AC1
Feb 12 11:00:30 debian64aa kernel: [20806.445419] Endpoint(6) is for WMM0 AC2
Feb 12 11:00:30 debian64aa kernel: [20806.445421] Endpoint(7) is for WMM0 AC3
Feb 12 11:00:30 debian64aa kernel: [20806.445423] Endpoint(9) is for WMM1 AC0
Feb 12 11:00:30 debian64aa kernel: [20806.445424] Endpoint(84) is for Data-In
Feb 12 11:00:30 debian64aa kernel: [20806.445426] Endpoint(85) is for Command Rsp
Feb 12 11:00:30 debian64aa kernel: [20806.445671] 80211> CurTxPower = 20 dBm
Feb 12 11:00:30 debian64aa kernel: [20806.445674] ====> Radar Channel 52
Feb 12 11:00:30 debian64aa kernel: [20806.445676] ====> Radar Channel 54
Feb 12 11:00:30 debian64aa kernel: [20806.445677] ====> Radar Channel 56
Feb 12 11:00:30 debian64aa kernel: [20806.445679] ====> Radar Channel 60
Feb 12 11:00:30 debian64aa kernel: [20806.445681] ====> Radar Channel 62
Feb 12 11:00:30 debian64aa kernel: [20806.445683] ====> Radar Channel 64
Feb 12 11:00:30 debian64aa kernel: [20806.445685] ====> Radar Channel 100
Feb 12 11:00:30 debian64aa kernel: [20806.445687] ====> Radar Channel 104
Feb 12 11:00:30 debian64aa kernel: [20806.445689] 80211> TxStream = 0
Feb 12 11:00:30 debian64aa kernel: [20806.446399] usbcore: registered new interface driver rt2870
Feb 12 11:00:30 debian64aa kernel: [20806.453169] -->RTUSBVendorReset
Feb 12 11:00:30 debian64aa kernel: [20806.473120] <--RTUSBVendorReset
Feb 12 11:00:30 debian64aa kernel: [20806.475257] ...........andes_usb_chk_crc
Feb 12 11:00:30 debian64aa kernel: [20806.569013] andes_usb_reset_wmt
Feb 12 11:00:30 debian64aa kernel: [20806.589553] -->RTUSBVendorReset
Feb 12 11:00:30 debian64aa kernel: [20806.609506] <--RTUSBVendorReset
Feb 12 11:00:30 debian64aa kernel: [20806.609919] fw version:0.0.00 build:1
Feb 12 11:00:30 debian64aa kernel: [20806.609923] build time:201406241830____
Feb 12 11:00:30 debian64aa kernel: [20806.609933] fw for E3 IC
Feb 12 11:00:30 debian64aa kernel: [20806.609935] ilm length = 59312(bytes)
Feb 12 11:00:30 debian64aa kernel: [20806.609937] dlm length = 32068(bytes)
Feb 12 11:00:30 debian64aa kernel: [20806.610766] loading fw........
Feb 12 11:00:30 debian64aa kernel: [20806.671473] cfg_mode=5
Feb 12 11:00:30 debian64aa kernel: [20806.671597] Key1Str is Invalid key length(0) or Type(0)
Feb 12 11:00:30 debian64aa kernel: [20806.671607] Key2Str is Invalid key length(0) or Type(0)
Feb 12 11:00:30 debian64aa kernel: [20806.671616] Key3Str is Invalid key length(0) or Type(0)
Feb 12 11:00:30 debian64aa kernel: [20806.671626] Key4Str is Invalid key length(0) or Type(0)
Feb 12 11:00:30 debian64aa kernel: [20806.671845] USBAggregation = 1
Feb 12 11:00:30 debian64aa kernel: [20806.671847] 1. Phy Mode = 31
Feb 12 11:00:30 debian64aa kernel: [20806.671848] NVM is Efuse and its size =1d[1e0-1fc] 
Feb 12 11:00:30 debian64aa kernel: [20806.701830] get_chl_grp:illegal channel (167)
Feb 12 11:00:30 debian64aa kernel: [20806.701834] get_chl_grp:illegal channel (167)
Feb 12 11:00:30 debian64aa kernel: [20806.701836] get_chl_grp:illegal channel (169)
Feb 12 11:00:30 debian64aa kernel: [20806.701838] get_chl_grp:illegal channel (169)
Feb 12 11:00:30 debian64aa kernel: [20806.701840] get_chl_grp:illegal channel (171)
Feb 12 11:00:30 debian64aa kernel: [20806.701842] get_chl_grp:illegal channel (171)
Feb 12 11:00:30 debian64aa kernel: [20806.701843] get_chl_grp:illegal channel (173)
Feb 12 11:00:30 debian64aa kernel: [20806.701845] get_chl_grp:illegal channel (173)
Feb 12 11:00:30 debian64aa kernel: [20806.704456] Country Region from e2p = ffff
[... cut ...]
Feb 12 11:04:43 debian64aa kernel: [21059.427720] Code: 48 c7 c7 6f 1b 95 a0 31 c0 e8 d1 4f be e0 8b 55 04 e9 97 f6 ff ff 31 c0 48 c7 c6 40 b0 94 a0 48 c7 c7 78 5a 96 a0 e8 b4 4f be e0 <80> bb 51 a3 13 00 01 0f 85 17 ee ff ff 83 3d 38 86 06 00 01 0f 
Feb 12 11:04:43 debian64aa kernel: [21059.427938] RIP  [<ffffffffa0926074>] CFG80211DRV_IoctlHandle+0x1364/0x14f0 [mt7662u_sta]
Feb 12 11:04:43 debian64aa kernel: [21059.427970]  RSP <ffff88042b8fba08>
Feb 12 11:04:43 debian64aa kernel: [21059.427981] CR2: 000000140013a355
Feb 12 11:04:43 debian64aa kernel: [21059.435143] ---[ end trace 1010fb61bd442416 ]---[/code]

jurobystricky, 2016-02-24:
Thanks for the log, will try to figure out what's going on. (Logs are always appreciated)

andy5995, 2016-02-25:
You're welcome. I still have the build directory; need any logs from there?

darreal44, 2016-04-30:
In my case (gihub version of today on ubuntu 14.04)

`Apr 26 15:17:24 newsoroh kernel: [16312.089114] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 0, pAd->WlanFunCtrl:0x208, Reg-WlanFunCtrl=0x209
Apr 26 15:17:24 newsoroh kernel: [16312.089885] -->RTUSBVendorReset
Apr 26 15:17:24 newsoroh kernel: [16312.109817] <--RTUSBVendorReset
Apr 26 15:17:24 newsoroh kernel: [16312.111113] ...........andes_usb_chk_crc
Apr 26 15:17:24 newsoroh kernel: [16312.200087] andes_usb_reset_wmt
Apr 26 15:17:24 newsoroh kernel: [16312.220878] -->RTUSBVendorReset
Apr 26 15:17:24 newsoroh kernel: [16312.240807] <--RTUSBVendorReset
Apr 26 15:17:24 newsoroh kernel: [16312.241568] fw version:0.0.00 build:1
Apr 26 15:17:24 newsoroh kernel: [16312.241571] build time:201406241830____
Apr 26 15:17:24 newsoroh kernel: [16312.241580] fw for E3 IC
Apr 26 15:17:24 newsoroh kernel: [16312.241582] ilm length = 59312(bytes)
Apr 26 15:17:24 newsoroh kernel: [16312.241583] dlm length = 32068(bytes)
Apr 26 15:17:24 newsoroh kernel: [16312.242015] loading fw........
Apr 26 15:17:24 newsoroh kernel: [16312.297620] cfg_mode=5
Apr 26 15:17:24 newsoroh kernel: [16312.297814] Key1Str is Invalid key length(0) or Type(0)
Apr 26 15:17:24 newsoroh kernel: [16312.297828] Key2Str is Invalid key length(0) or Type(0)
Apr 26 15:17:24 newsoroh kernel: [16312.297843] Key3Str is Invalid key length(0) or Type(0)
Apr 26 15:17:24 newsoroh kernel: [16312.297858] Key4Str is Invalid key length(0) or Type(0)
Apr 26 15:17:24 newsoroh kernel: [16312.298208] USBAggregation = 1
Apr 26 15:17:24 newsoroh kernel: [16312.298212] 1. Phy Mode = 31
Apr 26 15:17:24 newsoroh kernel: [16312.298214] NVM is Efuse and its size =1d[1e0-1fc] 
Apr 26 15:17:24 newsoroh kernel: [16312.310490] get_chl_grp:illegal channel (167)
Apr 26 15:17:24 newsoroh kernel: [16312.310492] get_chl_grp:illegal channel (167)
Apr 26 15:17:24 newsoroh kernel: [16312.310494] get_chl_grp:illegal channel (169)
Apr 26 15:17:24 newsoroh kernel: [16312.310495] get_chl_grp:illegal channel (169)
Apr 26 15:17:24 newsoroh kernel: [16312.310497] get_chl_grp:illegal channel (171)
Apr 26 15:17:24 newsoroh kernel: [16312.310498] get_chl_grp:illegal channel (171)
Apr 26 15:17:24 newsoroh kernel: [16312.310500] get_chl_grp:illegal channel (173)
Apr 26 15:17:24 newsoroh kernel: [16312.310501] get_chl_grp:illegal channel (173)
Apr 26 15:17:24 newsoroh kernel: [16312.311536] Country Region from e2p = ffff
Apr 26 15:17:24 newsoroh kernel: [16312.316146] mt76x2_get_external_lna_gain::LNA type=0x0, BLNAGain=0xffffff8d, ALNAGain0=0xffffff89, ALNAGain1=0xffffff88, ALNAGain2=0xffffff87
Apr 26 15:17:24 newsoroh kernel: [16312.316150] 2. Phy Mode = 31
Apr 26 15:17:24 newsoroh kernel: [16312.316152] 3. Phy Mode = 31
Apr 26 15:17:24 newsoroh kernel: [16312.316247] andes_usb_fw_init
Apr 26 15:17:24 newsoroh kernel: [16312.377719] AntCfgInit: primary/secondary ant 0/1
Apr 26 15:17:24 newsoroh kernel: [16312.377724] andes_load_cr:cr_type(2)
Apr 26 15:17:24 newsoroh kernel: [16312.378301] ChipStructAssign(): MT76x2 hook !
Apr 26 15:17:24 newsoroh kernel: [16312.378393] ---> InitFrequencyCalibration
Apr 26 15:17:24 newsoroh kernel: [16312.378395] InitFrequencyCalibrationMode:Unknow mode = 3
Apr 26 15:17:24 newsoroh kernel: [16312.378398] InitFrequencyCalibration: frequency offset in the EEPROM = 168
Apr 26 15:17:24 newsoroh kernel: [16312.378399] <--- InitFrequencyCalibration
Apr 26 15:17:24 newsoroh kernel: [16312.378412] RTMPSetPhyMode: channel is out of range, use first channel=1 
Apr 26 15:17:24 newsoroh kernel: [16312.378414] RTMPSetPhyMode: Update for STA
Apr 26 15:17:24 newsoroh kernel: [16312.380164] MCS Set = ff ff 00 00 01
Apr 26 15:17:24 newsoroh kernel: [16312.380235] 80211> re-init bands...
Apr 26 15:17:24 newsoroh kernel: [16312.380239] 80211> CurTxPower = 20 dBm
Apr 26 15:17:24 newsoroh kernel: [16312.380241] ====> Radar Channel 52
Apr 26 15:17:24 newsoroh kernel: [16312.380243] ====> Radar Channel 54
Apr 26 15:17:24 newsoroh kernel: [16312.380244] ====> Radar Channel 56
Apr 26 15:17:24 newsoroh kernel: [16312.380246] ====> Radar Channel 60
Apr 26 15:17:24 newsoroh kernel: [16312.380247] ====> Radar Channel 62
Apr 26 15:17:24 newsoroh kernel: [16312.380249] ====> Radar Channel 64
Apr 26 15:17:24 newsoroh kernel: [16312.380250] ====> Radar Channel 100
Apr 26 15:17:24 newsoroh kernel: [16312.380252] ====> Radar Channel 104
Apr 26 15:17:24 newsoroh kernel: [16312.380254] 80211> TxStream = 2
Apr 26 15:17:24 newsoroh kernel: [16312.380259] Chan 014 (frq 2484):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380264] Chan 149 (frq 5745):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380266] Chan 151 (frq 5755):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380268] Chan 153 (frq 5765):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380269] Chan 157 (frq 5785):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380271] Chan 159 (frq 5795):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380273] Chan 161 (frq 5805):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380274] Chan 165 (frq 5825):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380276] Chan 167 (frq 5835):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380277] Chan 169 (frq 5845):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380279] Chan 171 (frq 5855):	not allowed!
Apr 26 15:17:24 newsoroh kernel: [16312.380281] Chan 173 (frq 5865):	not allowed!
[... cut ...]
Apr 26 15:17:24 newsoroh kernel: [16312.657112] Code: ed ff ff 3c 01 0f 95 c0 0f b6 c0 e9 5d f8 ff ff 0f 1f 00 48 83 f9 09 0f 85 4e f8 ff ff 8b 05 68 86 06 00 85 c0 0f 85 e5 07 00 00 <80> bb 51 a3 13 00 01 0f 85 33 f8 ff ff 83 3d 4c 86 06 00 01 76 
Apr 26 15:17:24 newsoroh kernel: [16312.661097] RIP  [<ffffffffa0616ca0>] CFG80211DRV_IoctlHandle+0x930/0x13c0 [mt7662u_sta]
Apr 26 15:17:24 newsoroh kernel: [16312.663069]  RSP <ffff8802ff0bb600>
Apr 26 15:17:24 newsoroh kernel: [16312.665026] CR2: 000000140013a355
Apr 26 15:17:24 newsoroh kernel: [16312.677848] ---[ end trace c2f1fc283f223e3a ]---
`

jurobystricky, 2016-05-04:
Thanks for the log. Can you describe steps you took to power down/up? Is this (kernel panic) random or repeatable?

darreal44, 2016-05-25:
You are welcome.
I am now under 16.04. With the lastest version of Netgear A6210 driver on this github, it doesn't happen anymore. (but it happened with a previous version of Netgear A6210 driver on this github)
I was using mate. When I clicked on "System -> shutdown -> stand by" It crashed.

ratnesh1729, 2017-01-08:
@darreal44 which driver version was good - meaning the one you are using ?

darreal44, 2017-01-17:
the current origin/master work well with my current 16.04.
The only issue is when the usb dongle is removed quickly it freezes all the system.