jurobystricky/Netgear-A6210 #5: Make error

MMelrose, 2016-02-01:
Make is throwing this up (Feb1)...

pi@raspberrypi:~/Netgear-A6210 $ sudo make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.1.16-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/pi/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.1.16-v7+/build: No such file or directory. Stop.
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

jurobystricky, 2016-02-01:
Did you install kernel headers, ie.e something like:
 # apt-get install kernel-headers-$(uname -r)

MMelrose, 2016-02-01:
i followed this walkthrough...

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=132089

the last time I tried it (I've done it 4 times now all with different results) it got this error...

pi@raspberrypi:~/Netgear-A6210 $ sudo make install
make -C /home/pi/Netgear-A6210/os/linux -f Makefile.6 install
make[1]: Entering directory '/home/pi/Netgear-A6210/os/linux'
rm -rf /etc/Wireless/RT2870STA
mkdir /etc/Wireless/RT2870STA
mkdir: cannot create directory ‘/etc/Wireless/RT2870STA’: No such file or directory
Makefile.6:724: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/home/pi/Netgear-A6210/os/linux'
Makefile:76: recipe for target 'install' failed
make: *** [install] Error 2

jurobystricky, 2016-02-01:
Just create the directory manually for the time being.

$ sudo mkdir -p /etc/Wireless/RT2870STA

Seems the Makefile is missing "-p"

MMelrose, 2016-02-01:
If i run that command I get this...

pi@raspberrypi:~ $ sudo apt-get install kernel-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kernel-headers-4.1.13-v7
E: Couldn't find any package by regex 'kernel-headers-4.1.13-v7'

jurobystricky, 2016-02-01:
Try this:

sudo apt-get install linux-headers-$(uname -r)

MMelrose, 2016-02-02:
and i get his...

pi@raspberrypi:~ $ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.1.17-v7
E: Couldn't find any package by regex 'linux-headers-4.1.17-v7'

MMelrose, 2016-02-02:
Hi, Thanks for all the help and fix on the mkdir, however, the sudo make is still throwing up an Error 2...

pi@raspberrypi:~/Netgear-A6210 $ sudo make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.1.17-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/pi/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.1.17-v7+/build: No such file or directory.  Stop.
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

Do you know whats gone amiss? thanks again (sorry be being a pain).

MMelrose, 2016-02-03:
Okay, so, i manually created the /build directory and ran sudo make...

pi@raspberrypi:~/Netgear-A6210 $ sudo make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.1.17-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/pi/Netgear-A6210/os/linux modules
make[1]: Entering directory '/lib/modules/4.1.17-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.1.17-v7+/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

????????

jurobystricky, 2016-02-03:
There should be a file Makefile in your folder /lib/modules/4.1.17-v7+/build.
(And the Makefile should contain target 'modules'.)
What is the content of the folder /lib/modules/4.1.17-v7+/build ?

jurobystricky, 2016-02-03:
Let me rephrase the questions, how did you create the folder /lib/modules/4.1.17-v7+/build?
It should be done by $ sudo apt-get install kernel-headers-$(uname -r)
Are you building the driver on RaspberryPi or are you cross-compling?

MMelrose, 2016-02-03:
HI, there was no /build folder in /lib/modules/4.1.17-v7 so i figured I'd see what happened if i created it by sudo su and mkdir, I had previously tried to run the command 

sudo apt-get install kernel-headers-$(uname -r)

as you mentioned earlier, as well as

sudo apt-get install linux-headers-$(uname -r)

and they both return the same error...

pi@raspberrypi:~ $ sudo apt-get install kernal-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kernal-headers-4.1.17-v7
E: Couldn't find any package by regex 'kernal-headers-4.1.17-v7'

pi@raspberrypi:~ $ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.1.17-v7
E: Couldn't find any package by regex 'linux-headers-4.1.17-v7'

Thanks again.

MMelrose, 2016-02-03:
...oh, and it's all on a pi.

thanks

jurobystricky, 2016-02-03:
The driver will not build without the kernel headers. You need to figure out how to install them on pi.
Seems like Raspbian does not provide them for whatever reason.

jurobystricky, 2016-02-04:
Try to follow these steps, seems others got this working on pi:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=132089&p=880874

MMelrose, 2016-02-04:
HI many thanks, I am now sending this message wirelessly! I followed the instructions, did a bit if jiggery pokery and found the missing headers at...

https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/

downloaded and installed them, and continued along the walkthrough.

awesome!

Many many thanks for your time and help.

jurobystricky, 2016-02-04:
Glad to hear that you got it running! It is a nice feeling when things begin to behave...

DANHDAN, 2017-01-11:
Hello both, I'm not sure if i'm allowed to add to this post (almost a year old) but i could do with some guidance if you wouldnt mind please. I am stuck with not being able to "make". I have installed headers and just checked my build folder but there is no makefile file.

Current errors:

dan@PI:~$ cd Netgear-A6210
dan@PI:~/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/dan/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.4.41-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/dan/Netgear-A6210/os/linux modules
make[1]: Entering directory '/lib/modules/4.4.41-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.4.41-v7+/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
dan@PI:~/Netgear-A6210$ 

Many thanks in advance