jurobystricky/Netgear-A6210 #6: Not working on 4.3.0 - EDUP EP-AC1605

paolomeraviglia, 2016-02-24:
for first: THANK YOU FOR YOUR GREAT WORK!

I have an EDUP-AC1605: all the driver compile ok, with some warnings.

I am running debian testing:

uname -a : `Linux debian-netbook 4.3.0-1-amd64 #1 SMP Debian 4.3.5-1 (2016-02-06) x86_64 GNU/Linux`

the USB product id is the same of EDUP-1601

The system go in kernel panic if i attach / detach the usb dongle (so I cannot restart network-manager... I need to push down power button and manually restart everything).

If I boot up with usb dongle inserted, generally in 2 or 3 mins the system hangs.

In these 2-3 minutes I cannot connect to any network wireless, BUT the scanning works: in fact I can see the networks around me.

the kernel panic log say:

`Feb 24 01:07:37 debian-netbook kernel: [  289.268321] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
Feb 24 01:07:37 debian-netbook kernel: [  289.268362] BUG: unable to handle kernel paging request at 000000000013a355
Feb 24 01:07:37 debian-netbook kernel: [  289.268431] IP: [<ffffffffa10e0551>] CFG80211DRV_IoctlHandle+0x10c1/0x13b0 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [  289.268516] PGD b82c5067 PUD b2330067 PMD 0 
Feb 24 01:07:37 debian-netbook kernel: [  289.268557] Oops: 0000 [#1] SMP 
Feb 24 01:07:37 debian-netbook kernel: [  289.268588] Modules linked in: rfcomm tun sha256_ssse3 sha256_generic hmac drbg ansi_cprng ctr ccm bnep nls_utf8 nls_cp437 vfat fat mt7662u_sta(O) intel_rapl intel_powerclamp coretemp kvm_intel arc4 uvcvideo kvm videobuf2_vmalloc snd_hda_codec_hdmi iwlmvm snd_hda_codec_realtek snd_hda_codec_generic videobuf2_memops mac80211 videobuf2_core v4l2_common asus_nb_wmi asus_wmi sparse_keymap iTCO_wdt iTCO_vendor_support videodev wl(PO) crct10dif_pclmul media btusb btrtl btbcm btintel bluetooth crc32_pclmul snd_soc_rt5640 snd_soc_rl6231 snd_soc_core i915 cryptd rtsx_pci_ms snd_hda_intel memstick snd_hda_codec iwlwifi snd_hda_core cfg80211 snd_hwdep drm_kms_helper joydev pcspkr evdev snd_compress sg serio_raw efi_pstore efivars rfkill wmi drm snd_pcm 8250_fintek battery ac video snd_timer dw_dmac regmap_i2c tpm_tis snd_soc_sst_acpi dw_dmac_core i2c_algo_bit tpm snd shpchp i2c_designware_platform i2c_i801 i2c_designware_core lpc_ich soundcore button processor iosf_mbi parport_pc ppdev lp parport efivarfs autofs4 ext4 crc16 mbcache jbd2 sd_mod rtsx_pci_sdmmc crc32c_intel psmouse rtsx_pci mfd_core r8169 mii ahci libahci xhci_pci libata xhci_hcd usbcore scsi_mod usb_common thermal sdhci_acpi sdhci mmc_core i2c_hid hid
Feb 24 01:07:37 debian-netbook kernel: [  289.283721] CPU: 1 PID: 481 Comm: NetworkManager Tainted: P        W  O    4.3.0-1-amd64 #1 Debian 4.3.5-1
Feb 24 01:07:37 debian-netbook kernel: [  289.286307] Hardware name: ASUSTeK COMPUTER INC. X200MA/X200MA, BIOS X200MA.504 11/13/2014
Feb 24 01:07:37 debian-netbook kernel: [  289.288883] task: ffff88009a2e5100 ti: ffff88009a2d0000 task.ti: ffff88009a2d0000
Feb 24 01:07:37 debian-netbook kernel: [  289.291451] RIP: 0010:[<ffffffffa10e0551>]  [<ffffffffa10e0551>] CFG80211DRV_IoctlHandle+0x10c1/0x13b0 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [  289.294097] RSP: 0018:ffff88009a2d3688  EFLAGS: 00010282
Feb 24 01:07:37 debian-netbook kernel: [  289.296732] RAX: 000000000000003c RBX: 0000000000000004 RCX: 0000000000000000
Feb 24 01:07:37 debian-netbook kernel: [  289.299387] RDX: 0000000000000000 RSI: ffff88013fd0dd58 RDI: ffff88013fd0dd58
Feb 24 01:07:37 debian-netbook kernel: [  289.302020] RBP: ffff8800b25dc000 R08: 000000000000000a R09: 0000000000000703
Feb 24 01:07:37 debian-netbook kernel: [  289.304631] R10: ffff880036982220 R11: 0000000000000703 R12: ffff8800b25dc850
Feb 24 01:07:37 debian-netbook kernel: [  289.307227] R13: 0000000000005078 R14: ffff88009a2d37e0 R15: ffff88009330ac00
Feb 24 01:07:37 debian-netbook kernel: [  289.309826] FS:  00007f7f20b23940(0000) GS:ffff88013fd00000(0000) knlGS:0000000000000000
Feb 24 01:07:37 debian-netbook kernel: [  289.312441] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 24 01:07:37 debian-netbook kernel: [  289.315031] CR2: 000000000013a355 CR3: 00000001381d9000 CR4: 00000000001006e0
Feb 24 01:07:37 debian-netbook kernel: [  289.317617] Stack:
Feb 24 01:07:37 debian-netbook kernel: [  289.320188]  ffffffff810a6505 0000000000000000 ffffffff810a1bed 0000000000001247
Feb 24 01:07:37 debian-netbook kernel: [  289.322808]  ffffffff00000002 00000000ddf6f4db ffff8800b25dc850 00000000ddf6f4db
Feb 24 01:07:37 debian-netbook kernel: [  289.325385]  ffffffffa046b8c5 0000000000000009 ffff8800b25dc898 ffff8800b25dc850
Feb 24 01:07:37 debian-netbook kernel: [  289.327943] Call Trace:
Feb 24 01:07:37 debian-netbook kernel: [  289.330487]  [<ffffffff810a6505>] ? enqueue_entity+0x445/0x9f0
Feb 24 01:07:37 debian-netbook kernel: [  289.333039]  [<ffffffff810a1bed>] ? select_idle_sibling+0xad/0x100
Feb 24 01:07:37 debian-netbook kernel: [  289.335569]  [<ffffffffa046b8c5>] ? cfg80211_disconnect+0x185/0x1a0 [cfg80211]
Feb 24 01:07:37 debian-netbook kernel: [  289.338084]  [<ffffffffa0441eb6>] ? cfg80211_leave+0x26/0x40 [cfg80211]
Feb 24 01:07:37 debian-netbook kernel: [  289.340601]  [<ffffffffa10aaedb>] ? RTMP_COM_IoctlHandle+0x69b/0x760 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [  289.343121]  [<ffffffff8109a1fb>] ? check_preempt_curr+0x7b/0x90
Feb 24 01:07:37 debian-netbook kernel: [  289.345631]  [<ffffffffa10dc10f>] ? CFG80211NetdevNotifierEvent+0xff/0x190 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [  289.348109]  [<ffffffff81091215>] ? notifier_call_chain+0x45/0x70
Feb 24 01:07:37 debian-netbook kernel: [  289.350551]  [<ffffffff81489db5>] ? __dev_close_many+0x55/0x100
Feb 24 01:07:37 debian-netbook kernel: [  289.352992]  [<ffffffff81489fba>] ? __dev_close+0x3a/0x70
Feb 24 01:07:37 debian-netbook kernel: [  289.355402]  [<ffffffff814928a6>] ? __dev_change_flags+0x96/0x150
Feb 24 01:07:37 debian-netbook kernel: [  289.357778]  [<ffffffff81492983>] ? dev_change_flags+0x23/0x60
Feb 24 01:07:37 debian-netbook kernel: [  289.360119]  [<ffffffff8149fc3b>] ? do_setlink+0x5ab/0xae0
Feb 24 01:07:37 debian-netbook kernel: [  289.362456]  [<ffffffff814a085a>] ? rtnl_newlink+0x5aa/0x8a0
Feb 24 01:07:37 debian-netbook kernel: [  289.364791]  [<ffffffff81261291>] ? security_capable+0x41/0x60
Feb 24 01:07:37 debian-netbook kernel: [  289.367093]  [<ffffffff8149efb1>] ? rtnetlink_rcv_msg+0xe1/0x240
Feb 24 01:07:37 debian-netbook kernel: [  289.369355]  [<ffffffff8147afee>] ? __kmalloc_reserve.isra.32+0x2e/0x80
Feb 24 01:07:37 debian-netbook kernel: [  289.371608]  [<ffffffff8149eed0>] ? rtnetlink_rcv+0x30/0x30
Feb 24 01:07:37 debian-netbook kernel: [  289.373837]  [<ffffffff814c1001>] ? netlink_rcv_skb+0xa1/0xc0
Feb 24 01:07:37 debian-netbook kernel: [  289.376033]  [<ffffffff8149eec4>] ? rtnetlink_rcv+0x24/0x30
Feb 24 01:07:37 debian-netbook kernel: [  289.378200]  [<ffffffff814c07b1>] ? netlink_unicast+0x121/0x1a0
Feb 24 01:07:37 debian-netbook kernel: [  289.380378]  [<ffffffff814c0cd1>] ? netlink_sendmsg+0x4a1/0x600
Feb 24 01:07:37 debian-netbook kernel: [  289.382475]  [<ffffffff811ce478>] ? rw_copy_check_uvector+0x68/0x110
Feb 24 01:07:37 debian-netbook kernel: [  289.384500]  [<ffffffff814741e0>] ? sock_sendmsg+0x30/0x40
Feb 24 01:07:37 debian-netbook kernel: [  289.386445]  [<ffffffff81474ae1>] ? ___sys_sendmsg+0x281/0x290
Feb 24 01:07:37 debian-netbook kernel: [  289.388316]  [<ffffffff810e7573>] ? futex_wake+0x93/0x170
Feb 24 01:07:37 debian-netbook kernel: [  289.390106]  [<ffffffff813fbd5d>] ? uevent_show+0xdd/0x100
Feb 24 01:07:37 debian-netbook kernel: [  289.391818]  [<ffffffff814754fe>] ? __sys_sendmsg+0x4e/0x90
Feb 24 01:07:37 debian-netbook kernel: [  289.393452]  [<ffffffff81586432>] ? system_call_fast_compare_end+0xc/0x67
Feb 24 01:07:37 debian-netbook kernel: [  289.395024] Code: f9 df 8b 0f 41 89 4d 00 89 d1 8b 74 0f fc 41 89 74 0d fc e9 be fd ff ff 48 c7 c6 80 4e 10 a1 48 c7 c7 a0 f8 11 a1 e8 22 d7 07 e0 <80> bb 51 a3 13 00 01 0f 85 a4 f0 ff ff 83 3d fb 6e 06 00 01 0f 
Feb 24 01:07:37 debian-netbook kernel: [  289.398444] RIP  [<ffffffffa10e0551>] CFG80211DRV_IoctlHandle+0x10c1/0x13b0 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [  289.400116]  RSP <ffff88009a2d3688>
Feb 24 01:07:37 debian-netbook kernel: [  289.401715] CR2: 000000000013a355
Feb 24 01:07:37 debian-netbook kernel: [  289.413161] ---[ end trace b837acfcb5aa965c ]---`

paolomeraviglia, 2016-02-24:
Here the only warnings:
`  CC [M]  /home/paolo/Scaricati/MT/4.2.1/os/linux/../../common/cmm_data.o
/home/paolo/Scaricati/MT/4.2.1/os/linux/../../common/cmm_data.c: In function ‘TxPktClassification’:
/home/paolo/Scaricati/MT/4.2.1/os/linux/../../common/cmm_data.c:1258:4: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]`
`    && (TxFrameType == TX_LEGACY_FRAME)`
`    ^`

jurobystricky, 2016-02-24:
Thanks for taking the time to post the log. I'll need some time to digest it. There seem to be several issues (maybe the same one) with EDUP-1601, I'm waiting for one of those dongles myself. Once I get it, it should be fairly straightforward to determine what's going on.

paolomeraviglia, 2016-02-24:
If you need help I'll try to help you: As of now reading the KP log, I can see:

[ 289.398444] RIP [<ffffffffa10e0551>] CFG80211DRV_IoctlHandle+0x10c1/0x13b0 [mt7662u_sta]
Feb 24 01:07:37 debian-netbook kernel: [ 289.400116] RSP <ffff88009a2d3688>

Maybe could be sometring wrong in CFG80211DRV_IoctlHandle function?

Yesterday I tried with a ubuntu live cd with kernel 4.2.0: same issues, same KP.

paolomeraviglia, 2016-02-24:
with kernel 3.16.0.4 I have the same issue

paolomeraviglia, 2016-02-25:
maybe helpful:

with gdb, listing the function CFG80211DRV_IoctlHandle+0x10c1/0x13b0, I found:

`0x7c4c0 is in CFG80211DRV_IoctlHandle` `(/home/paolo/Scaricati/MT/mt7612u/4.2.1/os/linux/../../os/linux/cfg80211/cfg80211drv.c:57).`
`52	#endif /*CONFIG_STA_SUPPORT*/
`
`53	
`
`54	extern struct notifier_block cfg80211_netdev_notifier;
`
`55	
`
`56	int CFG80211DRV_IoctlHandle(void *pAdSrc, int cmd, void *pData, ULONG Data)
`
`57	{
`
`58		PRTMP_ADAPTER pAd = (PRTMP_ADAPTER)pAdSrc;
`
`59	
`
`60		switch (cmd) {
`
`61		case CMD_RTPRIV_IOCTL_80211_START:
`

jurobystricky, 2016-02-25:
Thanks, CFG80211DRV_IoctlHandle is a rather misnamed function. The BUG: unable to handle kernel paging request at 000000000013a355 happens in CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT, the code there is supposed to stop scanning if scanning in progress. This routine is called from network event notifier, and most likely got it pointers 
mangled. Try to simply comment out (and rebuild the driver) the line 
RTMP_DRIVER_80211_NETDEV_EVENT(pAd, ndev, state);
in the routine CFG80211NetdevNotifierEvent.

paolomeraviglia, 2016-02-25:
Row 2573 of cfg80211.c right?

paolomeraviglia, 2016-02-25:
Seems to be ok now, but i cannot connect to any network

paolomeraviglia, 2016-02-26:
here a cat of my syslog cutted and filtered:

http://pastebin.com/LgNWSgnb

ignore lines from 166 to 207

I'm sorry but I cannot filter lines from other wifi dongles inside my pc like that:

Feb 26 04:24:40 debian-netbook dhclient[7643]: DHCPACK of 192.168.1.134 from 192.168.1.254
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    address 192.168.1.134
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    plen 24 (255.255.255.0)
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    gateway 192.168.1.254
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    server identifier 192.168.1.254
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    lease time 1800
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    hostname 'debian-netbook'
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    nameserver '62.101.93.101'
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    nameserver '83.103.25.250'
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    domain name 'fastwebnet.it'
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>    wins '192.168.1.254'
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>  NetworkManager state is now CONNECTED_LOCAL
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>  NetworkManager state is now CONNECTED_GLOBAL
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>  dns-mgr: Writing DNS information to /sbin/resolvconf
Feb 26 04:24:40 debian-netbook dhclient[7643]: bound to 192.168.1.134 -- renewal in 678 seconds.
Feb 26 04:24:40 debian-netbook NetworkManager[7586]: <info>  startup complete

But as you can see kernel panic are still there:

Feb 26 04:24:41 debian-netbook kernel: [ 2712.166877] WARNING: CPU: 0 PID: 7365 at /build/linux-19r8NQ/linux-4.3.5/net/wireless/sme.c:714 __cfg80211_connect_result+0x3$
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166975] CPU: 0 PID: 7365 Comm: kworker/u4:0 Tainted: P        W  O    4.3.0-1-amd64 #1 Debian 4.3.5-1
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166977] Hardware name: ASUSTeK COMPUTER INC. X200MA/X200MA, BIOS X200MA.504 11/13/2014
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166989] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166991]  0000000000000000 00000000c8248438 ffffffff812ddf19 0000000000000000
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166995]  ffffffff81072afd ffff88008233b800 ffff88007686bda8 0000000000000000
Feb 26 04:24:41 debian-netbook kernel: [ 2712.166999]  0000000000000000 ffff88012ce7c000 ffffffffa04f4936 0000000000000002
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167002] Call Trace:
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167011]  [<ffffffff812ddf19>] ? dump_stack+0x40/0x57
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167015]  [<ffffffff81072afd>] ? warn_slowpath_common+0x7d/0xb0
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167029]  [<ffffffffa04f4936>] ? __cfg80211_connect_result+0x3b6/0x420 [cfg80211]
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167042]  [<ffffffffa04cf006>] ? cfg80211_process_wdev_events+0x116/0x190 [cfg80211]
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167053]  [<ffffffffa04cf0ac>] ? cfg80211_process_rdev_events+0x2c/0x60 [cfg80211]
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167064]  [<ffffffffa04ca01a>] ? cfg80211_event_work+0x1a/0x20 [cfg80211]
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167067]  [<ffffffff8108a3ff>] ? process_one_work+0x19f/0x3d0
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167070]  [<ffffffff8108a67d>] ? worker_thread+0x4d/0x450
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167073]  [<ffffffff8108a630>] ? process_one_work+0x3d0/0x3d0
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167077]  [<ffffffff810904ad>] ? kthread+0xcd/0xf0
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167080]  [<ffffffff810903e0>] ? kthread_create_on_node+0x190/0x190
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167084]  [<ffffffff815867cf>] ? ret_from_fork+0x3f/0x70
Feb 26 04:24:41 debian-netbook kernel: [ 2712.167088]  [<ffffffff810903e0>] ? kthread_create_on_node+0x190/0x190

jurobystricky, 2016-02-26:
Not really kernel panic, but a warning from http://lxr.free-electrons.com/source/net/wireless/sme.c#L714
definitely something to look into, but I don't think that is the culprit. There are additional messages in the log that deserve attention.

paolomeraviglia, 2016-02-29:
I found that, if I do iwlist wlan0 scanning, the result is really strange:

in front of my router I obtain a  60/100 quality signal for 5 ghz network and 10/100 quality signal for 2.4 network.

In all cases I cannot connect to any network. May I help you in some way?

jurobystricky, 2016-02-29:
Well, yesterday I installed debian 8.3 (AMD64) and rebuilt the driver (Netgear dongle). A bit to my surprise, it worked quite well. I could also unplug/replug the dongle and nothing broke, did not need to restart the network manager. I did, however, get some new compiler warnings which I need to look into.
As for the signal quality, I would take it with a grain of salt.

madmalkav, 2016-02-29:
As I commented on another issue, with the EDUP dongle the kernel panics
persists and there are some cuts, bad reception and low bitrate that makes
me think perhaps the EDUP dongle have some differences with the Netgear
one, at least on the antennae part.

Is there anything we can do to help you discover the differences between
both devices?

jurobystricky, 2016-02-29:
My own EDUP should arrive today (based on tracking #). That will hopefully reveal differences between the two dongles.

jurobystricky, 2016-03-02:
EDUP arrived. I can use it, but it looses connection frequently. But now I can at least investigate the problem using may machine.

madmalkav, 2016-03-02:
Apart of the connection stability, isn't the speed somewhat lower than
expected?

madmalkav, 2016-03-07:
I haven't tested it deeply, but I left a ping working and forgot about it and I had no more disconnects for a good hour. Two minutes after I stopped it I got a disconnect. I have been running a ping for another hour and no more disconnects for the moment.

paolomeraviglia, 2016-03-07:
Tomorrow i'll try it: I downgraded to 3.16 for stability issues

paolomeraviglia, 2016-04-01:
Hi,

Sorry for my silence
I tried on 5 different pc with linux debian 8.3.0 and kernel 3.16.0.4 with no success: the dongle find with terrible signal networks around me, but cannot connect.

This is the compilation log

http://pastebin.com/X5fCtpKE

this is my uname -a
`Linux debian-netbook 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt20-1+deb8u4 (2016-02-29) x86_64 GNU/Linux
`

May I help you now that I have a little bit of time?

Thanks

madmalkav, 2016-04-01:
Can you ask you why you still use kernel 3.16 ?

paolomeraviglia, 2016-04-01:
because i'm using standard debian stable 8.3.0

paolomeraviglia, 2016-04-01:
i'm using it because of debian 8.3.0 stable.

other info from kernel:

Apr  2 01:22:58 debian-netbook kernel: [  296.502138] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...

paolomeraviglia, 2016-04-01:
uhm: some strange errors:

`
Apr  2 01:36:01 debian-netbook kernel: [ 1079.766826] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
`

?

`
Apr  2 01:35:55 debian-netbook kernel: [ 1073.496427] cfg80211: Regulatory domain changed to country: DE `

(why not IT?)

`
Apr  2 01:33:24 debian-netbook NetworkManager[474]: <warn> (wlan1): failed to reset MAC address to 00:00:00:00:00:00 
`
uhm... strange MAC address...

`
Apr  2 01:33:24 debian-netbook NetworkManager[474]: <warn> Activation (wlan1/wireless): association took too long.
`

and finally kernel panic:

http://pastebin.com/K2PU1Mhy

paolomeraviglia, 2016-04-01:
mmm other message:
` Apr  2 01:46:26 debian-netbook kernel: [ 1705.090707] /home/paolo/git/Netgear-A6210/os/linux/../../sta/rtmp_data.c:413 assert pRxBlk->wcid == BSSID_WCIDfailed `

look at the directory: why it's using local path? I've already done make install!

jurobystricky, 2016-04-06:
Regarding your observations: The kernel message you see is not a panic, but a warning. Not that we want to see that either, but not nearly as severe.
As for the assert message, that is how assert is implemented, it embeds the file name/location into the binary image during compilation. It embeds an absolute path file name. It has nothing to do with installation. (Note that in the kernel warning you also an absolute path name based on the build system pf the kernel as well)
Having said that, I just pushed some changes which make my EDUP EP-AC1601 much more stable.
Please try them and let me know if they worked for you.

veaa, 2016-04-06:
Hi,

I’m still having the same problem with my EDUP EP-AC1601 and Kali:

[  142.264787] andes_usb_load_rom_patch: upload fw timeout
[  142.264794] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[  142.264912] receive cmd msg fail(-2)
[  142.264971] tx_kickout_fail_count = 0
[  142.264973] tx_timeout_fail_count = 0
[  142.264974] rx_receive_fail_count = 0
[  142.264975] alloc_cmd_msg = 1
[  142.264976] free_cmd_msg = 1
[  142.264978] !!! rt28xx init fail !!!
[  142.268032] -->RTUSBVendorReset
[  142.288266] <--RTUSBVendorReset
[  144.289112] andes_usb_load_rom_patch: upload fw timeout
[  144.289119] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[  144.289242] receive cmd msg fail(-2)
[  144.289304] tx_kickout_fail_count = 0
[  144.289307] tx_timeout_fail_count = 0
[  144.289308] rx_receive_fail_count = 0
[  144.289309] alloc_cmd_msg = 1
[  144.289310] free_cmd_msg = 1
[  144.289312] !!! rt28xx init fail !!!
[  144.291516] -->RTUSBVendorReset
[  144.311749] <--RTUSBVendorReset

madmalkav, 2016-04-06:
git pull; make clean; make; sudo make install and then reboot should be enough to install the new version? If so, I haven't got any improvement, i.e. I still need to run a ping in the background to avoid connection losts.

jurobystricky, 2016-04-06:
@madmalkav : Can you please post the full log?

madmalkav, 2016-04-06:
Please tell me what exactly you want me to log and I will upload tomorrow.

jurobystricky, 2016-04-06:
@madmalkav : Kernel log (dmesg), from the moment you boot (with the dongle plugged in) until a disconnect.

madmalkav, 2016-04-06:
Ok, I've got out of the ciry some hours ago, tomorrow when I get back will
send the log.

jonipalosaari, 2016-04-07:
@madmalkav @jurobystricky what connection manager do you use? 

maybe a long shot and not related at all, but somehow timeouts went away on my WNDA3100v3&gentoo after I moved openrc to systemd and **networkmanager to systemd-networkd**. Now it has been rock solid for weeks.

madmalkav, 2016-04-07:
http://pastebin.com/UWL96V2t

Here you have a dmesg. Probably lots of garbage as I kept using my PC normally to not modify the test conditions. You can see the device failing to start, me restarting my usb hub -more confortable that unplugging and replugging- device connecting and losing connection after about 5 minutes. Anyway, it was far quicker before the last commit so the situation have improved a little.

If I run a ping connection is far more stable, there are some connection losts but they are rare. I lose aprox. 10% of the packets.

jurobystricky, 2016-04-07:
@madmalkav : Thanks, some valuable errors logged. Will sit down and digest the log in more detail.

jurobystricky, 2016-04-08:
So I think can explain the ping: The driver drops connection (LinkDown) if there is no beacon for 4 seconds, yet the "keep alive" packets are sent only each 10 seconds. So ping serves as a better source of "keep alive". There is some convoluted logic involved, but if you change in file common/mlme.c the line (around 1950) :

if ((pAd->Mlme.OneSecPeriodicRound % 10) == 8) {

to 

if ((pAd->Mlme.OneSecPeriodicRound % 4) == 0) {

you should not need to run ping to avoid connection losses.

madmalkav, 2016-04-08:
I haven't been able to test your suggested fix for a long time but it seems to fix the connection lost issues. Now we only need to figure the packet lost and some really high latencies when the link is moving heavy amounts of data and we will be golden :)

summertm, 2016-05-31:
This is so close, but so far away.  
I bought this device, Edup 1605, a week ago for my Kuibuntu 16.04- kernel 4.4 setup and was very impressed when the driver built and installed without a hitch.  However, the scan request only works sometimes and never holds long enough to make an actual connection.  The signal strength is shown as nil, even though it is actually about 70%. when I use the device with Win7 loaded on the same machine.  
I would love to get this device working, but I shall have to send it back soon if no solution is found.  I am happy to test changes, but need to get it working in the next week, so I can still send it back if necessary.
I am not sure whether it is significant, but my device id is 761b rather than 7610 or 7612 reported elsewhere.
The speed achieved by the device in Windows is within a couple of Mbs of that achieved via a wired connection (32Mbs v 37Mbs) on a max of 38Mbs, so the potential is there.
I am wondering whether I should swap it for the Netgear A6210, or might this give similar problems.
Please help!

jurobystricky, 2016-05-31:
Please try the quick fix mentioned above :

if ((pAd->Mlme.OneSecPeriodicRound % 10) == 8)...

 to avoid connection losses. 
I can say from my own experience that Netgear A6210 is quite a bit more reliable. I get over 100mbs reading/writing large files Google drive.

summertm, 2016-06-01:
Thanks for the very quick reply.  I tried the fix, but to no avail.
I think I will go for the swap.
You are doing a brilliant job, and having searched around for some while 
for a 1200ac adapter for Linux, it looks like the combination you have 
provided offers the best bet at the current time.  So please stick at it 
for the sake of all the people like myself who buy the EDUP from ebay in 
the expectation of the cd providing the solution.
Regards
Mike Summerton

summertm, 2016-06-05:
I swapped the devices and now the 2.4Ghz band works OK but the 5GHz band never makes a proper connection.  Does anyone have any idea of the most likely cause?  Again, the device works Ok with the Windows driver and so the problem is in the driver/PC configuration.

madmalkav, 2016-06-05:
It works for me on 5Ghz with EDUP 1601, with some packet loses -as in
2.4Ghz-. Can you upload your dmesg?

summertm, 2016-06-06:
It's kind of you to look into this madmalkav.  The installation is directly using steals PPA dkms config, without any changes to the config.
The log showing an attempt to connect to 5GHz wifi is shown below.
06/06/2016 08:25		CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
06/06/2016 08:25		andes_usb_erasefw
06/06/2016 08:25		==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
06/06/2016 08:25		receive cmd msg fail(-2)
06/06/2016 08:25		tx_kickout_fail_count = 0
06/06/2016 08:25		tx_timeout_fail_count = 0
06/06/2016 08:25		rx_receive_fail_count = 0
06/06/2016 08:25		alloc_cmd_msg = 2452
06/06/2016 08:25		free_cmd_msg = 2452
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe4b0!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe528!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe5a0!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe438!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe2d0!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019fe348!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc9000198ffa0!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc9000197ed78!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc9000197edf8!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc90001990128!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc9000198feb0!
06/06/2016 08:25		RTMP_TimerListRelease: release timer obj ffffc900019900b0!
06/06/2016 08:25		==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 0, pAd->WlanFunCtrl:0x208, Reg-WlanFunCtrl=0x209
06/06/2016 08:25		-->RTUSBVendorReset
06/06/2016 08:25		<--RTUSBVendorReset
06/06/2016 08:25		...........andes_usb_chk_crc
06/06/2016 08:25		andes_usb_reset_wmt
06/06/2016 08:25		-->RTUSBVendorReset
06/06/2016 08:25		<--RTUSBVendorReset
06/06/2016 08:25	fw version	.0.00 build:1
06/06/2016 08:25	build time	01406241830____
06/06/2016 08:25		fw for E3 IC
06/06/2016 08:25		ilm length = 59312(bytes)
06/06/2016 08:25		dlm length = 32068(bytes)
06/06/2016 08:25		loading fw........
06/06/2016 08:25		cfg_mode=5
06/06/2016 08:25		Key1Str is Invalid key length(0) or Type(0)
06/06/2016 08:25		Key2Str is Invalid key length(0) or Type(0)
06/06/2016 08:25		Key3Str is Invalid key length(0) or Type(0)
06/06/2016 08:25		Key4Str is Invalid key length(0) or Type(0)
06/06/2016 08:25		USBAggregation = 1
06/06/2016 08:25		1. Phy Mode = 31
06/06/2016 08:25		NVM is Efuse and its size =1d[1e0-1fc]
06/06/2016 08:25	get_chl_grp	llegal channel (167)
06/06/2016 08:25	get_chl_grp	llegal channel (167)
06/06/2016 08:25	get_chl_grp	llegal channel (169)
06/06/2016 08:25	get_chl_grp	llegal channel (169)
06/06/2016 08:25	get_chl_grp	llegal channel (171)
06/06/2016 08:25	get_chl_grp	llegal channel (171)
06/06/2016 08:25	get_chl_grp	llegal channel (173)
06/06/2016 08:25	get_chl_grp	llegal channel (173)
06/06/2016 08:25		Country Region from e2p = ffff
06/06/2016 08:25		mt76x2_get_external_lna_gain::LNA type=0x0, BLNAGain=0xffffff8d, ALNAGain0=0xffffff89, ALNAGain1=0xffffff88, ALNAGain2=0xffffff87
06/06/2016 08:25		2. Phy Mode = 31
06/06/2016 08:25		3. Phy Mode = 31
06/06/2016 08:25		andes_usb_fw_init
06/06/2016 08:25	AntCfgInit	primary/secondary ant 0/1
06/06/2016 08:25	andes_load_cr	r_type(2)
06/06/2016 08:25	ChipStructAssign()	MT76x2 hook !
06/06/2016 08:25		---> InitFrequencyCalibration
06/06/2016 08:25		InitFrequencyCalibrationMode:Unknow mode = 3
06/06/2016 08:25		InitFrequencyCalibration: frequency offset in the EEPROM = 168
06/06/2016 08:25		<--- InitFrequencyCalibration
06/06/2016 08:25	RTMPSetPhyMode	channel is out of range, use first channel=1
06/06/2016 08:25	RTMPSetPhyMode	Update for STA
06/06/2016 08:25		MCS Set = ff ff 00 00 01
06/06/2016 08:25		80211> re-init bands...
06/06/2016 08:25		80211> CurTxPower = 20 dBm
06/06/2016 08:25		====> Radar Channel 52
[... 480 lines cut ...]
06/06/2016 08:26		80211> Connect bssid a8:d3:f7:96:ae:69
06/06/2016 08:26		PeerBeaconAtJoinAction(): HT-CtrlChannel=6, CentralChannel=>6
06/06/2016 08:26		PeerBeaconAtJoinAction(): Set CentralChannel=6
06/06/2016 08:26		AdjustChannelRelatedValue(): Input BW=0, rf_channel=6, vht_bw=1, Channel=6, vht_cent_ch=6!
06/06/2016 08:26		/var/lib/dkms/netgear-a6210/3.0.0.1/build/os/linux/../../sta/rtmp_data.c:413 assert pRxBlk->wcid == BSSID_WCIDfailed
06/06/2016 08:26	[255] 1	a0:63:91:bb:d1:01, 2: a8:d3:f7:96:ae:69, 3:a8:d3:f7:96:ae:69
06/06/2016 08:26		mt76x2_reinit_agc_gain:original agc_vga0 = 0x48, agc_vga1 = 0x48
06/06/2016 08:26		mt76x2_reinit_agc_gain:updated agc_vga0 = 0x48, agc_vga1 = 0x48
06/06/2016 08:26		mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x34, hi_lna1 = 0x34
06/06/2016 08:26		mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x34, hi_lna1 = 0x34
06/06/2016 08:26		original vga value(chain0) = 48
06/06/2016 08:26		original vga value(chain1) = 48
06/06/2016 08:26		mt76x2_calibration(channel = 6)
06/06/2016 08:26	cfg80211	Regulatory domain changed to country: GB
06/06/2016 08:26	cfg80211	DFS Master region: ETSI
06/06/2016 08:26	cfg80211	(start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
06/06/2016 08:26	cfg80211	(2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
06/06/2016 08:26	cfg80211	(5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
06/06/2016 08:26	cfg80211	(5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
06/06/2016 08:26	cfg80211	(5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
06/06/2016 08:26	cfg80211	(57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
06/06/2016 08:26		Rcv Wcid(1) AddBAReq
06/06/2016 08:26		Start Seq = 00000003
06/06/2016 08:26		80211> bss inform fail ==> 12
06/06/2016 08:29		80211> bss inform fail ==> 12

madmalkav, 2016-06-06:
I don't see anything that rings a bell. Let's wait for the developer to
look at it.

shingeri, 2016-09-29:
It has been awefully quite..since june. Did something change? I just purchased the EDUP EP-AC1605. Been add it since last night, but i can't get it even to scan. Compiled different drivers, but still no succes.

Anyone?

shingeri, 2016-10-11:
lsusb:
```
shinger@K-Laptop:~/Netgear-A6210$ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0bda:0811 Realtek Semiconductor Corp. 
Bus 003 Device 004: ID 8087:07dc Intel Corp. 
Bus 003 Device 003: ID 174f:14b8 Syntek 
Bus 003 Device 007: ID 0e8d:761b MediaTek Inc. 
Bus 003 Device 002: ID 046d:c050 Logitech, Inc. RX 250 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Added the device and vender etc.

`#ifdef MT76x2
        {USB_DEVICE(0x0846, 0x9014), .driver_info = RLT_MAC_BASE}, /* Netgear WNDA3100v3 */
        {USB_DEVICE(0x0B05, 0x180B), .driver_info = RLT_MAC_BASE}, /* ASUS USB-N53 */
        {USB_DEVICE(0x0846, 0x9053), .driver_info = RLT_MAC_BASE}, /* MT7612U, Netgear A6210 */
        {USB_DEVICE(0x0B05, 0x17EB), .driver_info = RLT_MAC_BASE}, /* ASUS USB-AC55 */
        {USB_DEVICE(0x045e, 0x02e6), .driver_info = RLT_MAC_BASE}, /* Microsoft XBox One Wireless Adapter */
        {USB_DEVICE(0x0E8D, 0x7612), .driver_info = RLT_MAC_BASE},
        {USB_DEVICE(0x0E8D, 0x761b), .driver_info = RLT_MAC_BASE}, /*EDUP AC1605*/
        {USB_DEVICE_AND_INTERFACE_INFO(0x0E8D, 0x7632, 0xff, 0xff, 0xff), .driver_info = RLT_MAC_BASE},
        {USB_DEVICE_AND_INTERFACE_INFO(0x0E8D, 0x7662, 0xff, 0xff, 0xff), .driver_info = RLT_MAC_BASE},
#endif`

I thought to be on the safe side, i also installed the latest linux kernel 4.8.1

uname -r output:
`shinger@K-Laptop:~/Netgear-A6210$ uname -r
4.8.1-040801-generic`

Linux headers already installed output:
`shinger@K-Laptop:~/Netgear-A6210$ sudo apt install linux-headers-$(uname -r)
[sudo] password for shinger: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-4.8.1-040801-generic is already the newest version (4.8.1-040801.201610071031).
The following packages were automatically installed and are no longer required:
  libechonest2.3 libopenshot8 libqt5script5 libreoffice-gtk linux-image-extra-4.4.0-31-generic
  linux-image-extra-4.4.0-34-generic linux-image-extra-4.4.0-36-generic linux-image-extra-4.4.0-38-generic
  linux-tools-4.4.0-31 linux-tools-4.4.0-31-generic linux-tools-4.4.0-34 linux-tools-4.4.0-34-generic
  linux-tools-4.4.0-36 linux-tools-4.4.0-36-generic linux-tools-4.4.0-38 linux-tools-4.4.0-38-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Trying to compile the driver:

```
shinger@K-Laptop:~/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/shinger/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.8.1-040801-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/shinger/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.8.1-040801-generic'
  CC [M]  /home/shinger/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from /home/shinger/Netgear-A6210/include/os/rt_linux.h:77:0,
                 from /home/shinger/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/shinger/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/shinger/Netgear-A6210/include/rt_config.h:34,
                 from /home/shinger/Netgear-A6210/os/linux/../../sta/assoc.c:28:
/home/shinger/Netgear-A6210/include/cfg80211.h:38:49: error: ‘IEEE80211_NUM_BANDS’ undeclared here (not in a function)
  struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
                                                 ^
scripts/Makefile.build:289: recipe for target '/home/shinger/Netgear-A6210/os/linux/../../sta/assoc.o' failed
make[2]: *** [/home/shinger/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
Makefile:1477: recipe for target '_module_/home/shinger/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/shinger/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.8.1-040801-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
```

So what am i doing wrong here?

paolomeraviglia, 2017-02-08:
with debian 8.3 (kernel 3.16.04) it continues to do random kernel panic

paolomeraviglia, 2017-02-08:
Here is the strace

`Feb  8 17:38:11 debian-netbook kernel: [   16.885804] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
Feb  8 17:38:11 debian-netbook kernel: [   16.885840] BUG: unable to handle kernel paging request at 000000000013a355
Feb  8 17:38:11 debian-netbook kernel: [   16.885910] IP: [<ffffffffa09b9824>] CFG80211DRV_IoctlHandle+0x1364/0x14f0 [mt7662u_sta]
Feb  8 17:38:11 debian-netbook kernel: [   16.885993] PGD a06a5067 PUD a06a4067 PMD 0 
Feb  8 17:38:11 debian-netbook kernel: [   16.886034] Oops: 0000 [#1] SMP 
Feb  8 17:38:11 debian-netbook kernel: [   16.886065] Modules linked in: rfcomm mt7662u_sta(O) binfmt_misc bnep nfsd auth_rpcgss oid_registry nfs_acl nfs lockd fscache sunrpc arc4 ecb hid_generic nls_utf8 usbhid btusb nls_cp437 bluetooth 6lowpan_iphc vfat fat iTCO_wdt asus_nb_wmi iTCO_vendor_support asus_wmi sparse_keymap iwlmvm intel_powerclamp intel_rapl coretemp snd_hda_codec_hdmi kvm_intel mac80211 kvm efi_pstore crc32_pclmul snd_hda_codec_realtek snd_hda_codec_generic cryptd uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev media pcspkr iwlwifi joydev evdev rtsx_pci_ms serio_raw efivars snd_hda_intel cfg80211 memstick rfkill snd_hda_controller i915 i2c_i801 lpc_ich snd_hda_codec snd_hwdep shpchp drm_kms_helper drm iosf_mbi i2c_algo_bit wmi snd_soc_rt5640 snd_soc_rl6231 snd_soc_core snd_compress battery video ac snd_pcm snd_timer snd soundcore regmap_i2c snd_soc_sst_acpi i2c_designware_platform i2c_designware_core button processor fuse parport_pc ppdev lp parport autofs4 ext4 crc16 mbcache jbd2 sg sd_mod crc_t10dif crct10dif_generic rtsx_pci_sdmmc crct10dif_pclmul crct10dif_common crc32c_intel psmouse r8169 mii rtsx_pci mfd_core thermal thermal_sys ahci libahci libata xhci_hcd sdhci_acpi sdhci i2c_hid hid i2c_core scsi_mod mmc_core usbcore usb_common
Feb  8 17:38:11 debian-netbook kernel: [   16.887168] CPU: 0 PID: 1161 Comm: ifconfig Tainted: G           O  3.16.0-4-amd64 #1 Debian 3.16.39-1
Feb  8 17:38:11 debian-netbook kernel: [   16.887241] Hardware name: ASUSTeK COMPUTER INC. X200MA/X200MA, BIOS X200MA.504 11/13/2014
Feb  8 17:38:11 debian-netbook kernel: [   16.887306] task: ffff8800a076e9a0 ti: ffff8800a07b0000 task.ti: ffff8800a07b0000
Feb  8 17:38:11 debian-netbook kernel: [   16.887365] RIP: 0010:[<ffffffffa09b9824>]  [<ffffffffa09b9824>] CFG80211DRV_IoctlHandle+0x1364/0x14f0 [mt7662u_sta]
Feb  8 17:38:11 debian-netbook kernel: [   16.887460] RSP: 0018:ffff8800a07b3c40  EFLAGS: 00010292
Feb  8 17:38:11 debian-netbook kernel: [   16.887503] RAX: 000000000000003c RBX: 0000000000000004 RCX: 0000000000000000
Feb  8 17:38:11 debian-netbook kernel: [   16.887559] RDX: ffff88013fc0ede0 RSI: ffff88013fc0d4b8 RDI: 0000000000000246
Feb  8 17:38:11 debian-netbook kernel: [   16.887615] RBP: ffff880139a80000 R08: 000000000000000a R09: 0000000000000354
Feb  8 17:38:11 debian-netbook kernel: [   16.887671] R10: 0000000000000000 R11: ffff8800a07b397e R12: ffff880139a808d0
Feb  8 17:38:11 debian-netbook kernel: [   16.887727] R13: 0000000000005078 R14: ffff8800a07b3d88 R15: 0000000000000000
Feb  8 17:38:11 debian-netbook kernel: [   16.887784] FS:  00007fe875bbb700(0000) GS:ffff88013fc00000(0000) knlGS:0000000000000000
Feb  8 17:38:11 debian-netbook kernel: [   16.887847] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb  8 17:38:11 debian-netbook kernel: [   16.887893] CR2: 000000000013a355 CR3: 00000000a0768000 CR4: 00000000001007f0
Feb  8 17:38:11 debian-netbook kernel: [   16.887949] Stack:
Feb  8 17:38:11 debian-netbook kernel: [   16.887966]  ffff8800a07b3c38 ffff8800a07b3ce8 ffff880139a808d0 ffff880139a808d0
Feb  8 17:38:11 debian-netbook kernel: [   16.888032]  ffff880139a80000 0000000000000003 ffff880036fc0000 ffffffffa0528c7d
Feb  8 17:38:11 debian-netbook kernel: [   16.888097]  0000000000000009 ffff880139a80918 0000000000000287 ffffffff81413f62
Feb  8 17:38:11 debian-netbook kernel: [   16.888162] Call Trace:
Feb  8 17:38:11 debian-netbook kernel: [   16.888197]  [<ffffffffa0528c7d>] ? cfg80211_disconnect+0x1dd/0x200 [cfg80211]
Feb  8 17:38:11 debian-netbook kernel: [   16.888258]  [<ffffffff81413f62>] ? skb_dequeue+0x52/0x70
Feb  8 17:38:11 debian-netbook kernel: [   16.888304]  [<ffffffff815025fb>] ? wireless_nlevent_flush+0x5b/0xa0
Feb  8 17:38:11 debian-netbook kernel: [   16.888369]  [<ffffffffa0982816>] ? RTMP_COM_IoctlHandle+0x746/0x820 [mt7662u_sta]
Feb  8 17:38:11 debian-netbook kernel: [   16.888440]  [<ffffffffa09b4f5a>] ? CFG80211NetdevNotifierEvent+0x10a/0x1b0 [mt7662u_sta]
Feb  8 17:38:11 debian-netbook kernel: [   16.888507]  [<ffffffff8108d79e>] ? notifier_call_chain+0x4e/0x70
Feb  8 17:38:11 debian-netbook kernel: [   16.888557]  [<ffffffff814202d8>] ? __dev_close_many+0x48/0xe0
Feb  8 17:38:11 debian-netbook kernel: [   16.888604]  [<ffffffff814204bf>] ? __dev_close+0x2f/0x50
Feb  8 17:38:11 debian-netbook kernel: [   16.888649]  [<ffffffff814282a6>] ? __dev_change_flags+0x96/0x160
Feb  8 17:38:11 debian-netbook kernel: [   16.888699]  [<ffffffff81428393>] ? dev_change_flags+0x23/0x60
Feb  8 17:38:11 debian-netbook kernel: [   16.888748]  [<ffffffff81492913>] ? devinet_ioctl+0x6a3/0x7a0
Feb  8 17:38:11 debian-netbook kernel: [   16.888796]  [<ffffffff814094b0>] ? sock_do_ioctl+0x20/0x50
Feb  8 17:38:11 debian-netbook kernel: [   16.888842]  [<ffffffff814099e8>] ? sock_ioctl+0x1e8/0x2c0
Feb  8 17:38:11 debian-netbook kernel: [   16.888888]  [<ffffffff811bceef>] ? do_vfs_ioctl+0x2cf/0x4b0
[... cut ...]
Feb  8 17:38:11 debian-netbook kernel: [   16.889080] Code: 48 c7 c7 66 59 9e a0 31 c0 e8 b3 88 b5 e0 8b 55 04 e9 97 f6 ff ff 31 c0 48 c7 c6 00 f0 9d a0 48 c7 c7 60 96 9f a0 e8 96 88 b5 e0 <80> bb 51 a3 13 00 01 0f 85 17 ee ff ff 83 3d 88 7d 06 00 01 0f 
Feb  8 17:38:11 debian-netbook kernel: [   16.894079] RIP  [<ffffffffa09b9824>] CFG80211DRV_IoctlHandle+0x1364/0x14f0 [mt7662u_sta]
Feb  8 17:38:11 debian-netbook kernel: [   16.896543]  RSP <ffff8800a07b3c40>
Feb  8 17:38:11 debian-netbook kernel: [   16.898986] CR2: 000000000013a355
Feb  8 17:38:11 debian-netbook kernel: [   16.911992] ---[ end trace ee6a9685e58ce8bc ]---
`