jurobystricky/Netgear-A6210 #9: Compilation error with make - cfg80211.c

aaronanglade, 2016-03-09:
Hi,

Make failed 

../os/linux/cfg80211/cfg80211.c: In function CFG80211_OpsStaDel
../os/linux/cfg80211/cfg80211.c: 2112:6 : error pMacAddr undeclared (first use in this function)
     if (pMacAddr == NULL) {

jurobystricky, 2016-03-09:
Hmm, I pushed a fix for this few hours ago. ..
There are various Linux kernel versions involved...
So what is your kernel version?

In any case please change the line 2106 from
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,2,0))
to
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,19,0))

aaronanglade, 2016-03-09:
Thanks for quick reply

My kernel version : 4.1.19-v7+ #852 SMP armv7

aaronanglade, 2016-03-09:
I tested. Compilation is ok. I change only the kernel version. Installation is ok.
Now i configure ;-)

Thanks again

closed 2016-03-17 by aaronanglade