jurobystricky/Netgear-A6210 #10: Compilation error

aaronanglade, 2016-03-13:
Hi,

I try to compile this on Debian 7.9. I have this error. Can you help me ?

export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/drivers/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/3.2.0-4-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/drivers/Netgear-A6210/os/linux modules
make[1]: entrant dans le rÃ©pertoire Â« /usr/src/linux-headers-3.2.0-4-amd64 Â»
  CC [M]  /root/drivers/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from /root/drivers/Netgear-A6210/include/rtmp_os.h:30:0,
                 from /root/drivers/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/drivers/Netgear-A6210/include/rt_config.h:34,
                 from /root/drivers/Netgear-A6210/os/linux/../../sta/assoc.c:28:
/root/drivers/Netgear-A6210/include/os/rt_linux.h:232:2: error: unknown type name âkuid_tâ
/root/drivers/Netgear-A6210/include/os/rt_linux.h:233:2: error: unknown type name âkgid_tâ
make[4]: *** [/root/drivers/Netgear-A6210/os/linux/../../sta/assoc.o] Erreur 1
make[3]: *** [_module_/root/drivers/Netgear-A6210/os/linux] Erreur 2
make[2]: *** [sub-make] Erreur 2
make[1]: *** [all] Erreur 2
make[1]: quittant le rÃ©pertoire Â« /usr/src/linux-headers-3.2.0-4-amd64 Â»
make: *** [debug] Erreur 2

Thanks for your answer

jurobystricky, 2016-03-13:
Please try to modify struct _OS_FS_INFO in the file include/os/rt_linux.h so it looks like this:

    typedef struct _OS_FS_INFO_
    {
    #if LINUX_VERSION_CODE >= KERNEL_VERSION(3,5,7)
	    kuid_t fsuid;
        kgid_t fsgid;
    #else
	    int fsuid;
    	int fsgid;
    #endif
	    mm_segment_t fs;
    } OS_FS_INFO;

I don't have any distro with kernel older than 3.5, so I cannot test the change myself. Please let me know the result.

aaronanglade, 2016-03-13:
Hi,

It's ok for fsuid et fsgid but I have another compilation error

  CC [M]  /root/drivers/Netgear-A6210/os/linux/../../os/linux/rt_profile.o
/root/drivers/Netgear-A6210/os/linux/../../os/linux/rt_profile.c: In function âRTMPReadParametersHookâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/rt_profile.c:186:4: error: implicit declaration of function âfile_inodeâ [-Werror=implicit-function-declaration]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/rt_profile.c:186:35: error: invalid type argument of â->â (have âintâ)
cc1: some warnings being treated as errors
make[4]: *** [/root/drivers/Netgear-A6210/os/linux/../../os/linux/rt_profile.o] Erreur 1
make[3]: *** [_module_/root/drivers/Netgear-A6210/os/linux] Erreur 2
make[2]: *** [sub-make] Erreur 2
make[1]: *** [all] Erreur 2
make[1]: quittant le rÃ©pertoire Â« /usr/src/linux-headers-3.2.0-4-amd64 Â»
make: *** [debug] Erreur 2

Have you an idea ?

jurobystricky, 2016-03-14:
Again, cannot test it but in the file os/linux/rt_profile.c, modify the routine RTMPReadParametersHook (around line 186) to look like this:

    #if LINUX_VERSION_CODE <= KERNEL_VERSION(3,8,0)
    			struct inode *inode;
		    	inode = srcf->f_dentry->d_inode;
			    fsize = (ULONG)inode->i_size;
    #else
			    fsize = (ULONG)file_inode(srcf)->i_size;
    #endif

aaronanglade, 2016-03-15:
It works but another error

 CC [M]  /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsMgmtTxâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1822:2: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsSetBeaconâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1878:2: warning: format â%sâ expects argument of type âchar *â, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1879:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1880:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1881:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1884:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1887:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsAddBeaconâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1933:2: warning: format â%sâ expects argument of type âchar *â, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1935:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1936:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1937:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1940:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1943:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2420:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2420:2: warning: (near initialization for âCFG80211_Ops.get_stationâ) [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2499:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2499:2: warning: (near initialization for âCFG80211_Ops.del_stationâ) [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2500:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2500:2: warning: (near initialization for âCFG80211_Ops.add_stationâ) [enabled by default]
[... cut ...]
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_p2p.c:29:
/usr/src/linux-headers-3.2.0-4-common/include/net/cfg80211.h:3092:6: note: declared here
make[4]: *** [/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_p2p.o] Erreur 1
make[3]: *** [_module_/root/drivers/Netgear-A6210/os/linux] Erreur 2
make[2]: *** [sub-make] Erreur 2
make[1]: *** [all] Erreur 2
make[1]: quittant le rÃ©pertoire Â« /usr/src/linux-headers-3.2.0-4-amd64 Â»
make: *** [debug] Erreur 2

Always the kernel version... i think
An idea ?

jurobystricky, 2016-03-15:
OK, so now modify the file os/linux/cfg80211/cfg80211_p2p.c around the line 112 to look like this:

    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,6,0)) 
		struct wireless_dev *pWdev = CFG80211_GetEventDevice(pAd)->ieee80211_ptr;
		cfg80211_remain_on_channel_expired(pWdev,
				pCfg80211_ctrl->Cfg80211ChanInfo.cookie, pCfg80211_ctrl->Cfg80211ChanInfo.chan,
				 GFP_KERNEL);
    #else
		cfg80211_remain_on_channel_expired(pAd->net_dev,
				pCfg80211_ctrl->Cfg80211ChanInfo.cookie, pCfg80211_ctrl->Cfg80211ChanInfo.chan,
				pCfg80211_ctrl->Cfg80211ChanInfo.ChanType, GFP_KERNEL);
    #endif /* LINUX_VERSION_CODE 3.6.11 */

aaronanglade, 2016-03-16:
Hi,

I have always an error

  CC [M]  /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsMgmtTxâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1822:2: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsSetBeaconâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1878:2: warning: format â%sâ expects argument of type âchar *â, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1879:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1880:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1881:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1884:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1887:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function âCFG80211_OpsAddBeaconâ:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1933:2: warning: format â%sâ expects argument of type âchar *â, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1935:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1936:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1937:2: warning: format â%dâ expects argument of type âintâ, but argument 2 has type âsize_tâ [-Wformat]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1940:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:1943:3: warning: passing argument 3 of âRTMP_COM_IoctlHandleâ discards âconstâ qualifier from pointer target type [enabled by default]
In file included from /root/drivers/Netgear-A6210/include/rt_config.h:49:0,
                 from /root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:50:
/root/drivers/Netgear-A6210/include/rtmp.h:6247:5: note: expected âvoid *â but argument is of type âconst u8 *â
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2420:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2420:2: warning: (near initialization for âCFG80211_Ops.get_stationâ) [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2499:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2499:2: warning: (near initialization for âCFG80211_Ops.del_stationâ) [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2500:2: warning: initialization from incompatible pointer type [enabled by default]
/root/drivers/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2500:2: warning: (near initialization for âCFG80211_Ops.add_stationâ) [enabled by default]
[... cut ...]
#endif /* LINUX_VERSION_CODE 3.6.11 */

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,34))
                        cfg80211_remain_on_channel_expired( pWdev,pCfg80211_ctrl->Cfg80211ChanInfo.cookie, pCfg80211_ctrl->Cfg80211ChanInfo.chan,                               
//jb was pCfg80211_ctrl->Cfg80211ChanInfo.ChanType, GFP_KERNEL);
                                 GFP_KERNEL);
#endif /* LINUX_VERSION_CODE 2.6.34 */

                pCfg80211_ctrl->Cfg80211RocTimerRunning = FALSE;
        }

Thanks for your answer

jurobystricky, 2016-03-16:
I installed debian 7.9, added the three changes listed above and the driver built and worked.
Please verify the code in the routine CFG80211RemainOnChannelTimeout looks like this:

    ...
    	{
    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,6,0))
		struct wireless_dev *pWdev = CFG80211_GetEventDevice(pAd)->ieee80211_ptr;
		DBGPRINT(RT_DEBUG_TRACE, ("CFG80211_ROC: RemainOnChannelTimeout -- FINISH\n"));

		cfg80211_remain_on_channel_expired( pWdev,
				pCfg80211_ctrl->Cfg80211ChanInfo.cookie, pCfg80211_ctrl->Cfg80211ChanInfo.chan,
				GFP_KERNEL);
    #else
		cfg80211_remain_on_channel_expired( pAd->net_dev,
				pCfg80211_ctrl->Cfg80211ChanInfo.cookie, pCfg80211_ctrl->Cfg80211ChanInfo.chan,
				pCfg80211_ctrl->Cfg80211ChanInfo.ChanType, GFP_KERNEL);
    #endif /* LINUX_VERSION_CODE 3.6.0 */

		pCfg80211_ctrl->Cfg80211RocTimerRunning = FALSE;
	}

aaronanglade, 2016-03-17:
It works for me. Thanks a lot.

closed 2016-03-17 by aaronanglade

tomsux, 2017-11-15:
Hi,
I try to compile this on centos6.8(2.6.32). I have this error. Can you help me ?

make -C /lib/modules/2.6.32-642.4.2.el6.i686/build DBGFLAGS=-DDBG SUBDIRS=/home/tom/a6210/Netgear-A6210-master/os/linux modules
make[1]: Entering directory `/usr/src/kernels/2.6.32-642.4.2.el6.i686'
  CC [M]  /home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.o
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsChannelSet’:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:233: error: ‘pDev’ undeclared (first use in this function)
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:233: error: (Each undeclared identifier is reported only once
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:233: error: for each function it appears in.)
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:653: warning: ‘enum tx_power_setting’ declared inside parameter list
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:653: warning: its scope is only this definition or declaration, which is probably not what you want
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:651: error: parameter 2 (‘Type’) has incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsStaGet’:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:776: error: ‘RATE_INFO_FLAGS_40_MHZ_WIDTH’ undeclared (first use in this function)
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:790: error: ‘STATION_INFO_TX_BITRATE’ undeclared (first use in this function)
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:798: error: ‘STATION_INFO_SIGNAL’ undeclared (first use in this function)
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsMgmtTx’:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1822: warning: passing argument 3 of ‘RTMP_COM_IoctlHandle’ discards qualifiers from pointer target type
/home/tom/a6210/Netgear-A6210-master/include/rtmp.h:6247: note: expected ‘void *’ but argument is of type ‘const u8 *’
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1862: warning: ‘struct beacon_parameters’ declared inside parameter list
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsSetBeacon’:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1871: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1871: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1872: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1872: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1873: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1874: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1890: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1892: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1892: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1894: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1896: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1896: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1898: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1899: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1902: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1903: error: dereferencing pointer to incomplete type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:1917: warning: ‘struct beacon_parameters’ declared inside parameter list
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsAddBeacon’:
[... cut ...]
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:2432: warning: initialization from incompatible pointer type
/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.c:2499: warning: initialization from incompatible pointer type
make[2]: *** [/home/tom/a6210/Netgear-A6210-master/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
make[1]: *** [_module_/home/tom/a6210/Netgear-A6210-master/os/linux] Error 2
make[1]: Leaving directory `/usr/src/kernels/2.6.32-642.4.2.el6.i686'
make: *** [debug] Error 2

genodeftest, 2017-11-15:
@tomsux :
1. Please use code blocks, otherwise your comment is barely readable and you're making everyone's life harder for no reason.
2. You should at least update to the latest CentOS release in case you expect any kind of software to keep working.
3. This kernel module has been developed against a recent version of the linux kernel. You should either update to a recent kernel (which is impossible on your outdated version of CentOS, so you need to update your operating system) or backport the driver.
4. Your issue is not related to the stuff above, so please open a separate bug report.