jurobystricky/Netgear-A6210 #11: Wireless no longer works after kernel upgrade

Trevonn, 2016-03-21:
I understand this is because the driver is only made and installed for the current running kernel version. Is there anyway to make it so that the driver is made and installed on all kernel versions similar to DKMS?

jurobystricky, 2016-03-25:
I think it should be possible, at least in theory. That is the purpose of DKMS after all.
(Need to go and ramp up on DKMS...)

steals, 2016-04-16:
I've created a DKMS config. You can find it here http://bazaar.launchpad.net/~steals/netgear-a6210/packaging/view/head:/dkms.conf

Also I created a PPA repo for Ubuntu with all required staff to install this driver as deb package. https://launchpad.net/~steals/+archive/ubuntu/netgear-a6210
(sudo apt-get install netgear-a6210-sta-dkms)

Trevonn, 2016-04-16:
Just what I wanted thank you.

Trevonn, 2016-04-16:
I would suggest adding this to the README then this issue can be closed.

jurobystricky, 2016-04-18:
@steals:Very nice job, indeed. I will place the info in the README file. Thanks.