jurobystricky/Netgear-A6210 #12: WNDA3100v3 - Wireless Won't Connect

jadenPete, 2016-03-23:
First, let me tell you, thank you! I am so very glad that someone is working on a driver for this device. Here's what I did:
    1. Downloaded the archive from github
    2. Extracted the archive
    3. Opened a terminal and went into the directory
    4. Ran the `make` command
    5. Ran `sudo make install`
    6. Ran `modprobe mt7662u_sta` (the driver it created)

Then, I looked, and poof! The network manager started receiving the signals for the wifi networks and displaying the networks. I tried connecting to my network, but it failed. I installed wicd. And after a whole minute of trying to authenticate, it said I had a bad password. I was using WPA2 - the encryption I normally use with that router. I tried the password multiple times, and looked at it (showing it in plain text), and it was my password!

I am running Ubuntu 14.04.4.
I am running the Linux Kernel 4.2.0.34.27 (Wily) from the main repositories.
I am using a Netgear WNDA3100v3 (as displayed in the title).

Hope I can get some help with this, and thanks!

jonipalosaari, 2016-03-23:
Google wpa_supplicant and try to configure it or use wep-encryption. Sounds like your driver works so this is not driver related issue.

jadenPete, 2016-03-23:
Tried the tutorial on the forums, didn't do anything. So I tried the WEP encryption, and same thing happened. What's really weird, is that my router's logs don't report any attempt to connect to the network, but when attempting to connect, I get some network activity, as shown in the attachment.
[image]
[image]

jadenPete, 2016-03-23:
BTW, I am using a tethered connection, as I regularly do.

jadenPete, 2016-03-24:
Ok I just tried to connect again, and it worked! I will close this issue, I think it's the hardware. I have had a few issues connecting in Windows, and the router is really far away.

closed 2016-03-24 by jadenPete