jurobystricky/Netgear-A6210 #13: Some Warnings left with gcc-4.9

bootsy52, 2016-04-14:
This is just an information about the success of building this driver on a Rasberry PI2+ with Kernel 4.4.7 and gcc-4.9 using rasbian OS. As I'm planning to buy EDUP EP-AC1605 because of the anntenas which is something I need. So I could test the module when I bought the wifi dongle. What is the current state, does it work somehow?

The only Warnings left during the compile process are these

In file included from /home/pi/src/Netgear-A6210/include/os/rt_linux.h:41:0,
                 from /home/pi/src/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/pi/src/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/pi/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
/home/pi/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c: In function ‘rt_ioctl_giwscan’:
include/net/iw_handler.h:559:3: warning: array subscript is below array bounds [-Warray-bounds]
   memcpy(stream + point_len, extra, iwe->u.data.length);
   ^
  CC [M]  /home/pi/src/Netgear-A6210/os/linux/../../mgmt/mgmt_vht.o
 ...
...
  CC [M]  /home/pi/src/Netgear-A6210/os/linux/../../common/cmm_data.o
/home/pi/src/Netgear-A6210/os/linux/../../common/cmm_data.c: In function ‘TxPktClassification’:
/home/pi/src/Netgear-A6210/os/linux/../../common/cmm_data.c:1085:4: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
    && (TxFrameType == TX_LEGACY_FRAME)
    ^
  CC [M]  /home/pi/src/Netgear-A6210/os/linux/../../common/rtmp_init.o
/home/pi/src/Netgear-A6210/os/linux/../../common/rtmp_init.c: In function ‘RTMPAllocAdapterBlock’:
/home/pi/src/Netgear-A6210/os/linux/../../common/rtmp_init.c:106:3: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘unsigned int’ [-Wformat=]
   DBGPRINT(RT_DEBUG_OFF, ("\n=== pAd = %p, size = %lu ===\n\n",
   ^

jurobystricky, 2016-04-14:
Thanks. I believe there were people who have successfully compiled the driver for Raspberry PI2.
You can probably safely ignore the first two warnings, I looked at the array subscript warning, and at this point it looks like false positive. Other gcc compiler versions don't issue ths warning.
The last warning should use %lz instead of %lu, as that is the correct format for size_t variables. (This works correctly for both 32 bit and 64 bit code)

bootsy52, 2016-04-26:
I also compiled the driver successfully but cannot test it, as the hardware has not
yet arrived. As I read some of these warnings were introduced with GCC-4.9 so 
that maybe the reason why other compiler versions don’t show them.

One question though, wouldn’t be a good idea to try to share this code / cooperate
with the manufacturer as there has been so much work done as I think why should this
be done twice.

P-i-U UG (haftungsbeschränkt) & Co. KG
Geschäftsführer
Dipl.-Wirt.-Ing. (FH) Carsten Menke
Zentrale: 0521 / 44 81 884 - 0
Talbrückenstr. 42

33611 Bielefeld