jurobystricky/Netgear-A6210 #14: Success stories!

swails, 2016-04-22:
First of all, I must say an emphatic thank you to @jurobystricky

A little Googling trying to find a driver for my new netgear N600 dongle brought me to this page, so I decided to give it a try.  Driver compiled easily and quickly, a quick little `insmod` followed by `ip addr show` and bam, there's `wlan0`.  I click on the network icon in my panel, select my router, and it connects.  It fumbles around a little at the beginning, but has been pretty stable since.  So to report on my configuration that works:

Gentoo Linux, kernel 3.18.9-gentoo, with a NETGEAR N600 WNDA3100 dongle

Thanks again for making my wireless adapter not useless!

jadenPete, 2016-04-22:
Same here :+1:

jdsmit60, 2016-07-03:
I would also like to say thank you to @jurobystricky for providing this driver. I have it working on Arch Linux, kernel version 4.5.4 with a NetGear A6210 dongle. Signal strength is a little poor compared to my laptop's built-in wireless card (Intel PRO 3945), but hey, it works and is much faster than my Intel card ever was.

I had a little trouble compiling at first but quickly realized that I didn't have linux-headers installed. Oops.

I noticed that this driver is nowhere to be found in the AUR (Arch User Repository) and I am thinking about creating a pkgbuild for this driver, something that would be a first-time thing for me... let's see if I can figure it out. If I do, I would like to ask @jurobystricky for permission to put it up on the AUR.

Once again, many thanks to @jurobystricky !

sabsowitz, 2016-07-09:
I'm using a Kali Linux VM. Just cloned the repository, ran `make`, and `make install`. Everything worked fine. The problem is that I'm not sure what to do now. Running `ifconfig` still shows just `lo` and `eth0`. Is there something I'm missing?

witi83, 2016-07-14:
@sabsowitz Have you tried to load the module using `modprobe mt7662u_sta`?

jadenPete, 2016-07-14:
Yeah you gotta load the module.

jadenPete, 2016-07-14:
And you can add it to `/etc/modules.conf` to make it start when you boot the computer

sabsowitz, 2016-07-15:
@witi83 @jadenPete Thanks. That's exactly what I was looking for. I hate being 'that guy,' and I knew I was missing the command to actually load the module. For whatever reason, I was having trouble finding that info.

jurobystricky, 2016-07-15:
The culprit may have been in "make install". Should be "sudo make install".

sabsowitz, 2016-07-16:
So I really hate to be that guy. But I've run `make` then `make install` (all with sudo), followed by `modprobe mt7662u_sta`. Haven't run into any errors, but still haven't gotten anywhere. Nothing showing up in `ifconfig` or `airmon-ng`. Please forgive me, but this is the first time I've done this on a VM. Is there something that I am missing?

DutchProgrammer, 2016-08-06:
Nice mine netgear wnda3100v3 is working on ubuntu 16.04 thanks to your driver !

jadenPete, 2016-08-06:
I have a Netgear WNDA3100v3 too!

paralaks, 2016-11-28:
It works with ASUS USB-AC55 on Ubuntu 16.04. Small issues: 
- Could not go beyond 300 Mhz
- There can be package loss as much as 10%
Still I am happy! :)

driventokill, 2016-12-31:
@jdsmit60 I'm trying it on arch linux with kernel `4.8.13-1-ARCH`.  My wireless adapter is netgear A6210 just like you. The `wlan0` showed up by with no mac address. Have you tried it with the latest kernel? 

`3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
`

ArnaudX, 2017-01-07:
hello 
seems it's not working with Debian Stretch 4.8.0-2-amd64 #1 SMP Debian 4.8.11-1 (2016-12-02)
dmesg :+1: [42446.022764] usb 4-1: new high-speed USB device number 14 using ehci-pci
[42446.176125] usb 4-1: New USB device found, idVendor=0846, idProduct=9053
[42446.176130] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[42446.176133] usb 4-1: Product: A6210
[42446.176135] usb 4-1: Manufacturer: NETGEAR
[42446.176138] usb 4-1: SerialNumber: 100
[42446.177649] 
               === pAd = ffffb38107741000, size = 1282584 ===

[42446.177652] driver version: 3.0.0.1 (Jan  7 2017 13:53:46) .
[42446.179131] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
[42446.188739] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[42446.188740] RtmpEepromGetDefault::e2p_dafault=1
[42446.188994] NVM is EFUSE mode
[42446.188996] Endpoint(8) is for In-band Command
[42446.188997] Endpoint(4) is for WMM0 AC0
[42446.188998] Endpoint(5) is for WMM0 AC1
[42446.188999] Endpoint(6) is for WMM0 AC2
[42446.188999] Endpoint(7) is for WMM0 AC3
[42446.189000] Endpoint(9) is for WMM1 AC0
[42446.189001] Endpoint(84) is for Data-In
[42446.189002] Endpoint(85) is for Command Rsp
[42446.189245] 80211> CurTxPower = 20 dBm
[42446.189246] ====> Radar Channel 52
[42446.189247] ====> Radar Channel 54
[42446.189248] ====> Radar Channel 56
[42446.189249] ====> Radar Channel 60
[42446.189249] ====> Radar Channel 62
[42446.189250] ====> Radar Channel 64
[42446.189251] ====> Radar Channel 100
[42446.189251] ====> Radar Channel 104
[42446.189253] 80211> TxStream = 0

and nothing is working
tested on Debian Jessie and working

andresrommier, 2017-02-06:
I was having similar issues (no networks detected) running on Arch Linux Kernel 4.9.

First I thought it was a problem with the kernel version so I switched to the LTS version (4.4), it didn't work so I decided to use Netctl and its app wifi-menu (in place of NetworkManager and its nm-applet), this worked (didn't try to run it on Linux 4.9 though).

karthicraghupathi, 2018-03-23:
@ArnaudX 

FWIW, I'm using this on Debian Stretch. The NM applet could not detect any Wi-Fi networks. I just chose the "Connect To Hidden Network" option and plugged in the name of the Wi-Fi network along with the password and I was able to succesfully connect and use the device.