jurobystricky/Netgear-A6210 #17: 5Ghz Network not Being Recongnized

jadenPete, 2016-04-30:
I'm running the Netgear WNDA3100v3 under the latest build of this driver, and for some reason, it won't detect my 5Ghz network, but it will with other networks. It's not the router's problem, because it works in Windows. So I ran depmod to see the issue, and I got the following:
`[   23.234323] Chan 108 (frq 5540):	not allowed!`
`[   23.234325] Chan 112 (frq 5560):	not allowed!`
`[   23.234327] Chan 116 (frq 5580):	not allowed!`
`[   23.234329] Chan 118 (frq 5590): 	not allowed!`
`[   23.234331] Chan 120 (frq 5600):	not allowed!`
`[   23.234333] Chan 124 (frq 5620):	not allowed!`
`[   23.234333] Chan 124 (frq 5620):	not allowed!`
`[   23.234335] Chan 126 (frq 5630):	not allowed!`
`[   23.234337] Chan 128 (frq 5640):	not allowed!`
`[   23.234339] Chan 132 (frq 5660):	not allowed!`
`[   23.234341] Chan 134 (frq 5670):	not allowed!`
`[   23.234343] Chan 136 (frq 5680):	not allowed!`
`[   23.234345] Chan 140 (frq 5700):	not allowed!`
`[   23.234348] Chan 167 (frq 5835):	not allowed!`
`[   23.234350] Chan 169 (frq 5845):	not allowed!`
`[   23.234352] Chan 171 (frq 5855):	not allowed!`
`[   23.234354] Chan 173 (frq 5865):	not allowed!`
`[   23.234356] Chan 184 (frq 4920):	not allowed!`
`[   23.234358] Chan 188 (frq 4940):	not allowed!`
`[   23.234360] Chan 192 (frq 4960):	not allowed!`
`[   23.234362] Chan 196 (frq 4980):	not allowed!`
`[   23.234364] Chan 208 (frq 6040):	not allowed!`
`[   23.234369] Chan 212 (frq 6060):	not allowed!`
`[   23.234369] Chan 216 (frq 6080):	not allowed!`

Seems fishy, because my network is on channel 157. Why isn't it detecting these channels?

ertheis, 2016-05-25:
Can confirm that 5ghz channels don't work on a Netgear WNDA3100v3. iwlist wlan0 freq produces:

wlan0     11 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Current Frequency:2.412 GHz (Channel 1)

jonipalosaari, 2016-07-14:
I can confirm 5Ghz channels works. I'm now connected to 5Ghz wifi and get ~80Mb/s down. 
I'm using systemd-networkd with wpa_supplicant on gentoo, if that matters. 

ping-tests:
`--- 192.168.1.1 ping statistics ---
247982 packets transmitted, 247982 received, 0% packet loss, time 2720578ms
rtt min/avg/max/mdev = 1.071/1.892/25.286/0.699 ms, pipe 2
`

And  there's wpa_cli results:
`> get_capability channels
Mode[G] Channels: 1 2 3 4 5 6 7 8 9 10 11 12 13
Mode[A] Channels: 36 38 40 44 46 48 52 54 56 60 62 64 100 104 108 112 116 118 120 124 126 128 132 134 136 140 149 151 153 157 159 161 165 167 169 171 173
Mode[B] Channels: 1 2 3 4 5 6 7 8 9 10 11 12 13
> get_capability freq
Mode[G] Channels:
 1 = 2412 MHz
 2 = 2417 MHz
 3 = 2422 MHz
 4 = 2427 MHz
 5 = 2432 MHz
 6 = 2437 MHz
 7 = 2442 MHz
 8 = 2447 MHz
 9 = 2452 MHz
 10 = 2457 MHz
 11 = 2462 MHz
 12 = 2467 MHz
 13 = 2472 MHz

Mode[A] Channels:
 36 = 5180 MHz
 38 = 5190 MHz
 40 = 5200 MHz
 44 = 5220 MHz
 46 = 5230 MHz
 48 = 5240 MHz
 52 = 5260 MHz (NO_IR) (DFS)
 54 = 5270 MHz (NO_IR) (DFS)
 56 = 5280 MHz (NO_IR) (DFS)
 60 = 5300 MHz (NO_IR) (DFS)
 62 = 5310 MHz (NO_IR) (DFS)
 64 = 5320 MHz (NO_IR) (DFS)
 100 = 5500 MHz (NO_IR) (DFS)
 104 = 5520 MHz (NO_IR) (DFS)
 108 = 5540 MHz (DFS)
 112 = 5560 MHz (DFS)
 116 = 5580 MHz (DFS)
 118 = 5590 MHz (DFS)
 120 = 5600 MHz (DFS)
 124 = 5620 MHz (DFS)
 126 = 5630 MHz (DFS)
 128 = 5640 MHz (DFS)
 132 = 5660 MHz (DFS)
 134 = 5670 MHz (DFS)
 136 = 5680 MHz (DFS)
 140 = 5700 MHz (DFS)
 149 = 5745 MHz
 151 = 5755 MHz
 153 = 5765 MHz
 157 = 5785 MHz
 159 = 5795 MHz
 161 = 5805 MHz
 165 = 5825 MHz
 167 = 5835 MHz
 169 = 5845 MHz
 171 = 5855 MHz
 173 = 5865 MHz

Mode[B] Channels:
 1 = 2412 MHz
 2 = 2417 MHz
 3 = 2422 MHz
 4 = 2427 MHz
 5 = 2432 MHz
 6 = 2437 MHz
 7 = 2442 MHz
 8 = 2447 MHz
 9 = 2452 MHz
 10 = 2457 MHz
 11 = 2462 MHz
 12 = 2467 MHz
 13 = 2472 MHz`

trensol, 2016-11-21:
Same issue here, would love to see 5ghz support

wlan0     11 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Current Frequency:2.417 GHz (Channel 2)