jurobystricky/Netgear-A6210 #18: kernel 4.4.8 problem

migueldvb, 2016-05-05:
I can compile the driver and a wlan0 device is created when I connect an AC1200 device but it disconnects frequently from the network. I am running kernel 4.4.8 on Archlinux

jurobystricky, 2016-05-05:
There is a fix which may help. (see issue #6)

The driver drops connection (LinkDown) if there is no beacon for 4 seconds, yet the "keep alive" packets are sent only each 10 seconds. There is some convoluted logic involved, but if you change in file common/mlme.c the line (around 1950) :

if ((pAd->Mlme.OneSecPeriodicRound % 10) == 8) {

to

if ((pAd->Mlme.OneSecPeriodicRound % 4) == 0) {

you should avoid connection losses.

jurobystricky, 2016-05-05:
I plan on a better fix eventually.

migueldvb, 2016-05-05:
Thanks! I will try your suggestion.