jurobystricky/Netgear-A6210 #19: Confirmed that this works on Fedora 23

schuylermartin45, 2016-06-10:
I couldn't find a better way to contact you guys but I wanted to thank you for your work, you all saved me a ton of headaches over the last day or two with this driver. I also wanted to let you know that I have successfully installed your driver twice on Fedora 23 (using two different kernels)

I just created a fresh install of Fedora 23 on my desktop and the driver works wonderfully. I even upgraded to the latest version of the kernel available on Fedora (4.5.6-200), recompiled and reinstalled the driver and it still works great.

Thanks,
Schuyler Martin

madmalkav, 2016-06-10:
No packet loss? Can you run a continuous ping for an hour or so and report
packet lost?

schuylermartin45, 2016-06-10:
The machine has been running updates on and off for the past 24 hours and I haven't noticed any networking issues or interruptions. I can set that test up for you as soon as I un-bork my graphics driver on this machine.

schuylermartin45, 2016-06-10:
I should probably also mention that my network configuration settings were preserved from prior to updating the kernel. But you guys probably don't modify or touch those configuration files.

schuylermartin45, 2016-06-11:
Here are the results from running ping for a little over an hour:
--- 8.8.8.8 ping statistics ---
3853 packets transmitted, 3615 received, 6% packet loss, time 3856788ms
rtt min/avg/max/mdev = 12.960/24.200/3095.555/64.551 ms, pipe 4

Note: I was using my desktop (occasionally web browsing) while this was running in the background.

madmalkav, 2016-06-11:
Interesting, you have less packet losses than me even if I ping my local
router to avoid external interference on the measure. Do you use
NetworkManager, wicd, or ...?

schuylermartin45, 2016-06-11:
NetworkManager

steals, 2016-06-11:
I have the following statistics:

- Ping from router (xiomi mi nano) to local pc (edup 1601):  
9042 packets transmitted, 8356 packets received, 7% packet loss
round-trip min/avg/max = 0.835/16.595/2008.831 ms

- Ping from local pc ( edup 1601) to local router (xiaomi mi nano):  
8180 packets transmitted, 7784 received, 4% packet loss, time 8194265ms
rtt min/avg/max/mdev = 0.823/25.173/2061.562/108.773 ms, pipe 3  

It looks like newer kernel(4.4+) improved wifi connection as previously I had ~ 15% packet loss when pinging local router.

chubei, 2016-06-30:
I tried this driver on Ubuntu 16.04 with ASUS USB-AC55 yesterday. It worked really well.

Didn't expect this when nearly all google results and ASUS support staff tell me this netcard won't work on Linux systems.

Really thank you guys. Saved my day.

jox58, 2016-09-09:
I have very little experience compiling against the kernel. When I tried to '`make`' on my 64 bit Fedora 24 I got the following error:
```
~HOME/Netgear-A6210/include/cfg80211.h:38:49: error: ‘IEEE80211_NUM_BANDS’ undeclared here (not in a function)
  struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
                                                 ^~~~~~~~~~~~~~~~~~~
```
I have no idea where or how IEEE80211_NUM_BANDS must be set. Can anymore offer me a suggestion?