jurobystricky/Netgear-A6210 #20: Make errors on debian

ReconCubed, 2016-06-16:
Getting some errors while trying to run make and make install on debian(kali)

```
    root@kali:~/Netgear-A6210# make
    export DBGFLAGS
    
    *** Building driver with debug messages ***
    
    cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
    make -C /lib/modules/4.3.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
    make[1]: *** /lib/modules/4.3.0-kali1-amd64/build: No such file or directory.  Stop.
    Makefile:59: recipe for target 'debug' failed
    make: *** [debug] Error 2
```

and

```
    root@kali:~/Netgear-A6210# sudo make install
    make -C /root/Netgear-A6210/os/linux -f Makefile.6 install
    make[1]: Entering directory '/root/Netgear-A6210/os/linux'
    rm -rf /etc/Wireless/RT2870STA
    mkdir -pv /etc/Wireless/RT2870STA
    mkdir: created directory ‘/etc/Wireless’
    mkdir: created directory ‘/etc/Wireless/RT2870STA’
    cp /root/Netgear-A6210/conf/RT2870STA.dat /etc/Wireless/RT2870STA/.
    install -d /lib/modules/4.3.0-kali1-amd64/kernel/drivers/net/wireless/
    install -m 644 -c mt7662u_sta.ko /lib/modules/4.3.0-kali1-amd64/kernel/drivers/net/wireless/
    install: cannot stat ‘mt7662u_sta.ko’: No such file or directory
    Makefile.6:721: recipe for target 'install' failed
    make[1]: *** [install] Error 1
    make[1]: Leaving directory '/root/Netgear-A6210/os/linux'
    Makefile:76: recipe for target 'install' failed
    make: *** [install] Error 2
```
Thanks! Currently running an apt-get upgrade. Hopefully this'll fix it.

ReconCubed, 2016-06-16:
apt-get upgrade did nothing :(

Stankye, 2016-08-25:
Same issue

jurobystricky, 2016-08-28:
looks like you don't have linux headers installed.
Please try:
apt-get install -y linux-headers-$(uname -r)

Falieson, 2019-01-31:
I have found that https://github.com/jeremyb31/rtl8822bu works without error