jurobystricky/Netgear-A6210 #21: Make problem (Manjaro)

HitCoder9768, 2016-07-04:
`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/hitcoder/Netgear-A6210-master/os/linux/Makefile
make -C /lib/modules/4.4.11-1-MANJARO/build DBGFLAGS=-DDBG SUBDIRS=/home/hitcoder/Netgear-A6210-master/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.4.11-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.4.11-1-MANJARO/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
`
The only kernel headers I can install are 4.4.13 under 4.4
There is no longer an option for 4.4.11, cannot build wifi drivers, main pc is now unusable.
Please fix ASAP because this is my main computer and my life is on my computer.

HitCoder9768, 2016-07-04:
I believe I have fixed it by manually changing the kernal directory at these lines

```
LINUX_SRC = /lib/modules/$(shell uname -r)/build
LINUX_SRC_MODULE = /lib/modules/$(shell uname -r)/kernel/drivers/net/wireless/
```

to

```
LINUX_SRC = /lib/modules/4.4.13-1-MANJARO/build
LINUX_SRC_MODULE = /lib/modules/4.4.13-1-MANJARO/kernel/drivers/net/wireless/
```

Scorch-Dev, 2016-09-30:
I'm having the same issue, but unfortunately this fix didn't do it for me. Still getting the same error, only difference is it provides a directory now that says 4.4.13-1-MANJARO. Any ideas somebody?

jurobystricky, 2016-09-30:
Seems you don't have the Linux kernel headers installed. This site may help you (manjaro):
http://paperiniktips.sviluppo-siti-web.com/arch-linux-how-to-find-and-install-the-right-kernel-headers/

Scorch-Dev, 2016-10-02:
@jurobystricky It worked perfectly, and thanks for getting back so fast