jurobystricky/Netgear-A6210 #23: hostAP support?

rezad1393, 2016-08-04:
is any of these adapter able to be use in AP mode (hostAP) ?
what I mean is can any of them be used as a wireless access point in linux?

jurobystricky, 2016-08-04:
I think there is conditional code there to support AP, ( I believe it is disabled by default)

rezad1393, 2016-08-04:
so yes ,no ?
any experience with it?
tplink and other brands support says they dont have softAP in linux but in a forum one of my firends said that he had softAP with tplink N900 .
it seems that even the support guys are confused.

jurobystricky, 2016-08-04:
Well, the answer is this: see config.mk. There are several configuration, we use
ifeq ($(RT28xx_MODE),STA) ....

But there are two additional ones (all mutually exclusive)  you may be interested in:
ifeq ($(RT28xx_MODE),AP)...
and
ifeq ($(RT28xx_MODE),APSTA)...

So the code to support pure AP or AP+STA driver should be present.
I never built for these configurations, so they may build and may even work.
But that would be rather an unexpected outcome. So good luck and any feedback is appreciated.

antismap, 2016-08-05:
I'm trying to get the AP mode to compile. First of all, the ap/ folder is missing. I copied it from the mediatek source that I got from their website. 
Many changes in the code are needed to fix thing such as include issues and missing defines. It still doesn't compile for me.
jurobystricky, is this driver a fork of the mediatek sources ?

jurobystricky, 2016-08-05:
see the REAME.md file for regarding the origin of the sources. And yes, it is pretty bad and ugly code.