jurobystricky/Netgear-A6210 #24: [Patch] Support for Kernel 4.7

witi83, 2016-08-10:
Hi @jurobystricky,

here is a first patch to get this driver up and running under Kernel 4.7. But I'm unsure whether this is the right place to define this stuff:

```diff
diff --git a/include/cfg80211.h b/include/cfg80211.h
index 275cf39..a402a2a 100644
--- a/include/cfg80211.h
+++ b/include/cfg80211.h
@@ -34,6 +34,13 @@ typedef enum _NDIS_HOSTAPD_STATUS {
 
 typedef struct __CFG80211_CB {
 
+    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 7, 0))
+    #define ieee80211_band nl80211_band
+    #define IEEE80211_BAND_2GHZ NL80211_BAND_2GHZ
+    #define IEEE80211_BAND_5GHZ NL80211_BAND_5GHZ
+    #define IEEE80211_NUM_BANDS NUM_NL80211_BANDS
+    #endif
+
        /* we can change channel/rate information on the fly so we backup them */
        struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
        struct ieee80211_channel *pCfg80211_Channels;
```

What do you think?

Cheers,
Witi

witi83, 2016-10-11:
Here we go:
https://github.com/jurobystricky/Netgear-A6210/pull/35

closed 2016-10-11 by witi83

No0to, 2019-09-03:
Sorry to comment on such an old thread, but when I try to install the patch using `git apply -v 24.patch`, I get this error:

`
Checking patch include/cfg80211.h...
error: while searching for:

typedef struct __CFG80211_CB {

       /* we can change channel/rate information on the fly so we backup them */
       struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
       struct ieee80211_channel *pCfg80211_Channels;

error: patch failed: include/cfg80211.h:34
error: include/cfg80211.h: patch does not apply
`
Does anybody know why this error is showing up and how to fix it?