jurobystricky/Netgear-A6210 #25: Ubuntu 16.04.1 Cannot get driver to work properly

wolfland, 2016-08-10:
I cannot seem to get my new A6210 Netgear adapter to work with this driver.
The driver seems to make/sudo make install fine as well as modprobe fine
but the adapter shows up as "device not managed"  if at all. I tried restarting
the network-manager service and it did not help. I did notice that there were
some segfaults happening.

I have included a cut/paste of my terminal session.

Any help would be appreciated and Thanks

[Netgear-A6210.txt](https://github.com/jurobystricky/Netgear-A6210/files/412007/Netgear-A6210.txt)

jurobystricky, 2016-08-10:
Thanks. I will install Ubuntu 16.04.1 and see if I can reproduce the problem.

wolfland, 2016-08-11:
Thanks.. If you need me to reproduce and gather more data or run some

test code just let me know.

chriscoyfish, 2016-08-14:
I also had some issues, but after fiddling around (unfortunately don't remember exact steps), I was able to get it the point where on boot I also get a SIGSEGV (here is the crash log: https://gist.github.com/chriscoyfish/49de32f33eed2e623ba29470ec746699 ) ...but if I "sudo service network-manager restart", the wlan0 interface gets detected and everything works as expected until after a reboot.

wolfland, 2016-08-14:
I tried again and this time it worked.  I am not sure why it didn't work 
last time
unless the recent kernel update fixed it.

Here is what I did..

from the directory that contains the extracted code:

$ make
$ sudo make install
$ sudo modprobe mt7662u_sta
$ sudo service network-manager restart

and this time it worked.

Thanks for the help and quick response and feel
free to close the issue.

jadenPete, 2016-08-14:
That happened to me too. Sometimes this driver can be a little stubborn. A couple months ago I was sitting with a useless WNDA3100v3, however I decided to try the driver and nearly lost hope when it wouldn't connect. However the next day I rebooted and it worked!

gforge, 2016-09-13:
@chriscoyfish also managed to get the driver installed but I seem to get the same SIGSEV error as you. 

I also have to do a:
`sudo modprobe mt7601u`

followed by:
`sudo service network-manager restart`

before it works.

XiaoLiuAI, 2017-06-03:
I used all the commands mentioned above. I can find the wifi setting and the detected wifi list. However, it cannot really connect to any of them. I tried to connect to hotspot of my iPhone, and it just keeps asking password. I tried to connect to a wifi without password, it always ends with "disconnected".

SamusArin, 2017-06-10:
uname

> 4.8.0-54-generic #57~16.04.1-Ubuntu x86_64

make **[ok]**
sudo make install **[ok]**
sudo modprobe mt7662u_sta **[ok]**
sudo service network-manager restart

note: wlp2s0 is built-in wifi interface.

> wlp2s0: deauthenticating from c8:b3:73:11:bc:e9 by local choice (Reason: 3=DEAUTH_LEAVING)
> IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
> -->RTUSBVendorReset
> <--RTUSBVendorReset
> fw version:0.0.00 build:1
> build time:201406241830____
> fw for E3 IC
> ilm length = 59312(bytes)
> dlm length = 32068(bytes)
> loading fw........
> Unexpected gfp: 0x2000000 (0x2000000). Fixing up to gfp: 0x80020 (__GFP_HIGH|__GFP_ATOMIC). Fix your code!
> Read file "/etc/Wireless/RT2870STA/RT2870STA.dat" failed(errCode=-14)!
> ERROR mt766u_sta:RTMPReadParametersHook failed, Status[=0xfffffff2]
> RTMP_TimerListRelease: release timer obj ffffba6c00a58440!
> RTMP_TimerListRelease: release timer obj ffffba6c00a584b0!
> RTMP_TimerListRelease: release timer obj ffffba6c00a58520!
> RTMP_TimerListRelease: release timer obj ffffba6c00a583d0!
> RTMP_TimerListRelease: release timer obj ffffba6c00a58280!
> RTMP_TimerListRelease: release timer obj ffffba6c00a582f0!
> RTMP_TimerListRelease: release timer obj ffffba6c009e9f70!
> RTMP_TimerListRelease: release timer obj ffffba6c009d8d78!
> RTMP_TimerListRelease: release timer obj ffffba6c009d8df0!
> RTMP_TimerListRelease: release timer obj ffffba6c009ea0e0!
> RTMP_TimerListRelease: release timer obj ffffba6c009e9e90!
> RTMP_TimerListRelease: release timer obj ffffba6c009ea070!
> receive cmd msg fail(-2)
> tx_kickout_fail_count = 0
> tx_timeout_fail_count = 0
> rx_receive_fail_count = 0
> alloc_cmd_msg = 1
> free_cmd_msg = 1
> ------------[ cut here ]------------
> WARNING: CPU: 1 PID: 3965 at .../ng_a6210/Netgear-A6210-master/os/linux/../../common/rtmp_init_inf.c:491 rt28xx_init+0x50/0x600 [mt7662u_sta]
> Modules linked in: mt7662u_sta(OE) snd_hda_codec_conexant snd_hda_codec_generic snd_hda_intel intel_rapl arc4 x86_pkg_temp_thermal intel_powerclamp rtl8192ce coretemp rtl_pci snd_hda_codec rtl8192c_common snd_hda_core rtlwifi snd_hwdep crct10dif_pclmul mac80211 cfg80211 snd_pcm crc32_pclmul ghash_clmulni_intel snd_seq_midi snd_seq_midi_event cryptd snd_rawmidi snd_seq intel_cstate intel_rapl_perf joydev input_leds serio_raw snd_seq_device snd_timer mei_me toshiba_acpi shpchp sparse_keymap snd mei industrialio mac_hid lpc_ich toshiba_bluetooth wmi soundcore parport_pc ppdev lp parport autofs4 ums_realtek uas usb_storage i915 psmouse ahci libahci atl1c i2c_algo_bit drm_kms_helper syscopyarea video sysfillrect fjes sysimgblt fb_sys_fops drm
> **CPU: 1 PID: 3965 Comm: NetworkManager Tainted: G        W  OE   4.8.0-54-generic #57~16.04.1-Ubuntu**
> Hardware name: TOSHIBA Satellite C655/Portable PC, BIOS 1.30 11/21/2011
>  0000000000000286 00000000502bb6bb ffff9f5983f5f618 ffffffffa5a2e7b3
>  0000000000000000 0000000000000000 ffff9f5983f5f658 ffffffffa568314b
>  000001eb83f5f668 ffffba6c009d6000 0000000000000001 ffffffffc0a07b24
> Call Trace:
>  [<ffffffffa5a2e7b3>] dump_stack+0x63/0x90
>  [<ffffffffa568314b>] __warn+0xcb/0xf0
>  [<ffffffffa568327d>] warn_slowpath_null+0x1d/0x20
>  [<ffffffffc0996190>] rt28xx_init+0x50/0x600 [mt7662u_sta]
>  [<ffffffffc09cec07>] rt28xx_open+0xc7/0x150 [mt7662u_sta]
>  [<ffffffffc09a64a7>] RTMP_COM_IoctlHandle+0x6d7/0x790 [mt7662u_sta]
>  [<ffffffffc09cee81>] MainVirtualIF_open+0x51/0xe0 [mt7662u_sta]
>  [<ffffffffc09ceb40>] ? RT28xx_get_ether_stats+0x140/0x140 [mt7662u_sta]
>  [<ffffffffc09cedb0>] ? rt28xx_ioctl+0x80/0x80 [mt7662u_sta]
>  [<ffffffffa5d824c8>] __dev_open+0xc8/0x140
>  [<ffffffffa5d827dd>] __dev_change_flags+0x9d/0x160
>  [<ffffffffa5d828c9>] dev_change_flags+0x29/0x70
>  [<ffffffffa5d91b83>] do_setlink+0x693/0xce0
>  [<ffffffffa5a5c791>] ? __nla_reserve+0x41/0x50
>  [<ffffffffa5a5c8d0>] ? __nla_put+0x20/0x30
>  [<ffffffffa5e9a520>] ? _raw_read_unlock_bh+0x20/0x30
>  [<ffffffffa5e33f1d>] ? inet6_fill_ifla6_attrs+0x43d/0x460
>  [<ffffffffa5e33f5a>] ? inet6_fill_link_af+0x1a/0x30
>  [<ffffffffa5d94755>] ? rtnl_fill_ifinfo+0xbe5/0x10a0
[... 18 lines cut ...]
>  [<ffffffffa568b892>] ? __do_proc_dointvec+0x322/0x3d0
>  [<ffffffffa5a3ddca>] ? lockref_put_or_lock+0x5a/0x80
>  [<ffffffffa584c065>] ? dput+0x175/0x250
>  [<ffffffffa5855964>] ? mntput+0x24/0x40
>  [<ffffffffa58351a3>] ? __fput+0x193/0x230
>  [<ffffffffa5852ab5>] ? __fget_light+0x25/0x60
>  [<ffffffffa5d61eb4>] __sys_sendmsg+0x54/0x90
>  [<ffffffffa5d61f02>] SyS_sendmsg+0x12/0x20
>  [<ffffffffa5e9a876>] entry_SYSCALL_64_fastpath+0x1e/0xa8
> ---[ end trace 631d7aaf1251e90a ]---
> !!! rt28xx init fail !!!

ifconfig wlan1

> wlan1     Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
>           BROADCAST MULTICAST  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sudo ifconfig wlp2s0 down [ok]
sudo ifconfig wlan1 up

Terminal hangs, cannot unload mt7662u_sta module, and cannot shutdown (hangs, requires hard power-off).

yszheda, 2017-08-10:
I encountered the same problem as @XiaoLiuAI describes.

- hardware: ASUS USB-AC55 USB Wireless adapter
- kernel version and OS: 4.8.0-36-generic #36~16.04.1-Ubuntu
- `dmesg`: 

```
[  754.091203] 80211> Connect bssid 1c:b7:2c:75:8e:bc
[  754.199513] PeerBeaconAtJoinAction(): HT-CtrlChannel=44, CentralChannel=>46
[  754.199516] PeerBeaconAtJoinAction(): Set CentralChannel=46
[  754.199521] AdjustChannelRelatedValue(): Input BW=1, rf_channel=46, vht_bw=1, Channel=44, vht_cent_ch=46!
[  754.262700] /media/ys/0D6911000D691100/Downloads/Netgear-A6210-master/os/linux/../../sta/rtmp_data.c:413 assert pRxBlk->wcid == BSSID_WCIDfailed
[  754.262707] [255] 1: 9c:5c:8e:b1:3e:4c, 2: 1c:b7:2c:75:8e:bc, 3:1c:b7:2c:75:8e:bc
[  754.263312] mt76x2_reinit_agc_gain:original agc_vga0 = 0x52, agc_vga1 = 0x52
[  754.263315] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x52, agc_vga1 = 0x52
[  754.263476] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x2c, hi_lna1 = 0x2c
[  754.263478] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x2c, hi_lna1 = 0x2c
[  754.263597] original vga value(chain0) = 52
[  754.263639] original vga value(chain1) = 52
[  754.263803] mt76x2_calibration(channel = 46)
[  756.062309] Rcv Wcid(1) AddBAReq
[  756.062313] Start Seq = 00000005
[  757.125376] 80211> bss inform fail ==> 4
[  757.125385] 80211> bss inform fail ==> 4
[  757.125392] 80211> bss inform fail ==> 4
[  757.659138] mt76x2_switch_channel:: Wait until MAC 0x1200 bit0 and BBP 0x2130 become 0 > 300 times
[  757.659141] mt76x2_switch_channel:: doing IBI and core soft progress
[  774.554762] RTMP_TimerListRelease: release timer obj ffffc6aac1391f78!
[  775.061065] 80211> Connect bssid d4:ee:07:3e:f4:1c
[  775.134018] PeerBeaconAtJoinAction(): HT-CtrlChannel=13, CentralChannel=>11
[  775.134020] PeerBeaconAtJoinAction(): Set CentralChannel=11
[  775.134025] AdjustChannelRelatedValue(): Input BW=1, rf_channel=11, vht_bw=1, Channel=13, vht_cent_ch=11!
[  775.918072] PeerBeaconAtJoinAction(): HT-CtrlChannel=11, CentralChannel=>11
[  775.918075] PeerBeaconAtJoinAction(): Set CentralChannel=11
[  775.918079] AdjustChannelRelatedValue(): Input BW=0, rf_channel=11, vht_bw=1, Channel=11, vht_cent_ch=11!
[  776.264883] 80211> Connect bssid 1c:b7:2c:75:8e:b8
[  776.362993] PeerBeaconAtJoinAction(): HT-CtrlChannel=13, CentralChannel=>11
[  776.362996] PeerBeaconAtJoinAction(): Set CentralChannel=11
[  776.363001] AdjustChannelRelatedValue(): Input BW=1, rf_channel=11, vht_bw=1, Channel=13, vht_cent_ch=11!
[  777.150056] PeerBeaconAtJoinAction(): HT-CtrlChannel=11, CentralChannel=>11
[  777.150058] PeerBeaconAtJoinAction(): Set CentralChannel=11
[  777.150063] AdjustChannelRelatedValue(): Input BW=0, rf_channel=11, vht_bw=1, Channel=11, vht_cent_ch=11!
[  777.216062] ------------[ cut here ]------------
[  777.216124] WARNING: CPU: 7 PID: 231 at /build/linux-hwe-eyfT8D/linux-hwe-4.8.0/net/wireless/sme.c:724 __cfg80211_connect_result+0x3ab/0x410 [cfg80211]
[  777.216126] Modules linked in: bnep dm_crypt eeepc_wmi asus_wmi sparse_keymap mt7662u_sta(OE) cfg80211 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic intel_rapl x86_pkg_temp_thermal snd_hda_intel coretemp snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device kvm snd_timer snd input_leds joydev irqbypass soundcore hci_uart btbcm btqca crct10dif_pclmul btintel crc32_pclmul bluetooth shpchp serio_raw ghash_clmulni_intel aesni_intel mei_me aes_x86_64 mac_hid mei lrw glue_helper ablk_helper cryptd acpi_als kfifo_buf intel_lpss_acpi intel_lpss industrialio acpi_pad parport_pc ppdev lp parport autofs4 hid_generic usbhid nouveau i915 mxm_wmi ttm i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops psmouse r8169
[  777.216229]  drm ahci mii libahci wmi video i2c_hid hid fjes
[  777.216248] CPU: 7 PID: 231 Comm: kworker/u16:5 Tainted: G        W  OE   4.8.0-36-generic #36~16.04.1-Ubuntu
[  777.216251] Hardware name: System manufacturer System Product Name/B250M-KYLIN, BIOS 0808 06/16/2017
[  777.216292] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  777.216296]  0000000000000286 00000000dadcbcd3 ffff8e2ef4fc7ca8 ffffffff9aa2d7b3
[  777.216303]  0000000000000000 0000000000000000 ffff8e2ef4fc7ce8 ffffffff9a68313b
[  777.216308]  000002d400000000 ffff8e2ef3ae1400 0000000000000000 ffff8e2ef3aef000
[  777.216314] Call Trace:
[  777.216322]  [<ffffffff9aa2d7b3>] dump_stack+0x63/0x90
[  777.216329]  [<ffffffff9a68313b>] __warn+0xcb/0xf0
[  777.216335]  [<ffffffff9a68326d>] warn_slowpath_null+0x1d/0x20
[  777.216365]  [<ffffffffc087f16b>] __cfg80211_connect_result+0x3ab/0x410 [cfg80211]
[  777.216390]  [<ffffffffc08557f4>] cfg80211_process_wdev_events+0x144/0x1b0 [cfg80211]
[  777.216414]  [<ffffffffc0855898>] cfg80211_process_rdev_events+0x38/0x70 [cfg80211]
[  777.216434]  [<ffffffffc085001e>] cfg80211_event_work+0x1e/0x30 [cfg80211]
[  777.216439]  [<ffffffff9a69d8eb>] process_one_work+0x16b/0x4a0
[  777.216442]  [<ffffffff9a69dc6b>] worker_thread+0x4b/0x500
[  777.216446]  [<ffffffff9a69dc20>] ? process_one_work+0x4a0/0x4a0
[  777.216451]  [<ffffffff9a6a4008>] kthread+0xd8/0xf0
[  777.216458]  [<ffffffff9ae9679f>] ret_from_fork+0x1f/0x40
[  777.216463]  [<ffffffff9a6a3f30>] ? kthread_create_on_node+0x1e0/0x1e0
[  777.216466] ---[ end trace afb6405f878672c0 ]---
[  777.218639] mt76x2_reinit_agc_gain:original agc_vga0 = 0x5e, agc_vga1 = 0x5e
[  777.218642] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x5e, agc_vga1 = 0x5e
[  777.218803] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x27, hi_lna1 = 0x27
[  777.218805] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x27, hi_lna1 = 0x27
[  777.218927] original vga value(chain0) = 5e
[  777.218969] original vga value(chain1) = 5e
[  777.219133] mt76x2_calibration(channel = 11)
[  778.216134] Rcv Wcid(1) AddBAReq
[  778.216138] Start Seq = 00000000
```

kcauchy, 2017-09-27:
@chriscoyfish Same problem like you. Mine is Ubuntu 16.04.3 LTS. Kernel 4.10.0-35.

Sometimes a restart for network manager works, sometime it doesn't.. Don't know why...