jurobystricky/Netgear-A6210 #26: X11 freezes on connect [4.7.0-1-ARCH]

intxcc, 2016-08-11:
Used:
- Arch 64 Bit with latest updates
- Netgear A6210
- mt7662u_sta

Just tried to build the driver for the newest kernel with this patch https://github.com/jurobystricky/Netgear-A6210/issues/24. Using NetworkManager everything seems fine at first, although the Mac is shown as 00:00:00:00:00:00. When trying to connect to an access point X11 freezes and mt7662u_sta crashed.

```
[  747.328092] usb 2-1.2: new high-speed USB device number 4 using ehci-pci
[  747.989645] rtusb init rt2870 --->
[  747.990002] 
               === pAd = ffffc90000d73000, size = 1292952 ===

[  747.990008] driver version: 3.0.0.1 (Aug 11 2016 23:43:18) .
[  747.990756] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
[  748.001389] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[  748.001404] RtmpEepromGetDefault::e2p_dafault=1
[  748.001883] NVM is EFUSE mode
[  748.001888] Endpoint(8) is for In-band Command
[  748.001891] Endpoint(4) is for WMM0 AC0
[  748.001904] Endpoint(5) is for WMM0 AC1
[  748.001906] Endpoint(6) is for WMM0 AC2
[  748.001908] Endpoint(7) is for WMM0 AC3
[  748.001910] Endpoint(9) is for WMM1 AC0
[  748.001912] Endpoint(84) is for Data-In
[  748.001924] Endpoint(85) is for Command Rsp
[  748.002156] 80211> CurTxPower = 20 dBm
[  748.002160] ====> Radar Channel 52
[  748.002162] ====> Radar Channel 54
[  748.002173] ====> Radar Channel 56
[  748.002175] ====> Radar Channel 60
[  748.002177] ====> Radar Channel 62
[  748.002178] ====> Radar Channel 64
[  748.002180] ====> Radar Channel 100
[  748.002182] ====> Radar Channel 104
[  748.002184] 80211> TxStream = 0
[  748.002860] usbcore: registered new interface driver rt2870
[  779.838334] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  779.839457] -->RTUSBVendorReset
[  779.859658] <--RTUSBVendorReset
[  779.861832] ...........andes_usb_chk_crc
[  779.955330] andes_usb_reset_wmt
[  779.976164] -->RTUSBVendorReset
[  779.996381] <--RTUSBVendorReset
[  779.996927] fw version:0.0.00 build:1
[  779.996934] build time:201406241830____
[  779.996949] fw for E3 IC
[  779.996953] ilm length = 59312(bytes)
[  779.996955] dlm length = 32068(bytes)
[  779.998012] loading fw........
[  780.060071] cfg_mode=5
[  780.060292] Key1Str is Invalid key length(0) or Type(0)
[  780.060308] Key2Str is Invalid key length(0) or Type(0)
[  780.060323] Key3Str is Invalid key length(0) or Type(0)
[  780.060339] Key4Str is Invalid key length(0) or Type(0)
[  780.060757] USBAggregation = 1
[  780.060761] 1. Phy Mode = 31
[  780.060763] NVM is Efuse and its size =1d[1e0-1fc] 
[  780.089975] get_chl_grp:illegal channel (167)
[  780.089992] get_chl_grp:illegal channel (167)
[  780.089995] get_chl_grp:illegal channel (169)
[  780.089997] get_chl_grp:illegal channel (169)
[  780.090000] get_chl_grp:illegal channel (171)
[  780.090002] get_chl_grp:illegal channel (171)
[  780.090004] get_chl_grp:illegal channel (173)
[  780.090006] get_chl_grp:illegal channel (173)
[  780.092468] Country Region from e2p = ffff
[  780.105729] mt76x2_get_external_lna_gain::LNA type=0x0, BLNAGain=0xffffff8d, ALNAGain0=0xffffff89, ALNAGain1=0xffffff88, ALNAGain2=0xffffff87
[  780.105739] 2. Phy Mode = 31
[  780.105743] 3. Phy Mode = 31
[... 158 lines cut ...]
[ 1327.939646]  [<ffffffff815c5664>] ? wireless_nlevent_flush+0x54/0x90
[ 1327.939707]  [<ffffffffa0b07ea5>] RTMP_COM_IoctlHandle+0x6b5/0x780 [mt7662u_sta]
[ 1327.939758]  [<ffffffff8154b99d>] ? inetdev_event+0x3d/0x500
[ 1327.939811]  [<ffffffffa0b36d86>] CFG80211NetdevNotifierEvent+0x106/0x190 [mt7662u_sta]
[ 1327.939873]  [<ffffffff8109a34a>] notifier_call_chain+0x4a/0x70
[ 1327.939956]  [<ffffffff8109a486>] raw_notifier_call_chain+0x16/0x20
[ 1327.940009]  [<ffffffff814cb6c5>] call_netdevice_notifiers_info+0x35/0x60
[ 1327.940082]  [<ffffffff814cba55>] __dev_close_many+0x55/0x100
[ 1327.940134]  [<ffffffff814cbb8c>] dev_close_many+0x8c/0x130
[ 1327.940187]  [<ffffffff814d0fca>] dev_close.part.19+0x4a/0x70
[ 1327.940241]  [<ffffffff814d100a>] dev_close+0x1a/0x20
[ 1327.940301]  [<ffffffffa08e6d55>] cfg80211_shutdown_all_interfaces+0x45/0xb0 [cfg80211]
[ 1327.940360]  [<ffffffffa08e6e26>] cfg80211_rfkill_set_block+0x26/0x30 [cfg80211]
[ 1327.940412]  [<ffffffffa057af19>] rfkill_set_block+0x89/0x130 [rfkill]
[ 1327.940456]  [<ffffffffa057b173>] rfkill_fop_write+0x103/0x1a0 [rfkill]
[ 1327.942553]  [<ffffffff811f94d7>] __vfs_write+0x37/0x140
[ 1327.944652]  [<ffffffff811fa2a6>] vfs_write+0xb6/0x1a0
[ 1327.946727]  [<ffffffff811fb725>] SyS_write+0x55/0xc0
[ 1327.948771]  [<ffffffff815dd732>] entry_SYSCALL_64_fastpath+0x1a/0xa4
[ 1327.950794] Code: 63 e0 e9 c5 fa ff ff 48 c7 c7 98 d8 b7 a0 e8 47 1f 63 e0 e9 d8 f4 ff ff 48 c7 c6 60 2e b6 a0 48 c7 c7 48 d9 b7 a0 e8 2f 1f 63 e0 <80> bb 51 a3 13 00 01 0f 85 eb ed ff ff 83 3d bf 7e 06 00 01 0f 
[ 1327.955115] RIP  [<ffffffffa0b3e4ad>] CFG80211DRV_IoctlHandle+0x138d/0x1400 [mt7662u_sta]
[ 1327.957189]  RSP <ffff8800ce173b58>
[ 1327.959150] CR2: 0000000f0013a355
[ 1327.969373] ---[ end trace 75d2d708fd76d3fd ]---
```

jadenPete, 2016-08-12:
Linux kernel panics? Now I haven't seen that in a long time...

intxcc, 2016-08-12:
Yes, me neither. At first I thought the error might come from these dmesg lines.

```
[...]
[  780.287550] Chan 212 (frq 6060): not allowed!
[...]
```

But even if this has something to do with a misconfiguration, the driver shouldn't be able to crash the whole system.

jadenPete, 2016-08-12:
Oh I have those issues to with the dmesg lines. I am strongly convinced it is why I can't get 5Ghz working  on my Netgear WNDA3100v3. Those channels in dmesg are where my 5Ghz network is in, and where 5Ghz is. 2.4Ghz works though, because it is channels 1-14. (1-11 in the USA). I know it is not a problem with my card, because it works in Windoze.

jadenPete, 2016-08-12:
I have also heard that this driver doesn't do well with newer kernels. [Try this updated fork I use](https://github.com/worralph/netgear-a6210-driver-dkms). It has DKMS support, is updated with the latest commits, fixed the GCC warning messages, and has AUR support (I think). It also is patched for newer kernels!

bobberb, 2016-10-12:
I encountered this on my arch-ification of a box today. Will check the fork and report back.