jurobystricky/Netgear-A6210 #27: Constant disconnecting

flyingpies155, 2016-08-26:
I am running on Arch with the Linux-ck 4.7.1 kernel, and my Wi-Fi is constantly disconnecting, with a popup asking for the password (but it's already filled in, I just have to click 'connect') [See here if that's not clear](http://i.imgur.com/mH2cNV7.jpg)

I'm not sure what to put here to help, but it seems to only happen when I'm using a decent amount of bandwidth (eg. watching YouTube or Twitch).

Anyone else having this issue?

jurobystricky, 2016-08-27:
There is a fix which may help. (see issue #6)

The driver drops connection (LinkDown) if there is no beacon for 4 seconds, yet the "keep alive" packets are sent only each 10 seconds. There is some convoluted logic involved, but if you change in file common/mlme.c the line (around 1950) :

if ((pAd->Mlme.OneSecPeriodicRound % 10) == 8) {

to

if ((pAd->Mlme.OneSecPeriodicRound % 4) == 0) {

you should avoid connection losses.
It is not an ideal fix, but in your case it should help.

flyingpies155, 2016-08-27:
Thanks. I haven't really had the issue since I posted, but I'll update if it ever happens again.