jurobystricky/Netgear-A6210 #29: kernel panic on ubuntu trusty

bonnaffe, 2016-09-07:
commit 7c0dee7943c835910d2a2759f7e45ea08fddda24
environment : ubuntu 14.04 (trusty tahr), 32-bit, kernel 3.13.0.87
panic happens during system init, easily reproducible

[image]

OCR-translated text below (subject to typos)
----------------------------------------------------------
```
[ 32.566242] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: f9acb177 
[ 32.566292] CPU: 3 PID: 1302 Comm: RtmpMlmeTask Tainted: G O% 3,13.0-87-generic #133-Ubuntu 
[ 32.566330] Hardware name: MSI MS-7758/B75A-G43 (MS-7758), BIOS U10,0 02/21/2012 
[ 32.566361] 00000086 00000086 eca87db8 c165d3ed f9942f40 00000001 eca87dd8 c16580a7
[ 32.566404] c183bec8 clacic80 00000008 f9942f40 00000001 f988c000 eca87de8 c105777e
[ 32.566447] c183beec f9acb177 eca87e44 f9acb/77 01010000 00000000 00000000 edc04b67 
[ 32.566490] Call Trace: 
[ 32.566507] [<c165d3ed>] dump_stack+Ox58/0x72
[ 32.566541] [<c16580a7>] panic+0x87/0x18b 
[ 32.566564] [<c105777e>] __stack_chk_fail+Oxle/Ox20 
[ 32.566603] [<f9acb177>1 ? RTMPSetSupportMCS+0x627/0x650 [mt7662u_sta] 
[ 32.566643] [<f9acb177>] RTMPSetSupportMCS+0x627/0x650 [mt7662u_sta] 
[ 32.566683] [<f9a76290>] PeerAssoclispAction+0x3e0/0x470 [mt7662u_sta] 
[ 32.566716] [<c103040a>] ? mce_setup+Oxaa/Ox120 
[ 32.566747] [<f9a9907e>] MlmeThread+0x27e/Ox4d0 [mt7662u_sta] 
[ 32.566785] [<f9a98e00>] ? ARC4_Discard_ReyLength+Ox60/0x60 [mt7662u_sta] 
[ 32.566817] [<c107678b>] kthread+Ox9b/OxbO
[ 32.566838] [<c166b777>] ret_from_kernel_thread+Ox1b/Ox28
[ 32.566862] [<c10766f0>] ? kthread_create_on_node+Ox140/0x140 
[ 32.566892] drm_kms_helper: panic occurred, switching back to text console
```

jadenPete, 2016-09-08:
Geez what is with the kernel panics on this driver.

jurobystricky, 2016-09-08:
easily reproducible...How about exact steps to reproduce it?

jadenPete, 2016-09-09:
I'm running ubuntu trusty and I don't get a kernel panic

jonipalosaari, 2016-09-09:
@bonnaffe This happened first time and after installation? What if you plug your device after boot?

@jadenPete that's cool, but tells nothing :)

jadenPete, 2016-09-10:
Might want to try kernel 4.4. I'm running 14.04.5, and J get no issues. No reboot needed. I did have my device plugged in already before I loaded the driver though.

bonnaffe, 2016-09-14:
@jurobystricky to reproduce on kernel 3.13.0-87, you can go the coldplug way: boot the system with the device plugged in, and the kernel crashes before gdm is started. 

To get more details, you can also reproduce it the hotplug way : 
`# modprobe mt7662u_sta`

no crash on module insertion, syslog shows : 

> cfg80211: Calling CRDA to update world regulatory domain
rtusb init rt2870 --->
usbcore: registered new interface driver rt2870
cfg80211: World regulatory domain updated:
cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm)
cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)

Then plug the device (USB2 and USB3 ports show the same behavior), you will get additional logs (see attachment).
[ubuntu14.04_kernel3.13.0-87_syslog-on-device-plug.txt](https://github.com/jurobystricky/Netgear-A6210/files/472224/ubuntu14.04_kernel3.13.0-87_syslog-on-device-plug.txt)

Then immediately the system freezes, sometimes my display switches to console mode with the kernel stack logs (cf bug description)

bonnaffe, 2016-09-15:
@jadenPete I had issues booting Trusty on a kernel more recent than 3.13.0-87, so I've done a fresh install. Ubuntu 16.04.1 (xenial), 64-bit.
Same exact crash and call trace.

jadenPete, 2016-09-16:
Must be a driver specific to your hardware

mehlis, 2016-12-19:
Or the function RTMPSetSupportMCS trashes the stack....on my system the VHT is not working and "iw list" reports 8x8 AC capabilities...so there is something wrong with MCS rates.