jurobystricky/Netgear-A6210 #31: Manjaro - Nothing happens

LordHultonus, 2016-09-18:
So I just installed the driver on my latest Manjaro with 4.4.20 kernel, after reboot... nothing happens. I keep unplugging and plugging the adapter back again but it looks like it doesnt even recognize it as a device plugged in...

jurobystricky, 2016-09-20:
Does the USB dongle plugging/unplugging show up in dmesg?

LordHultonus, 2016-09-21:
Nope, but PC recognise that somenthing was plugged in (adapter appears as " Bus 002 Device 004: ID 0846:9053 NetGear, Inc. " after running lsusb command. 
I suppose that drivers just dont scan theese USB ports for adapter so its not workin.

jonipalosaari, 2016-09-21:
and you've modproped driver?

LordHultonus, 2016-09-21:
Nope. Im kinda newbie to linux, never used modprobe before.

jonipalosaari, 2016-09-21:
lsmod should output "mt7662u_sta". If not, then "modprobe mt7662u_sta".

LordHultonus, 2016-09-21:
Its there, no need to modprobe...
mt7662u_sta 
Used by: 0

Apparently it doesnt know which device should it use...

jurobystricky, 2016-09-22:
Can you post the log? (dmesg)

LordHultonus, 2016-09-22:
Actually I've found out a very interesting thing...
When I plug in the adapter, whole system appears to get crazy. Plasma stops working, console doesnt do what I tell it to do (even basic commands like sudo reboot dont work). When I unplug the adapter, everything is back to normal. What the hell?

jurobystricky, 2016-09-22:
Well, this is where a log file would be really helpful.

LordHultonus, 2016-09-23:
http://pastebin.com/DE849Z2g

Done, looks like the few last lines have somenthing to do with the adapter.

jurobystricky, 2016-09-23:
Seems there is a communication problem with the USB dongle.
Strange I did not see anything like:
New USB device found, idVendor=0846, idProduct=9053
...

Can you try to plug the dongle into another USB port? (Or a hub)

LordHultonus, 2016-09-23:
I did! 
Same thing. System starts to freeze and almost refuses to work completely. When I unplug the card, everything is back to normal. Plug it back - system gets retarded.

jurobystricky, 2016-09-23:
Can you try to disable USB 3.0 in BIOS, reboot and see what happens?

LordHultonus, 2016-09-23:
The information about adapter now takes whoooole dmesg. 

http://pastebin.com/Q568md7N

ulli-kroll, 2016-09-26:
FWIW
`[   15.716750] RTUSB_VendorRequest failed(-110)`
#define     ETIMEDOUT       110

LordHultonus, 2016-09-26:
The motherboard is Gigabyte GA-Z170-D3H. Im using USB ports on the front panel of my PC Case (Phaneks Enthoo Pro). On Windows everything is absolutely normal so its defenitely software problem. I can try plugging the card directly to the motherboard and check if anything changes. I dont think so though.

ulli-kroll, 2016-09-27:
I think the problem is not hardware, the driver is a PITA.
I know this from my work on mt7610u and mt7612u

The problem is inside rtusb_io.c
`RTUSB_CONTROL_MSG(pObj->pUsb_Dev, 0, Request, RequestType, Value,
						Index, pAd->UsbVendorReqBuf, TransferBufferLength,
						CONTROL_TIMEOUT_JIFFIES, RET);
`
which is a macro and calls at the end usb_control_msg() from the kernel.
The param CONTROL_TIMEOUT_JIFFIES is wrong
`#define CONTROL_TIMEOUT_JIFFIES ( (300 * OS_HZ) / 1000`
`#define OS_HZ                 HZ`
which must be
`@timeout: time in msecs to wait for the message to complete before timi
ng`
because HZ can change at build time.
This formula can evaluated (with HZ = 100)  to 30msec.
I would set this higher to 50 (or 100), this is is **my** guess

jurobystricky, 2016-09-27:
I agree, there should be no harm extending the time-out. At best it fixes the problem, worst case scenario is we get a time-out error message a bit later, most of the time there will be no change. (Prudent fix would be to modify the code and replace CONTROL_TIMEOUT_JIFFIES  with CONTROL_TIMEOUT_MS)

ulli-kroll, 2016-10-01:
I would prefer some higher value like 100ms or so.
The problem here if this device behind max. USB3 hubs.
In mt7601u this is set to
`drivers/net/wireless/mediatek/mt7601u/usb.h:22:#define MT_VEND_REQ_TOUT_MS      300`

The mt7612u driver (mine) is unstable in quality, it depends on the weather ...
Compared with mt7610u this driver is ancient

LordHultonus, 2016-11-06:
so I've tried to fix this. The system doesnt slow down or crash anymore but the adapter still doesnt work. It looks like Manjaro doesnt detect it and there are no networks visible.