jurobystricky/Netgear-A6210 #32: ARCH - Cannot make, errors

lell186, 2016-10-04:
Used:
Arch Linux 64 Bit with latest updates
Netgear A6210 (mt7612u)

I have been trying to build the new driver for the last couple of days (on and off) and keep coming up with this error. Anyone able to help? Do i need to downgrade my linux headers?

`cp -f os/linux/Makefile.6 /home/luke/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.7.6-1-ARCH/build DBGFLAGS=-DDBG SUBDIRS=/home/luke/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.7.6-1-ARCH/build'
  CC [M]  /home/luke/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from /home/luke/Netgear-A6210/include/os/rt_linux.h:77:0,
                 from /home/luke/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/luke/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/luke/Netgear-A6210/include/rt_config.h:34,
                 from /home/luke/Netgear-A6210/os/linux/../../sta/assoc.c:28:
/home/luke/Netgear-A6210/include/cfg80211.h:38:49: error: ‘IEEE80211_NUM_BANDS’ undeclared here (not in a function)
  struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
                                                 ^~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:290: /home/luke/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
make[1]: *** [Makefile:1458: _module_/home/luke/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.7.6-1-ARCH/build'
make: *** [Makefile:64: debug] Error 2
`

thomasluquet, 2016-10-04:
got the same PB

witi83, 2016-10-06:
https://github.com/jurobystricky/Netgear-A6210/issues/24

lell186, 2016-10-06:
Hi witi,

Apologies, I am a bit of a newbie when it comes to Linux. Can you show me where this needs to be patched into?
 
Thanks in advance!

witi83, 2016-10-07:
1. Open a shell and navigate to the main directory of this project.
2. Save this patch in #24 as e.g. `24.patch`
3. Apply it using `git apply 24.patch`

lell186, 2016-10-07:
I hate to be that person but i am getting stuck here too.. Here are my results:

I start by creating a new .patch file in my ~/Netgear-A6210/ directory
`touch 24.patch`
I then edit the file with nano
`nano 24.patch`
and paste in your segment of code (below) and save
```
diff --git a/include/cfg80211.h b/include/cfg80211.h
index 275cf39..a402a2a 100644
--- a/include/cfg80211.h
+++ b/include/cfg80211.h
@@ -34,6 +34,13 @@ typedef enum _NDIS_HOSTAPD_STATUS {

 typedef struct __CFG80211_CB {

+    #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 7, 0))
+    #define ieee80211_band nl80211_band
+    #define IEEE80211_BAND_2GHZ NL80211_BAND_2GHZ
+    #define IEEE80211_BAND_5GHZ NL80211_BAND_5GHZ
+    #define IEEE80211_NUM_BANDS NUM_NL80211_BANDS
+    #endif
+
        /* we can change channel/rate information on the fly so we backup them */
        struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
        struct ieee80211_channel *pCfg80211_Channels;
```

I then run the command `git apply -v 24.patch`

This is the output 

> Checking patch include/cfg80211.h...
> error: while searching for:
> hecking patch include/cfg80211.h...
> error: while searching for:
> 
> typedef struct __CFG80211_CB {
> 
>        /* we can change channel/rate information on the fly so we backup them */
>        struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
>        struct ieee80211_channel *pCfg80211_Channels;
> 
> error: patch failed: include/cfg80211.h:34
> error: include/cfg80211.h: patch does not apply

Thanks for all your help and sorry for being a pain..

jonipalosaari, 2016-10-07:
Is this some kind of linux support site? Every damn ubuntu/arch/some new cool distro guy register to github to make "issue" to this project? Driver works just fine and have been working for long time. Have to quit following this, cause this is ridiculous.

closed 2016-10-07 by lell186

blissrage, 2017-12-16:
i have the same issue as lell186 . . . i don't know the ins and outs of header files, makefile, etc. so dropping comment in hopes someone posts solution

xloem, 2021-04-15:
i don't know why jonipalosaari said that.  it is important to have drivers be compatible across distributions.  it is not a distribution support issue, but rather a software development issue.