jurobystricky/Netgear-A6210 #33: Any way to keep the driver persistent with bootup?

Hamblok0, 2016-10-05:
Still relatively new to linux, just curious if there's a way I can prevent having to run through the process of reinstalling the driver every time I boot to get my Netgear AC1200 to work each time. 

I typically have to run these commands in this order so that Ubuntu can see my wifi adapter, and therefore be able to connect to any wifi network: 

$ make
$ sudo make install
$ sudo modprobe mt7662u_sta
$ sudo service network-manager restart

Using Ubuntu 16.04

jurobystricky, 2016-10-05:
To build and install the driver you should need to run (preferably without the dongle inserted):

$ make
$ sudo make install

This needs to be done only once, unless you updated your kernel, in which case you need to rebuild the driver again with the new kernel header files.

The command 
$ sudo service network-manager restart

should be only needed after unplugging/re-plugging the dongle. It really should not be needed at all, but the network manager fails to detect a dongle being unplugged. 

In order to use the newly built driver you can:
A. Reboot with (or without) dongle inserted. (Insert the dongle after boot completed if booting w/o the dongle). This should be all that is needed.

use the newly built driver  without rebooting:

B. re-insert the dongle and run
$ sudo service network-manager restart

If A or B does not behave as it should, please provide a syslog or dmesg output

jadenPete, 2016-10-05:
In order to make the driver load on startup add mt7662u (or the module name) to /etc/modules.conf