jurobystricky/Netgear-A6210 #34: Working fine on CentOS 7 (4.8 kernel) but shutdown/restart hangs

MrTekNo, 2016-10-09:
Installed kernel 4.8 on CentOS 7 and run make (with the fix listed here for 4.7).

Driver is working properly but I can no longer do a restart or shutdown (need to press the power button).

When I do not insert the A6210 and boot I can properly shutdown/restart.

I can see a screen with 1 page of a dump of the driver but that's it (as the system stalled).

Anyway to fix this or maybe some hints on how to try to find out why this is happening?

Is there anyone else not able to shutdown when the A6210 is active?

Much appreciated.

MrTekNo, 2016-10-12:
Nobody having any problems (with any other Linux distro) when the A6210 is active and trying to shutdown or restart? No crashes?

gt945, 2016-11-05:
I can confirm this issue.
when shutdown, restart or unplug dongle, the kernel hangs.

funkey, 2016-12-06:
Same here. Ubuntu 16.10, kernel 4.8.0-30.

kirbyfan64, 2017-04-24:
Same exact issue here...

deepdumbo, 2018-07-19:
Same here. Ubuntu 16.04.4