jurobystricky/Netgear-A6210 #36: struct net_device’ has no member named ‘wireless_handlers

dbrazziel, 2016-10-20:
Build failure on kernel 4.7:

/opt/Netgear-WNDA3100v3/os/linux/../../os/linux/rt_main_dev.c: In function ‘rt28xx_open’:
/opt/Netgear-WNDA3100v3/os/linux/../../os/linux/rt_main_dev.c:288:11: error: ‘struct net_device’ has no member named ‘wireless_handlers’; did 
you mean ‘rx_handler’?
    net_dev->wireless_handlers = (struct iw_handler_def *) &rt28xx_iw_handler_def;

mehlis, 2016-12-19:
Support for Linux 4.7 was added with https://github.com/jurobystricky/Netgear-A6210/commit/f9266f154f24aa86b0a8cedcf6f65454579094dd . please retry.

lschneiderbauer, 2017-04-27:
just tried with linux 4.9.16; similar error in
./os/linux/rt_linux.c:1420:11: error: 'struct net_device' has no member named 'wireless_handlers'
pNetDev->wireless_handlers = pDevOpHook->iw_handler;
(up to typos, since copied by hand)

lschneiderbauer, 2017-04-29:
I fixed that error by enabling CONFIG_WIRELESS_EXT in the kernel. You cannot enable it directly though, you have to enable some wireless module which pulls CONFIG_WIRELESS_EXT in.
The module compiles now.