jurobystricky/Netgear-A6210 #37: Manjaro; can't see any wifi networks

HitCoder9768, 2016-10-25:
I have the driver installed, the computer can see the device is installed perfectly fine. I have tried an old version of this driver as well as the latest. I have also tried downgrading the built in network manager.
This used to work but recently I ran some updates on my system and it stopped working.
I'm happy to give any details if asked.

jonipalosaari, 2016-10-26:
"some updates".. kernel? You have reinstalled the driver?

HitCoder9768, 2016-10-26:
Yes. I have reinstalled the driver and I did around 200 application updates, and I think the kernel was also updated.
I tried an older version of the driver I had backed up, as well as the latest one from here.

CalebSmith376, 2016-10-29:
It looks like your adapter is in your 3.0 port and mine showed no networks when it was in the 3.0.

Try the 2.0 port and see if it works.

HitCoder9768, 2016-10-31:
No, I do not have ANY 3.0 ports on my machine. Thank you.

mcraveiro, 2016-11-03:
Hi @HitCoder9768, 

I seem to have similar symptoms to you. Can you please have a look at #39 and see if you got the same messages in dmesg? Also can you see any networks at all?

I'm also on 2.0 USB.

Cheers

Marco