jurobystricky/Netgear-A6210 #38: Not working with 3.0

CalebSmith376, 2016-10-29:
Hey my driver isnt working with my 3.0 port, it just freezes my pc after about 30 seconds.

I really need the 3.0 it runs too slow without it

CalebSmith376, 2016-10-29:
Sorry I was on mobile and I couldn't add much information, but let me elaborate on some things.

1) When I plug it in after boot, it says "device not managed" (I've tried the managed = true fix and it doesn't do anything).
2) When I plug it in prior to booting, it says "Disconnected" when I look at it when Ubuntu booted.

On Windows, I was getting about 50 Mb/s when I had it plugged into my USB 3.0 but getting around 100 Kb/s in a USB 2.0, which is exactly what's going on here, but the 3.0 isn't detecting any wireless connections.

Any ideas?

jordanbancino, 2017-07-20:
Interesting. My Netear A6210 works with my USB 3.0 ports just fine. Although, no matter which USB port I connect it to, I will get the "device not managed" warning, however, I can still connect to the internet. 

Will other devices (such as flash drives, keyboards, or mice) work with the USB 3.0 ports?