jurobystricky/Netgear-A6210 #39: Problems with Netgear WNDA3100v3 N600 WiFi USB Adapter

mcraveiro, 2016-10-31:
Hi,

thanks very much for contributing this driver. I recently bought a Netgear WNDA3100v3 N600 WiFi USB Adapter which I am trying to get to work. I cloned your repo at ```736f47b```, built the driver and installed it without any problems. I am on Debian Testing, kernel 4.7:

```
Linux lorenz 4.7.0-1-amd64 #1 SMP Debian 4.7.5-1 (2016-09-26) x86_64 GNU/Linux
```

I have restarted network manager as per instructions. When I plug in the device, the driver kicks in as expected and I see a list of Wireless networks to connect to. However, when I try to connect to my network I get an error:

```
Connection Failed: Activation of network connection failed
```

Interestingly enough, sometimes the list goes down to just one network, but most of the time it shows me quite a few networks. Not sure if this is relevant. At any rate, I looked at dmesg, and I see this repeated a couple of times every second (apologies for the large paste):

```
[1948357.442219] ERROR mt766u_sta:RTMPSetTimer failed, Halt in Progress!
[1948357.468970] andes_usb_erasefw
[1948357.469096] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
[1948357.473294] tx_kickout_fail_count = 0
[1948357.473296] tx_timeout_fail_count = 0
[1948357.473297] rx_receive_fail_count = 0
[1948357.473297] alloc_cmd_msg = 21
[1948357.473298] free_cmd_msg = 21
[1948357.527699] RTMP_TimerListRelease: release timer obj ffffc90003fc24b0!
[1948357.527702] RTMP_TimerListRelease: release timer obj ffffc90003fc2528!
[1948357.527703] RTMP_TimerListRelease: release timer obj ffffc90003fc25a0!
[1948357.527705] RTMP_TimerListRelease: release timer obj ffffc90003fc2438!
[1948357.527706] RTMP_TimerListRelease: release timer obj ffffc90003fc22d0!
[1948357.527707] RTMP_TimerListRelease: release timer obj ffffc90003fc2348!
[1948357.527708] RTMP_TimerListRelease: release timer obj ffffc90003f53fa0!
[1948357.527709] RTMP_TimerListRelease: release timer obj ffffc90003f42d78!
[1948357.527710] RTMP_TimerListRelease: release timer obj ffffc90003f42df8!
[1948357.527711] RTMP_TimerListRelease: release timer obj ffffc90003f54128!
[1948357.527712] RTMP_TimerListRelease: release timer obj ffffc90003f53eb0!
[1948357.527713] RTMP_TimerListRelease: release timer obj ffffc90003f540b0!
[1948357.538595] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 0, pAd->WlanFunCtrl:0x208, Reg-WlanFunCtrl=0x209
[1948357.539845] -->RTUSBVendorReset
[1948357.559813] <--RTUSBVendorReset
[1948357.560096] fw version:0.0.00 build:1
[1948357.560097] build time:201406241830____
[1948357.560101] fw for E3 IC
[1948357.560102] ilm length = 59312(bytes)
[1948357.560103] dlm length = 32068(bytes)
[1948357.560723] loading fw........
[1948357.621096] cfg_mode=5
[1948357.621197] Key1Str is Invalid key length(0) or Type(0)
[1948357.621205] Key2Str is Invalid key length(0) or Type(0)
[1948357.621213] Key3Str is Invalid key length(0) or Type(0)
[1948357.621221] Key4Str is Invalid key length(0) or Type(0)
[1948357.621403] USBAggregation = 1
[1948357.621405] 1. Phy Mode = 31
[1948357.621406] NVM is Efuse and its size =1d[1e0-1fc] 
[1948357.650223] get_chl_grp:illegal channel (167)
[1948357.650225] get_chl_grp:illegal channel (167)
[1948357.650226] get_chl_grp:illegal channel (169)
[1948357.650227] get_chl_grp:illegal channel (169)
[1948357.650228] get_chl_grp:illegal channel (171)
[1948357.650228] get_chl_grp:illegal channel (171)
[1948357.650229] get_chl_grp:illegal channel (173)
[1948357.650230] get_chl_grp:illegal channel (173)
[1948357.652847] Country Region from e2p = ffff
[1948357.652850] NICReadEEPROMParameters():Err! chip not support 5G band 0!
[1948357.652851] cfg_mode=9
[1948357.652937] RTMPSetPhyMode: channel is out of range, use first channel=1 
[1948357.652938] RTMPSetPhyMode: Update for STA
[1948357.669603] mt76x2_get_external_lna_gain::LNA type=0x11, BLNAGain=0x0, ALNAGain0=0x0, ALNAGain1=0x0, ALNAGain2=0x0
[1948357.669607] 2. Phy Mode = 14
[1948357.669608] 3. Phy Mode = 14
[1948357.669851] andes_usb_fw_init
[1948357.884482] AntCfgInit: primary/secondary ant 0/1
[1948357.884487] andes_load_cr:cr_type(2)
[1948357.885727] ChipStructAssign(): MT76x2 hook !
[1948357.885976] ---> InitFrequencyCalibration
[1948357.885978] InitFrequencyCalibrationMode:Unknow mode = 3
[1948357.885979] InitFrequencyCalibration: frequency offset in the EEPROM = 168
[1948357.885981] <--- InitFrequencyCalibration
[1948357.885988] RTMPSetPhyMode: Update for STA
[1948357.890229] MCS Set = ff ff 00 00 01
[1948357.890352] 80211> re-init bands...
[1948357.890354] 80211> CurTxPower = 20 dBm
[1948357.890357] ====> Radar Channel 52
[1948357.890358] ====> Radar Channel 54
[1948357.890359] ====> Radar Channel 56
[1948357.890361] ====> Radar Channel 60
[1948357.890362] ====> Radar Channel 62
[1948357.890372] ====> Radar Channel 64
[1948357.890373] ====> Radar Channel 100
[1948357.890374] ====> Radar Channel 104
[1948357.890376] 80211> TxStream = 2
[1948357.890381] Chan 167 (frq 5835):	not allowed!
[1948357.890382] Chan 169 (frq 5845):	not allowed!
[1948357.890384] Chan 171 (frq 5855):	not allowed!
[1948357.890385] Chan 173 (frq 5865):	not allowed!
[1948357.890386] Chan 184 (frq 4920):	not allowed!
[1948357.890388] Chan 188 (frq 4940):	not allowed!
[1948357.890389] Chan 192 (frq 4960):	not allowed!
[1948357.890390] Chan 196 (frq 4980):	not allowed!
[1948357.890392] Chan 208 (frq 6040):	not allowed!
[1948357.890393] Chan 212 (frq 6060):	not allowed!
[1948357.890394] Chan 216 (frq 6080):	not allowed!
[1948357.890396] RTMPDrvOpen(1):Check if PDMA is idle!
[1948357.890480] RTMPDrvOpen(2):Check if PDMA is idle!
[1948357.897530] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
[1948357.904979] TX0 power compensation = 0x38
[1948357.905227] TX1 power compensation = 0x38
```

Does this provide any clues as to what could be going wrong?

Many thanks for your time.

Marco

mcraveiro, 2016-11-03:
Interestingly enough, I just updated my kernel and when I try to build I now get:

```
cp -f os/linux/Makefile.6 /home/marco/Development/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.7.0-1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/home/marco/Development/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.7.0-1-amd64'
  CC [M]  /home/marco/Development/Netgear-A6210/os/linux/../../sta/assoc.o
/home/marco/Development/Netgear-A6210/os/linux/../../sta/assoc.c:1:0: error: code model kernel does not support PIC mode
...
```

This even after a clean. Going to try a fresh build...

mcraveiro, 2016-11-03:
Ok, no this seems to be the case now, can't build due to PIC. Kernel:

```
Linux lorenz 4.7.0-1-amd64 #1 SMP Debian 4.7.8-1 (2016-10-19) x86_64 GNU/Linux
```

Was fine on

```
Linux lorenz 4.7.0-1-amd64 #1 SMP Debian 4.7.5-1 (2016-09-26) x86_64 GNU/Linux

```

Investigating...

mcraveiro, 2016-11-03:
Seems like this is going to be a problem for everyone soon. According to [this](http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.8-rc2/0002-UBUNTU-SAUCE-no-up-disable-pie-when-gcc-has-it-enabl.patch) patch:

```
In Ubuntu 16.10, gcc's defaults have been set to build Position
Independent Executables (PIE) on amd64 and ppc64le (gcc was configured
this way for s390x in Ubuntu 16.04 LTS). This breaks the kernel build on
amd64. The following patch disables pie for x86 builds (though not yet
verified to work with gcc configured to build PIE by default i386 --
we're not planning to enable it for that architecture).
```

I tried to make sense of the ```-fno-pie``` option from that patch:

```
+KBUILD_CFLAGS += $(call cc-option, -fno-pie)
+KBUILD_CFLAGS += $(call cc-option, -no-pie)
+KBUILD_AFLAGS += $(call cc-option, -fno-pie)
+KBUILD_CPPFLAGS += $(call cc-option, -fno-pie)
```

But haven't quite managed to add it to the makefile just yet. Any suggestions are appreciated.

PandaHugMonster, 2016-11-12:
I have the same problem.

RTMPSetPhyMode: channel is out of range, use first channel=1 
[  374.930212] RTMPSetPhyMode: Update for STA
[  374.938535] mt76x2_get_external_lna_gain::LNA type=0x11, BLNAGain=0x0, ALNAGain0=0x0, ALNAGain1=0x0, ALNAGain2=0x0
[  374.938536] 2. Phy Mode = 14
[  374.938536] 3. Phy Mode = 14
[  374.938676] andes_usb_fw_init
[  375.123512] AntCfgInit: primary/secondary ant 0/1
[  375.123513] andes_load_cr:cr_type(2)
[  375.124146] ChipStructAssign(): MT76x2 hook !
[  375.124269] ---> InitFrequencyCalibration
[  375.124269] InitFrequencyCalibrationMode:Unknow mode = 3
[  375.124270] InitFrequencyCalibration: frequency offset in the EEPROM = 168
[  375.124270] <--- InitFrequencyCalibration
[  375.124273] RTMPSetPhyMode: Update for STA
[  375.126263] MCS Set = ff ff 00 00 01
[  375.126334] 80211> re-init bands...
[  375.126335] 80211> CurTxPower = 20 dBm
[  375.126335] ====> Radar Channel 52
[  375.126336] ====> Radar Channel 54
[  375.126336] ====> Radar Channel 56
[  375.126336] ====> Radar Channel 60
[  375.126336] ====> Radar Channel 62
[  375.126337] ====> Radar Channel 64
[  375.126337] ====> Radar Channel 100
[  375.126337] ====> Radar Channel 104
[  375.126338] 80211> TxStream = 2
[  375.126339] Chan 014 (frq 2484):	not allowed!
[  375.126340] Chan 149 (frq 5745):	not allowed!
[  375.126340] Chan 151 (frq 5755):	not allowed!
[  375.126340] Chan 153 (frq 5765):	not allowed!
[  375.126341] Chan 157 (frq 5785):	not allowed!
[  375.126341] Chan 159 (frq 5795):	not allowed!
[  375.126341] Chan 161 (frq 5805):	not allowed!
[  375.126341] Chan 165 (frq 5825):	not allowed!
[  375.126342] Chan 167 (frq 5835):	not allowed!
[  375.126342] Chan 169 (frq 5845):	not allowed!
[  375.126342] Chan 171 (frq 5855):	not allowed!
[  375.126342] Chan 173 (frq 5865):	not allowed!
[  375.126343] Chan 184 (frq 4920):	not allowed!
[  375.126343] Chan 188 (frq 4940):	not allowed!
[  375.126343] Chan 192 (frq 4960):	not allowed!
[  375.126344] Chan 196 (frq 4980):	not allowed!
[  375.126344] Chan 208 (frq 6040):	not allowed!
[  375.126344] Chan 212 (frq 6060):	not allowed!
[  375.126345] Chan 216 (frq 6080):	not allowed!
[  375.126345] RTMPDrvOpen(1):Check if PDMA is idle!
[  375.126399] RTMPDrvOpen(2):Check if PDMA is idle!
[  375.175817] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
[  375.183953] TX0 power compensation = 0x38
[  375.184078] TX1 power compensation = 0x38
[  375.185015] ERROR mt766u_sta:RTMPSetTimer failed, Halt in Progress!
[  375.206272] andes_usb_erasefw
[  375.206337] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
[  375.206596] receive cmd msg fail(-2)
[  375.206601] tx_kickout_fail_count = 0
[  375.206601] tx_timeout_fail_count = 0
[  375.206601] rx_receive_fail_count = 0
[  375.206602] alloc_cmd_msg = 22
[  375.206602] free_cmd_msg = 22
[  375.258734] RTMP_TimerListRelease: release timer obj ffffc90002fc7440!
[  375.258735] RTMP_TimerListRelease: release timer obj ffffc90002fc74b0!
[  375.258735] RTMP_TimerListRelease: release timer obj ffffc90002fc7520!
[  375.258736] RTMP_TimerListRelease: release timer obj ffffc90002fc73d0!
[  375.258736] RTMP_TimerListRelease: release timer obj ffffc90002fc7280!
[  375.258736] RTMP_TimerListRelease: release timer obj ffffc90002fc72f0!
[  375.258737] RTMP_TimerListRelease: release timer obj ffffc90002f58f70!
[  375.258737] RTMP_TimerListRelease: release timer obj ffffc90002f47d78!
[  375.258737] RTMP_TimerListRelease: release timer obj ffffc90002f47df0!
[  375.258738] RTMP_TimerListRelease: release timer obj ffffc90002f590e0!
[  375.258738] RTMP_TimerListRelease: release timer obj ffffc90002f58e90!
[  375.258739] RTMP_TimerListRelease: release timer obj ffffc90002f59070!
[  375.265761] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 0, pAd->WlanFunCtrl:0x208, Reg-WlanFunCtrl=0x209
[  375.266662] -->RTUSBVendorReset
[  375.286746] <--RTUSBVendorReset
[  375.286951] fw version:0.0.00 build:1
[  375.286952] build time:201406241830____
[  375.286954] fw for E3 IC
[  375.286955] ilm length = 59312(bytes)
[  375.286955] dlm length = 32068(bytes)
[  375.287484] loading fw........
[  375.343853] cfg_mode=5
[  375.343925] Key1Str is Invalid key length(0) or Type(0)
[  375.343930] Key2Str is Invalid key length(0) or Type(0)
[  375.343935] Key3Str is Invalid key length(0) or Type(0)
[  375.343941] Key4Str is Invalid key length(0) or Type(0)
[  375.344069] USBAggregation = 1
[  375.344070] 1. Phy Mode = 31
[  375.344071] NVM is Efuse and its size =1d[1e0-1fc] 
[  375.358456] get_chl_grp:illegal channel (167)
[  375.358457] get_chl_grp:illegal channel (167)
[  375.358457] get_chl_grp:illegal channel (169)
[  375.358458] get_chl_grp:illegal channel (169)
[  375.358458] get_chl_grp:illegal channel (171)
[  375.358458] get_chl_grp:illegal channel (171)
[  375.358458] get_chl_grp:illegal channel (173)
[  375.358459] get_chl_grp:illegal channel (173)
[  375.359941] Country Region from e2p = ffff
[  375.359943] NICReadEEPROMParameters():Err! chip not support 5G band 0!
[  375.359943] cfg_mode=9
[  375.360005] RTMPSetPhyMode: channel is out of range, use first channel=1 
[  375.360006] RTMPSetPhyMode: Update for STA
[  375.368316] mt76x2_get_external_lna_gain::LNA type=0x11, BLNAGain=0x0, ALNAGain0=0x0, ALNAGain1=0x0, ALNAGain2=0x0
[  375.368317] 2. Phy Mode = 14
[  375.368318] 3. Phy Mode = 14
[  375.368439] andes_usb_fw_init

Here is my uname: Linux taggerd 4.8.7-1-ARCH #1 SMP PREEMPT Thu Nov 10 17:22:48 CET 2016 x86_64 GNU/Linux

kcauchy, 2017-09-27:
Same Error.

kcauchy, 2017-09-27:
@mcraveiro The the workaround requires a recompiling for entire linux kernel?

What if I want just an update to this driver? 

Thanks.

kcauchy, 2017-09-27:
@mcraveiro Seems not my problem... Sorry for that.

My ubuntu is 16.04 LTS.

mcraveiro, 2017-09-27:
@kcauchy as I understand it, the problem is the kernel is compiled with one set of settings and the driver is not (i.e. driver is not PIC/PIE). I need to retry building at some point, but haven't had the time...