jurobystricky/Netgear-A6210 #40: 5GHz -25dBm Access Point not present

klugja, 2016-11-03:
From Windows, using Netgear Genie, I can see it and connect.  When I boot the same system into Linux (3.13.0-100-generic Ubuntu), it is not there.  Everything is in close proximity (within 2 meters).  I also have a non-PC Linux system, and it sees -25dBm for the access point signal.  It is an open access point.

lsusb:
Bus 003 Device 003: ID 0846:9014 NetGear, Inc.

This is a NetGear N600.  

Last change I have:
commit 736f47b9fa2f9e44f04e59487d0836d720890dfa (HEAD, origin/master, origin/HEAD, master)
Merge: 7c0dee7 f4da44e
Author: jurobystricky <juro.bystricky@intel.com>
Date:   Tue Oct 11 13:15:59 2016 -0700

    Merge pull request #35 from witi83/master
    
    Support for Kernel > 4.7 and <= 4.8.x

jadenPete, 2016-11-06:
I can't see 5gHz networks either

klugja, 2016-11-08:
There must be a compatibility issue, as I can see twenty-one 5GHz WiFi stations here, but not the closest one that is next to my PC.  I can see the same 5GHz WiFi with Windows, and ChromeOS.