jurobystricky/Netgear-A6210 #41: [Arch Linux] No network found

JacopoGobbi, 2016-11-06:
Hi, I followed the whole procedure, and tried different solutions but nothing works, the adapter is identified but I can't connect to any network:
[image]

Please let me know if you have any idea on what I could try.
Thanks

kaynz, 2016-11-09:
Same here (A6210)

kprkpr, 2016-11-13:
Same here (EDUP AC-1605)
From 4.8 doesnt works

d3xter, 2017-01-07:
Same here with 4.4.0-53 and Netgear A6210.

Output of dmesg is:
[  211.889370] -->RTUSBVendorReset
[  211.909343] <--RTUSBVendorReset
[  211.910615] ...........andes_usb_chk_crc
[  212.003477] andes_usb_reset_wmt
[  212.023850] -->RTUSBVendorReset
[  212.043818] <--RTUSBVendorReset
[  212.044223] fw version:0.0.00 build:1
[  212.044225] build time:201406241830____
[  212.044229] fw for E3 IC
[  212.044230] ilm length = 59312(bytes)
[  212.044230] dlm length = 32068(bytes)
[  212.044849] loading fw........
[  212.106143] cfg_mode=5
[  212.106462] Key1Str is Invalid key length(0) or Type(0)
[  212.106487] Key2Str is Invalid key length(0) or Type(0)
[  212.106517] Key3Str is Invalid key length(0) or Type(0)
[  212.106544] Key4Str is Invalid key length(0) or Type(0)
[  212.107097] USBAggregation = 1
[  212.107102] 1. Phy Mode = 31
[  212.107106] NVM is Efuse and its size =1d[1e0-1fc] 
[  212.136350] get_chl_grp:illegal channel (167)
[  212.136353] get_chl_grp:illegal channel (167)
[  212.136354] get_chl_grp:illegal channel (169)
[  212.136355] get_chl_grp:illegal channel (169)
[  212.136356] get_chl_grp:illegal channel (171)
[  212.136358] get_chl_grp:illegal channel (171)
[  212.136359] get_chl_grp:illegal channel (173)
[  212.136360] get_chl_grp:illegal channel (173)
[  212.138965] Country Region from e2p = ffff
[  212.151586] mt76x2_get_external_lna_gain::LNA type=0x0, BLNAGain=0xffffff8d, ALNAGain0=0xffffff89, ALNAGain1=0xffffff88, ALNAGain2=0xffffff87
[  212.151588] 2. Phy Mode = 31
[  212.151590] 3. Phy Mode = 31
[  212.151837] andes_usb_fw_init
[  212.327692] AntCfgInit: primary/secondary ant 0/1
[  212.327694] andes_load_cr:cr_type(2)
[  212.328940] ChipStructAssign(): MT76x2 hook !
[  212.329191] ---> InitFrequencyCalibration
[  212.329192] InitFrequencyCalibrationMode:Unknow mode = 3
[  212.329194] InitFrequencyCalibration: frequency offset in the EEPROM = 138
[  212.329195] <--- InitFrequencyCalibration
[  212.329203] RTMPSetPhyMode: channel is out of range, use first channel=1 
[  212.329207] RTMPSetPhyMode: Update for STA
[  212.333192] MCS Set = ff ff 00 00 01
[  212.333317] 80211> re-init bands...
[  212.333319] 80211> CurTxPower = 20 dBm
[  212.333321] ====> Radar Channel 52
[  212.333322] ====> Radar Channel 54
[  212.333323] ====> Radar Channel 56
[  212.333324] ====> Radar Channel 60
[  212.333325] ====> Radar Channel 62
[  212.333326] ====> Radar Channel 64
[  212.333327] ====> Radar Channel 100
[  212.333328] ====> Radar Channel 104
[  212.333330] 80211> TxStream = 2
[  212.333335] Chan 167 (frq 5835):	not allowed!
[  212.333336] Chan 169 (frq 5845):	not allowed!
[  212.333337] Chan 171 (frq 5855):	not allowed!
[  212.333338] Chan 173 (frq 5865):	not allowed!
[  212.333339] Chan 184 (frq 4920):	not allowed!
[  212.333340] Chan 188 (frq 4940):	not allowed!
[  212.333341] Chan 192 (frq 4960):	not allowed!
[  212.333342] Chan 196 (frq 4980):	not allowed!
[  212.333343] Chan 208 (frq 6040):	not allowed!
[  212.333344] Chan 212 (frq 6060):	not allowed!
[  212.333345] Chan 216 (frq 6080):	not allowed!
[  212.333346] RTMPDrvOpen(1):Check if PDMA is idle!
[  212.333444] RTMPDrvOpen(2):Check if PDMA is idle!
[... 71 lines cut ...]
[  217.597573] 80211> re-init bands...
[  217.597575] 80211> CurTxPower = 20 dBm
[  217.597578] ====> Radar Channel 52
[  217.597579] ====> Radar Channel 54
[  217.597580] ====> Radar Channel 56
[  217.597582] ====> Radar Channel 60
[  217.597583] ====> Radar Channel 62
[  217.597584] ====> Radar Channel 64
[  217.597585] ====> Radar Channel 100
[  217.597587] ====> Radar Channel 104
[  217.597589] 80211> TxStream = 2
[  217.597595] Chan 167 (frq 5835):	not allowed!
[  217.597596] Chan 169 (frq 5845):	not allowed!
[  217.597598] Chan 171 (frq 5855):	not allowed!
[  217.597599] Chan 173 (frq 5865):	not allowed!
[  217.597600] Chan 184 (frq 4920):	not allowed!
[  217.597602] Chan 188 (frq 4940):	not allowed!
[  217.597603] Chan 192 (frq 4960):	not allowed!
[  217.597604] Chan 196 (frq 4980):	not allowed!
[  217.597605] Chan 208 (frq 6040):	not allowed!
[  217.597607] Chan 212 (frq 6060):	not allowed!
[  217.597608] Chan 216 (frq 6080):	not allowed!
[  217.597610] RTMPDrvOpen(1):Check if PDMA is idle!
[  217.597701] RTMPDrvOpen(2):Check if PDMA is idle!
[  217.621443] 80211> CFG80211_OpsPmksaFlush ==>

Cube00, 2017-01-14:
Same here on Fedora 25, any updates?

jordanbancino, 2017-07-20:
This is also observed in Ubuntu 17.04. I use the Netgear A6210 also.