jurobystricky/Netgear-A6210 #43: Kali Linux 2016.2 using kali rolling repository. 'PIC MODE NOT SUPPORTED'

C0alDust, 2016-11-09:
Backstory:
Recently got kali and purchased a Netgear A6210 so i could use airmonitoring and do some wireless bits and bobs. Saw your driver and thought 'bingo', as previous attempts to get the adapter working had failed.

Issue:
first 2 command in the read me work fine. 
command 3: '$make'

returns an error, the error message is as follows
`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.8.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.8.0-kali1-amd64'
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/assoc.o
/root/Netgear-A6210/os/linux/../../sta/assoc.c:1:0: error: code model kernel does not support PIC mode
 /*
 
/usr/src/linux-headers-4.8.0-kali1-common/scripts/Makefile.build:294: recipe for target '/root/Netgear-A6210/os/linux/../../sta/assoc.o' failed
make[4]: *** [/root/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
/usr/src/linux-headers-4.8.0-kali1-common/Makefile:1488: recipe for target '_module_/root/Netgear-A6210/os/linux' failed
make[3]: *** [_module_/root/Netgear-A6210/os/linux] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.8.0-kali1-amd64'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2`

Running '$uname -a' returns:
`Linux kali 4.8.0-kali1-amd64 #1 SMP Debian 4.8.5-1kali1 (2016-11-04) x86_64 GNU/Linux`

Having a right headache getting kali to recognise the adapter.

Used your driver in Linux Mint and works a charm! no issues at all :)

Any help would be great,
Thanks,
CoalDust

Reikku, 2016-11-10:
Exactly the same issue for me with
uname -a
Linux kali 4.6.0-kali1-amd64 #1 SMP Debian 4.6.4-1kali1 (2016-07-21) x86_64 GNU/Linux

LIN-X42, 2016-11-12:
Mine can be build,But cannot switch to monitor mode

C0alDust, 2016-11-12:
What version of kali are you using?
type 'uname -a' in terminal and post the results up :)

mcraveiro, 2016-11-12:
Hi guys,

I'm investigating another problem but I've also hit the PIC issue at #39. If anyone has any kernel experience you may be able to make sense of my notes on that ticket, seems like we need to add some flags to the kernel module to build with PIC. Haven't quite figured out how to do this yet though :-(

Cheers

geeksmith, 2016-11-23:
On Debian stretch (Linux 4.8.0.1-amd64) I got past this issue by:

- patching /usr/src/linux-source-4.8/Makefile with the [patch you mentioned](http://kernel.ubuntu.com/%7Ekernel-ppa/mainline/v4.8-rc2/0002-UBUNTU-SAUCE-no-up-disable-pie-when-gcc-has-it-enabl.patch) [in issue #39](https://github.com/jurobystricky/Netgear-A6210/issues/39).
- changing the symlink in /lib/modules/($uname -r)/build to point to /usr/src/linux-source-4.8/ (this could alternatively be edited in Netgear-A6210/Makefile).

This allowed me to follow the regular, documented build process. I'm not sure how this could be done without editing the kernel Makefile, or in an easier way for the end user. Hopefully the gcc, kernel, and distro folks will work this one out sooner than later.

C0alDust, 2016-11-23:
@geeksmith 
Thanks for your response but i'm having trouble interpreting you solution to the issue.
If possible could you do a bit of a **dummies guide** to applying your methods of fixing the PIC issue, i'll admit this is the first time i have had to deal with makefiles and compiler related issues in depth. So a simpler step by step explanation on how to 'patch the makefile with the patch in #39' and / or 'changing the symlink in ...' would be fantastic! 
Many Thanks,
C0alDust

geeksmith, 2016-11-29:
@C0alDust 
The process is a little cumbersome for those not familiar with modifying and compiling source code with the standard *nix utilities. I'll do my best to break this down into steps that are easier to follow, though you'll possibly have to modify this for your kernel version and distro.

1. Download the kernel source for your distro. On a debian-based distro, you can do so with the command "apt-get source linux". Run this command in a directory where you intend to keep the kernel source code. The command will create a directory called "linux-source-X.Y" where X and Y are the major and minor version of your kernel.
2. Apply [this patch](http://kernel.ubuntu.com/%7Ekernel-ppa/mainline/v4.8-rc2/0002-UBUNTU-SAUCE-no-up-disable-pie-when-gcc-has-it-enabl.patch) to the kernel source. There are 2 options to complete this step
  2.1. Download the patch file and put it in the kernel source directory. Then run this command: `patch -p1 < 0002-UBUNTU-SAUCE-no-up-disable-pie-when-gcc-has-it-enabl.patch`
  2.2. You could directly edit the file named "Makefile" in the kernel source you downloaded, i.e. linux-source-X.Y/Makefile. In my Makefile, I inserted the following lines after linux 615, right after the line that says "all: vmlinux":
```
# force no-pie for distro compilers that enable pie by default
KBUILD_CFLAGS += $(call cc-option, -fno-pie)
KBUILD_CFLAGS += $(call cc-option, -no-pie)
KBUILD_AFLAGS += $(call cc-option, -fno-pie)
KBUILD_CPPFLAGS += $(call cc-option, -fno-pie)
```
3. You must now point the Netgear-A6210 driver code to this kernel source code directory. By default, the driver code points to /lib/modules/X.Y.Z-arch/build, which is a shortcut to another directory. Modify "~/linux-4.8.5" to the location of the directory containing the kernel source code you patched. It must be the full path to the directory in the last command:
```
sudo rm /lib/modules/$(uname -r)/build
sudo ln -sf ~/linux-4.8.5 /lib/modules/$(uname -r)/build
```

After this, you should be able to follow the README instructions for building the driver. I hope these more explicit instructions are helpful. If not, please post the errors you see.

Reikku, 2016-11-30:
Thank you @geeksmith for very detailed instructions.
Would you be kind and verify whether the line creating the link is correct?

:~/Netgear-A6210# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.8.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: Entering directory '/root/source/linux-4.8.5'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/Netgear-A6210/os/linux/../../sta/auth_rsp.o
/bin/sh: 1: ./scripts/recordmcount: not found
scripts/Makefile.build:294: recipe for target '/root/Netgear-A6210/os/linux/../../sta/auth_rsp.o' failed
make[2]: *** [/root/Netgear-A6210/os/linux/../../sta/auth_rsp.o] Error 127
Makefile:1492: recipe for target '_module_/root/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/root/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/root/source/linux-4.8.5'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

geeksmith, 2016-12-01:
I missed a step, thanks for catching that. I also had to copy the Module.symvers file into the kernel source directory. To get that file, I installed the linux-headers package for my distro and copied it from there into the source directory:
```
sudo apt-get install linux-headers-amd64
cd linux-source-X.Y
cp /usr/src/linux-headers-$(uname -r)/Module.symvers .
```

After completing this step, you should be able to go back to the Netgear directory and compile the module.

C0alDust, 2016-12-07:
Hi @geeksmith 
Sorry for the late responce.

Currently using Kali Linux 2016.2. As a consequence i'm having issues following your instructions.
When typing `apt-get source linux`

i get the following return
`E: You must put some 'source' URIs in your sources.list`

As such i can't continue with your instructions
CoalDust

geeksmith, 2016-12-08:
Take a look at the contents of /etc/apt/sources.list -- you'll see several lines similar to this, each line representing a remote package repository:
deb http://ftp.us.debian.org/debian/ testing main contrib non-free

To add a source URI for a repository, copy and paste the line, then replace the "deb" at the beginning of the new line with "deb-src", like this:
deb-src http://ftp.us.debian.org/debian/ testing main contrib non-free

After doing that for all the repositories in that file, save it and run "sudo apt-get update". After that finishes, you should be able to install the Linux kernel source code. Good luck!

C0alDust, 2016-12-08:
@geeksmith 

The following is the contents of sources.list

`# 

# deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

#deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

deb http://http.kali.org/kali kali-rolling main non-free contrib
# deb-src http://http.kali.org/kali kali-rolling main non-free contrib`

After uncommenting the deb-src and typing apt-get source linux

I get the error:
`E: You must put some 'source' URIs in your sources.list` again....

Kali is a bit of a quirky one :P

geeksmith, 2016-12-09:
Did you run "sudo apt-get update" first?

C0alDust, 2016-12-09:
Yep, ran that line and then apt-get source linux and same error as before. It may be because 2016.2 uses a rolling release system that's constantly updating.

geeksmith, 2016-12-10:
Because Kali is Debian-based, it follows the same rolling release system. The changes to sources.list should work. If I get time to install a Kali VM, I'll give this a try and let you know how to make it work. Just for completeness, will you please paste the contents of your /etc/apt/sources.list?

C0alDust, 2016-12-10:
# 

# deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

#deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

deb http://http.kali.org/kali kali-rolling main non-free contrib
#deb-src http://http.kali.org/kali kali-rolling main non-free contrib

geeksmith, 2016-12-10:
Was the hash in front of the deb-src line when you rand apt-get upgrade? The hash disables the line, so you'll have to remove it in order to enable the deb-src URI.

C0alDust, 2016-12-11:
@geeksmith 
Yes, i took the hash off that line as per your instruction earlier.. correction... *Literally running commands again now* after re-taking that hash off it seems to be working! So i now have the kernel source a folder named 'linux-4.8.11'

After running the copying the patch file in and patch the makefil successfully, then running the sudo rm... sudo ln...  and your extra bit for the module.symvers. After attempting to run the make command on the netgear folder i got an error that suggested i use 'make oldconfig && make prepare' on the kernel source to fix things. After running that on the folder containing the kernel code i get the following when running 'make' on the netgear folder.

`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Desktop/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.8.0-kali2-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Desktop/Netgear-A6210/os/linux modules
make[1]: Entering directory '/root/linux-4.8.11'
  CC [M]  /root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from ./include/linux/ktime.h:25:0,
                 from ./include/linux/rcupdate.h:47,
                 from ./include/linux/srcu.h:33,
                 from ./include/linux/notifier.h:15,
                 from ./include/linux/memory_hotplug.h:6,
                 from ./include/linux/mmzone.h:777,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /root/Desktop/Netgear-A6210/include/os/rt_linux.h:14,
                 from /root/Desktop/Netgear-A6210/include/rtmp_os.h:30,
                 from /root/Desktop/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/Desktop/Netgear-A6210/include/rt_config.h:34,
                 from /root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.c:28:
./include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
 #include <generated/timeconst.h>
                                 ^
compilation terminated.
scripts/Makefile.build:294: recipe for target '/root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o' failed
make[2]: *** [/root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
Makefile:1496: recipe for target '_module_/root/Desktop/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/root/Desktop/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/root/linux-4.8.11'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
`
Its a lot to read through but hopefully you can follow :)

Thanks!
C0alDust

geeksmith, 2016-12-12:
Try this while in the kernel source dir:
cp /boot/config-$(uname -r) .

This will eliminate the need for those two commands.

C0alDust, 2016-12-14:
@geeksmith 
Thanks so much for your help so far, but that last fix has hit a brick wall

typing `cp..... (uname -r)` into  prompt whilst in the kernel folder returns the following

```
root@kali:~# cp /boot/config-$(uname -r)
cp: missing destination file operand after '/boot/config-4.8.0-kali2-amd64'
Try 'cp --help' for more information.
```

Cheers,
C0alDust

geeksmith, 2016-12-14:
You missed the " ." at the end of the command I posted above.

C0alDust, 2016-12-15:
@geeksmith 
With the full stop the last command worked whilst inside the downloaded kernel folder. 

But when running `make` on the driver folder the same issue occurs

```
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Desktop/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.8.0-kali2-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Desktop/Netgear-A6210/os/linux modules
make[1]: Entering directory '/root/linux-4.8.11'
  CC [M]  /root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from ./include/linux/ktime.h:25:0,
                 from ./include/linux/rcupdate.h:47,
                 from ./include/linux/srcu.h:33,
                 from ./include/linux/notifier.h:15,
                 from ./include/linux/memory_hotplug.h:6,
                 from ./include/linux/mmzone.h:777,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /root/Desktop/Netgear-A6210/include/os/rt_linux.h:14,
                 from /root/Desktop/Netgear-A6210/include/rtmp_os.h:30,
                 from /root/Desktop/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/Desktop/Netgear-A6210/include/rt_config.h:34,
                 from /root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.c:28:
./include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
 #include <generated/timeconst.h>
                                 ^
compilation terminated.
scripts/Makefile.build:294: recipe for target '/root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o' failed
make[2]: *** [/root/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
Makefile:1496: recipe for target '_module_/root/Desktop/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/root/Desktop/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/root/linux-4.8.11'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
```

geeksmith, 2016-12-15:
Looks like the kernel source is missing a file that would normally be there after building the kernel. Try running make in the kernel source directory before building the driver. If it still fails, maybe somebody else here has run into this before.

C0alDust, 2016-12-15:
@geeksmith 
Running make on the kernel returns 
```
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/package.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
/bin/sh: 1: bc: not found
Kbuild:67: recipe for target 'include/generated/timeconst.h' failed
make[1]: *** [include/generated/timeconst.h] Error 127
Makefile:1027: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
'''

So same file still throwing a spanner in the works.

geeksmith, 2016-12-16:
You're missing some basic utilities required to build the kernel. Run this command and try again:
sudo apt-get build-dep linux

C0alDust, 2016-12-16:
After building the kernel again, then compiling the driver code it was working to some extend, was able to plug the adapter in and the system would recognise it but wouldn't find any networks. Reading geeksmiths previous post i then ran `apt-get build-dep linux` on the kernel folder, then re-built the kernel and recompiled the drivers. 

Restarting the VM prompted a rather large issue, the system would no longer boot. Going to old school black screen with command prompt stating: `Gave up waiting for root device`.

So a WARNING for all who attempt this ***be careful it may brick the install :)***

Going to reinstall kali then have another bash at it :P