jurobystricky/Netgear-A6210 #44: monitor mode support

bw100, 2016-11-16:
I have built and installed the driver onto  fedora 20 ( 3.19.8-100.fc20.x86_64) and ferdora 24 (4.8.6-201.fc24.x86_64) machines. I did not change any make or config files. 

I can not successfully capture packets using wireshark after setting the wlan interface into monitor mode. 

I appear to succesfully use iwconfig to set mode and channel 
     iwconfig wlan0 mode monitor
     iwconfig wlan0 channel 1

iwconfig out gives 

wlan0     Ralink STA  ESSID:""  Nickname:"mt7612u_sta"
          Mode:Monitor  Frequency=2.412 GHz  Access Point: Not-Associated   
          Bit Rate:1 Mb/s   
          RTS thr:off   Fragment thr:off
          Link Quality=10/100  Signal level:0 dBm  Noise level:0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

However , "iw dev" gives me 

phy#1
        Interface wlan0
                ifindex 6
                wdev 0x100000001
                addr b0:7f:b9:4d:d5:27
                type managed

i.e the hardware device is still in "managed" mode

when I try to use "iw" I either get "device or resource busy (-16)  or 

e.g  when trying to add an interface of type monitor
      iw phy phy1 interface add mon0 type monitor

i  get command not supported errors (-95) 

looking at output from "iw phy phy1 info"

        Supported interface modes:
                 * IBSS
                 * managed
                 * AP

and 

  software interface modes (can always be added): is empty

i.e no "monitor" mode support comparing to an interface I can successfully sniff with 
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
                 * P2P-device

       software interface modes (can always be added):
                 * AP/VLAN
                 * monitor

So, do I need to modify some config somewhere to enable monitor mode or am i just out of luck?

ITNano, 2018-06-26:
Bump. Anyone who have had any luck with this issue? I am experiencing the same behaviour. Running on Raspberry Pi 3B+, Raspbian stretch with the 4.14-fix by genodeftest.

vladdoster, 2019-03-19:
Any update on this?

ITNano, 2019-03-19:
I never got it to work. After giving up I instead went for a Panda PAU06 USB wireless card which worked flawlessly out-of-the-box with my setup (Raspbian installation) and seems to be tested with most major linux-based OSes. If you get permanently stuck like me and plan on changing USB wireless card I would highly recommend it. I'm leaving the link here for convenience: https://www.amazon.com/gp/aw/d/B00JDVRCI0/ .

vladdoster, 2019-03-20:
@ITNano Ah you read my mind. I ended up going with the [Panda Wireless PAU09 N600](https://www.amazon.com/gp/product/B01LY35HGO/). Thank you!