jurobystricky/Netgear-A6210 #45: compiled&installed without any errors but the device doesnt start!

mail2onur, 2016-11-16:
I recently bought EDUP EP-AC1601 wifi dongle and noticed that it doesn't work in Linux Mint 18, Kernel v4.4. This is such a shame because it performs so good in Win10 with excellent speed and quite high signal coverage.

I tried the driver from EDUP's website, Mediatek website and i have failed to compile these drivers as you can guess and then i found this github page while searching on Google. This driver is the only one which gets complied and installed without any errors but unfortunately i couldn't make the wifi dongle work no matter what i tried. This is so strange because i didn't see any error while installing this driver.

I tried resetting network manager via terminal command, remove&insert the usb dongle, restarting pc but none helps. I don't see any activity in network manager. I can see the device getting removed and inserted via lsusb and dpmesg commands but it doesnt call the necessary firmware from kernel to make it work.

Can anyone help me to get this work in Linux Mint 18 because i don't wanna return to Win10 just to be able to connect to the internet!

mail2onur, 2016-11-22:
I tried to install the driver again exactly as i did before and magically now i have my device listed in network manager after restart. Tough, it can only see 5ghz networks but none of the 2,4ghz signals. 

I cannot connect to my router`s 5ghz network either. The signal strength is abnormally low (around %9-10) and after few connection attempts, the signal strength drops to zero and the network disappears from the list. Also removing and reinserting the device sometimes causes entire system to freeze.

Tbh i dont know what to do at this point. I am out of ideas.

mail2onur, 2016-11-30:
I have read the readme files and i saw that it is possible to adjust TX and RX values from Linux terminal iwpriv command if it`s enabled in config.mk file before compiling it;
https://github.com/jurobystricky/Netgear-A6210/blob/master/doc/sta_ate_iwpriv_usage.txt

I have added necessary parameters ("HAS_ATE=y" and "HAS_QA_SUPPORT=y") into config.mk file but then i couldn't compile the driver due to an error. It says "no rule to make target" :/
```
CC [M]  /home/onur/Netgear-A6210-master/os/linux/../../ate/common/rt_ate.o
 
make[2]: *** No rule to make target '/home/onur/Netgear-A6210-master/os/linux/../../ate/common/rt_qa.o', needed by '/home/onur/Netgear-A6210-master/os/linux/mt7662u_sta.o'.  Stop.
Makefile:1396: recipe for target '_module_/home/onur/Netgear-A6210-master/os/linux' failed
make[1]: *** [_module_/home/onur/Netgear-A6210-master/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
```
Can anyone help me about this? or rather than trying to use iwpriv command, where are these RX&TX values are stored, in which file? Is it possible for me to tweak those without frying the chip?

LidiaGarrucho, 2017-01-22:
I have a Mediatek USB wifi with Vendor ID 0x0e8d and device ID 0x761b. I've been trying several drivers and none worked except this one, but with the same problem than you (extreme low received power). I manage to connect to the hotspot, even to ping it, but I loose 4/5 of all packets. As the received power is low, maybe there is a problem in the values that configure the AGC of the RX chain. I did a dmesg and it showed this interesting lines:

> [ 1345.195773] ERROR mt766u_sta:RTMPResumeMsduTransmission, R66CurrentValue=0...
[ 1345.197058] 80211> Connect bssid f4:b8:a7:f1:90:d4
[ 1345.236012] PeerBeaconAtJoinAction(): HT-CtrlChannel=6, CentralChannel=>6
[ 1345.236013] PeerBeaconAtJoinAction(): Set CentralChannel=6
[ 1345.236015] AdjustChannelRelatedValue(): Input BW=0, rf_channel=6, vht_bw=1, Channel=6, vht_cent_ch=6!
[ 1345.285188] mt76x2_reinit_agc_gain:original agc_vga0 = 0x4a, agc_vga1 = 0x4a
[ 1345.285195] mt76x2_reinit_agc_gain:updated agc_vga0 = 0x4a, agc_vga1 = 0x4a
[ 1345.285532] mt76x2_reinit_hi_lna_gain:original hi_lna0 = 0x33, hi_lna1 = 0x33
[ 1345.285536] mt76x2_reinit_hi_lna_gain:updated hi_lna0 = 0x33, hi_lna1 = 0x33
[ 1345.285964] original vga value(chain0) = 4a
[ 1345.286050] original vga value(chain1) = 4a
[ 1345.286357] mt76x2_calibration(channel = 6)

Is there a way to change the mt76x2_reinit_agc_gain value, or to force a default value higher than the showed up?