jurobystricky/Netgear-A6210 #54: Kernel 4.4.38 kernel Oops

gt945, 2016-12-29:
If i disable networkmanager. I can connect the AP by blow command
`wpa_supplicant -B -i wlan0 -c wpa_supplicant.conf -D nl80211
dhcpcd wlan0`

But, if I plugin the dongle and restart networkmanager, kernel oops 

uname -a
Linux Arch-C6A8BC 4.4.38-1-lts #1 SMP Sat Dec 10 20:21:58 CET 2016 x86_64 GNU/Linux

[  388.545428] 80211> CFG80211_OpsScan ==> wlan0(2)
[  388.545459] CFG80211DRV_IoctlHandle: 504a ffff8802475d7918 0
[  388.545468] CFG80211DRV_IoctlHandle: 504e           (null) 2
[  388.545475] 80211> ExtraIEs Null in ProbeRequest from upper layer...
[  388.545480] 80211> Wildcard SSID In ProbeRequest.
[  388.545492] CFG80211DRV_IoctlHandle: 5061 ffff88023223f400 2f
[  388.545583] SCANNING, suspend MSDU transmission ...
[  388.546269] bbp_set_bw(): Set PhyBW as 20MHz.l
[  388.546282] SYNC - BBP R4 to 20MHz.l
[  388.581330] wlp2s0: deauthenticating from 00:35:7a:00:00:c8 by local choice (Reason: 3=DEAUTH_LEAVING)
[  388.596848] CFG80211NetdevNotifierEvent ndev:ffff88024863e000  state:4
[  388.596856] CFG80211NetdevNotifierEvent default: [4]
[  388.600673] TX0 power compensation = 0x38
[  388.600775] TX1 power compensation = 0x38
[  388.601092] mt76x2_switch_channel(): Switch to Ch#1(2T2R), BBP_BW=0, bbp_ch_idx=0)
[  388.674215] CFG80211NetdevNotifierEvent ndev:ffff88024863e000  state:4
[  388.674218] CFG80211NetdevNotifierEvent default: [4]
[  388.674385] CFG80211NetdevNotifierEvent ndev:ffff88009e099000  state:9
[  388.674387] CFG80211NetdevNotifierEvent NETDEV_GOING_DOWN [9]
[  388.674389] CFG80211DRV_IoctlHandle: 5078 ffff88009e099000 9
[  388.674391] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
[  388.674392] CFG_SCAN: close the scan cmd in device close phase
[  388.674393] 80211> cfg80211_scan_done
[  388.674455] ===> rt28xx_close ffffc90001a01000
[  388.674456] ==> MlmeHalt
[  388.674462] <---MlmeThread
[  388.674471] --->Disable TSF synchronization
[  388.679784] <== MlmeHalt
[  388.679788] MacTableReset
[  388.679831] RT28xxUsbMlmeRadioOFF()
[  388.689936] --> RT65xxUsbAsicRadioOff
[  388.689938] ----> RT65xxDisableTxRx
[  388.689939] RT65xxDisableTxRx Tx success = 0
[  388.689940] RT65xxDisableTxRx Tx success = 9
[  388.689941] ====> StopDmaTx
[  388.691396] <==== StopDmaTx
[  388.691759] ERROR mt766u_sta:MlmeEnqueueForRecv(): fRTMP_ADAPTER_HALT_IN_PROGRESS
[  388.691796] ERROR mt766u_sta:MlmeEnqueueForRecv(): fRTMP_ADAPTER_HALT_IN_PROGRESS
[  388.691841] ERROR mt766u_sta:MlmeEnqueueForRecv(): fRTMP_ADAPTER_HALT_IN_PROGRESS
[  388.694458] ====> StopDmaRx
[  388.694498] <==== StopDmaRx
[  388.694499] andes_usb_erasefw
[  388.694539] ==>rlt_wlan_chip_onoff(): OnOff:0, Reset= 0, pAd->WlanFunCtrl:0x20b, Reg-WlanFunCtrl=0x20b
[  388.694578] WlanFunCtrl= 0x208
[  388.694675] <== rlt_wlan_chip_onoff():pAd->WlanFunCtrl.word = 0x208, Reg->WlanFunCtrl=0x208!
[  388.694677] <---- RT65xxDisableTxRx
[  388.694755] receive cmd msg fail(-2)
[  388.694760] tx_kickout_fail_count = 0
[  388.694760] tx_timeout_fail_count = 0
[  388.694761] rx_receive_fail_count = 0
[  388.694761] alloc_cmd_msg = 26
[  388.694762] free_cmd_msg = 26
[  388.694763] --->RTUSBCancelPendingBulkInIRP
[  388.694800] <---RTUSBCancelPendingBulkInIRP
[  388.694801] <== RT65xxUsbAsicRadioOff
[  388.745163] --->RTUSBCancelPendingBulkInIRP
[  388.745174] <---RTUSBCancelPendingBulkInIRP
[  388.745190] <---RTUSBCmdThread
[  388.745231] <---RtmpTimerQThread
[  388.745264] CFG80211DRV_IoctlHandle: 5073           (null) 0
[  388.745265]  CFG80211_reSetToDefault
[... 1707 lines cut ...]
			DBGPRINT(RT_DEBUG_TRACE, ("Install PTK: %d\n", IoctlSec.Alg));
   7c5c5:	e9 25 fb ff ff       	jmpq   7c0ef <CFG80211DRV_IoctlHandle+0xfff>
   7c5ca:	48 c7 c6 00 00 00 00 	mov    $0x0,%rsi
   7c5d1:	48 c7 c7 00 00 00 00 	mov    $0x0,%rdi
   7c5d8:	e8 00 00 00 00       	callq  7c5dd <CFG80211DRV_IoctlHandle+0x14ed>
			DBGPRINT(RT_DEBUG_TRACE, ("WPA\n"));
   7c5dd:	e9 92 f0 ff ff       	jmpq   7b674 <CFG80211DRV_IoctlHandle+0x584>
   7c5e2:	48 c7 c6 00 00 00 00 	mov    $0x0,%rsi
   7c5e9:	48 c7 c7 00 00 00 00 	mov    $0x0,%rdi
		memcpy(KeyBuf, pConnInfo->pKey, pConnInfo->KeyLen);
   7c5f0:	e8 00 00 00 00       	callq  7c5f5 <CFG80211DRV_IoctlHandle+0x1505>
   7c5f5:	41 80 bc 24 51 a3 13 	cmpb   $0x1,0x13a351(%r12)
   7c5fc:	00 01 
	DBGPRINT(RT_DEBUG_TRACE, (" %s\n", __FUNCTION__));
   7c5fe:	0f 85 99 ec ff ff    	jne    7b29d <CFG80211DRV_IoctlHandle+0x1ad>
   7c604:	83 3d 00 00 00 00 01 	cmpl   $0x1,0x0(%rip)        # 7c60b <CFG80211DRV_IoctlHandle+0x151b>
   7c60b:	0f 86 9a f1 ff ff    	jbe    7b7ab <CFG80211DRV_IoctlHandle+0x6bb>
   7c611:	48 83 3d 00 00 00 00 	cmpq   $0x0,0x0(%rip)        # 7c619 <CFG80211DRV_IoctlHandle+0x1529>
   7c618:	00 
			DBGPRINT(RT_DEBUG_ERROR,
   7c619:	0f 85 8c f1 ff ff    	jne    7b7ab <CFG80211DRV_IoctlHandle+0x6bb>
   7c61f:	48 c7 c7 00 00 00 00 	mov    $0x0,%rdi
			if (pAd->cfg80211_ctrl.FlgCfg80211Scanning == TRUE) {
   7c626:	e8 00 00 00 00       	callq  7c62b <CFG80211DRV_IoctlHandle+0x153b>
   7c62b:	e9 7b f1 ff ff       	jmpq   7b7ab <CFG80211DRV_IoctlHandle+0x6bb>

gt945, 2017-01-16:
finally, after do some workaround, this dongle works very well , no crash, no hang
1. use linux-lts  package in archlinux
2. disable networkmanager
3. connect to ap by manual
create  wpa_supplicant.conf and input ssid name and passphrase,
for example:
    `network={`
    `           ssid="SSID"`
    `            psk="qazxcdew"`
   ` }`
then 
`wpa_supplicant -B -i wlan0 -c wpa_supplicant.conf -D nl80211`
4. get ipaddress by manual
 `dhcpcd wlan0`

jdubzisfaded, 2017-03-07:
thank you sir. I think this will help much!