jurobystricky/Netgear-A6210 #55: No network found - Arch linux

confirmed12, 2017-01-07:
Hello I successfully builded upt the driver but when I check the network list it says not networks . I'm using Arch linux as well the Netgear WNDA3100v3 stick. I have tried restarting the network manager but still not networks. Someone help me with this?

LidiaGarrucho, 2017-01-25:
Can you try to place your router REALLY close to the computer, just to test if it is receiving or not? In my case I can receive with a AUKEY WF-R13 some networks but really really weak, maybe the gain of the receiving chain is wrongly set in the drivers.

flyingpies155, 2017-03-19:
I am having the same issue. When I was still setting Arch up and tried to use wifi-menu, it would immediately fail to connect. In Gnome, I have the same issue where it can't find network. If I try to run `sudo ip link set wlan0 up` I get an error saying `RTNETLINK answers: Operation not permitted`. I'm having this issue both on the normal Arch kernel and on the Linux-ck kernel (Currently running ck 4.10.4). Anyone have an ideas for how to solve this?

flyingpies155, 2017-03-25:
Update: I am now on Linux-ck 4.10.5, and it gets farther (`ip link set wlan0` up actually works now), but still doesn't find any networks. Here is the dmesg: http://pastebin.com/QGWn4zR0