jurobystricky/Netgear-A6210 #57: Driver fails to start with kernels 4.8, 4.9 and kernel hangs

MrEngman, 2017-01-10:
I am getting the following error when trying to run the driver on a Raspberry Pi B+ ARMv6 and the Pi appears to hang.

The error is identical on kernels 4.8.13, 4.9.0 and 4.9.2 other than the actual address values and the two lines
```
Jan 10 02:45:07 raspberrypi kernel: [   11.825112] ERROR mt7662u_sta:
Jan 10 02:45:07 raspberrypi kernel: [   11.825119] Failed to allocate memory - TxRxRing
```
appear as one with kernel 4.8.13.

`Jan 10 04:07:03 raspberrypi kernel: [   11.894239] ERROR mt7662u_sta:Failed to allocate memory - TxRxRing
`

Error on Raspberry Pi kernel 4.9.2, with 4.9.0 and 4.8.13 identical.
```
Jan 10 02:45:07 raspberrypi kernel: [    4.431064] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
Jan 10 02:45:07 raspberrypi kernel: [    4.572694] usb 1-1.2: New USB device found, idVendor=0b05, idProduct=180b
Jan 10 02:45:07 raspberrypi kernel: [    4.582793] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 10 02:45:07 raspberrypi kernel: [    4.593193] usb 1-1.2: Product: 802.11ac WLAN
Jan 10 02:45:07 raspberrypi kernel: [    4.600569] usb 1-1.2: Manufacturer: MediaTek Inc.
Jan 10 02:45:07 raspberrypi kernel: [    4.608335] usb 1-1.2: SerialNumber: 000000000
Jan 10 02:45:07 raspberrypi kernel: [    7.796949] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
Jan 10 02:45:07 raspberrypi kernel: [    8.534235] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Jan 10 02:45:07 raspberrypi kernel: [   11.611994] mt7662u_sta: loading out-of-tree module taints kernel.
Jan 10 02:45:07 raspberrypi kernel: [   11.823296] rtusb init rt2870 --->
Jan 10 02:45:07 raspberrypi kernel: [   11.824968]
Jan 10 02:45:07 raspberrypi kernel: [   11.824968] === pAd = dd24d000, size = 1181608 ===
Jan 10 02:45:07 raspberrypi kernel: [   11.824968]
Jan 10 02:45:07 raspberrypi kernel: [   11.824979] driver version: 3.0.0.1 (Jan 10 2017 00:35:34) .
Jan 10 02:45:07 raspberrypi kernel: [   11.825048] <-- ERROR in Alloc Bulk buffer for RxContext!
Jan 10 02:45:07 raspberrypi kernel: [   11.825054] ---> RTMPFreeTxRxRingMemory
Jan 10 02:45:07 raspberrypi kernel: [   11.825107] <--- RTMPFreeTxRxRingMemory
Jan 10 02:45:07 raspberrypi kernel: [   11.825112] ERROR mt7662u_sta:
Jan 10 02:45:07 raspberrypi kernel: [   11.825119] Failed to allocate memory - TxRxRing
Jan 10 02:45:07 raspberrypi kernel: [   11.825171] ------------[ cut here ]------------
Jan 10 02:45:07 raspberrypi kernel: [   11.846742] Modules linked in: mt7662u_sta(O+) cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem uio_pdrv_genir$
Jan 10 02:45:07 raspberrypi kernel: [   11.864945] CPU: 0 PID: 131 Comm: systemd-udevd Tainted: G           O    4.9.2+ #945
Jan 10 02:45:07 raspberrypi kernel: [   11.878466] Hardware name: BCM2835
Jan 10 02:45:07 raspberrypi kernel: [   11.884672] task: da5a0da0 task.stack: da6a6000
Jan 10 02:45:07 raspberrypi kernel: [   11.892053] PC is at kfree+0x144/0x18c
Jan 10 02:45:07 raspberrypi kernel: [   11.899696] LR is at os_free_mem+0x24/0x3c [mt7662u_sta]
Jan 10 02:45:07 raspberrypi kernel: [   11.907886] pc : [<c01342c4>]    lr : [<bf22299c>]    psr: 40000113
Jan 10 02:45:07 raspberrypi kernel: [   11.907886] sp : da6a7b68  ip : da6a7b90  fp : da6a7b8c
Jan 10 02:45:07 raspberrypi kernel: [   11.924889] r10: 0011fb24  r9 : 0011fb30  r8 : da6a7c0c
Jan 10 02:45:07 raspberrypi kernel: [   11.932875] r7 : bf29abf0  r6 : bf22299c  r5 : dd24d000  r4 : dd24d000
Jan 10 02:45:07 raspberrypi kernel: [   11.942175] r3 : db4292d4  r2 : 00000000  r1 : 00000000  r0 : dd24d000
Jan 10 02:45:07 raspberrypi kernel: [   11.951365] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jan 10 02:45:07 raspberrypi kernel: [   11.961243] Control: 00c5387d  Table: 1a6ac008  DAC: 00000055
Jan 10 02:45:07 raspberrypi kernel: [   12.566887] [<c01342c4>] (kfree) from [<bf22299c>] (os_free_mem+0x24/0x3c [mt7662u_sta])
Jan 10 02:45:07 raspberrypi kernel: [   12.584767] [<bf22299c>] (os_free_mem [mt7662u_sta]) from [<bf1e5954>] (RTMPAllocAdapterBlock+0xcc/0x300 [mt7662u_sta])
Jan 10 02:45:07 raspberrypi kernel: [   12.605438] [<bf1e5954>] (RTMPAllocAdapterBlock [mt7662u_sta]) from [<bf24fa9c>] (rtusb_probe+0x80/0x478 [mt7662u_sta])
Jan 10 02:45:07 raspberrypi kernel: [   12.625265] [<bf24fa9c>] (rtusb_probe [mt7662u_sta]) from [<c0416050>] (usb_probe_interface+0x164/0x2dc)
Jan 10 02:45:07 raspberrypi kernel: [   12.642712] [<c0416050>] (usb_probe_interface) from [<c03b0928>] (driver_probe_device+0x214/0x2c8)
Jan 10 02:45:07 raspberrypi kernel: [   12.659503] [<c03b0928>] (driver_probe_device) from [<c03b0a98>] (__driver_attach+0xbc/0xc0)
Jan 10 02:45:07 raspberrypi kernel: [   12.675756] [<c03b0a98>] (__driver_attach) from [<c03aea98>] (bus_for_each_dev+0x7c/0xb0)
Jan 10 02:45:07 raspberrypi kernel: [   12.691742] [<c03aea98>] (bus_for_each_dev) from [<c03b02b8>] (driver_attach+0x28/0x30)
Jan 10 02:45:07 raspberrypi kernel: [   12.707549] [<c03b02b8>] (driver_attach) from [<c03afe54>] (bus_add_driver+0x19c/0x224)
Jan 10 02:45:07 raspberrypi kernel: [   12.723351] [<c03afe54>] (bus_add_driver) from [<c03b1148>] (driver_register+0x88/0x108)
Jan 10 02:45:07 raspberrypi kernel: [   12.739259] [<c03b1148>] (driver_register) from [<c0414aa0>] (usb_register_driver+0x70/0x13c)
Jan 10 02:45:07 raspberrypi kernel: [   12.756692] [<c0414aa0>] (usb_register_driver) from [<bf2bd074>] (init_module+0x74/0xa0 [mt7662u_sta])
Jan 10 02:45:07 raspberrypi kernel: [   12.774986] [<bf2bd074>] (init_module [mt7662u_sta]) from [<c0009a1c>] (do_one_initcall+0x4c/0x188)
Jan 10 02:45:07 raspberrypi kernel: [   12.791982] [<c0009a1c>] (do_one_initcall) from [<c00e3c54>] (do_init_module+0x6c/0x1dc)
Jan 10 02:45:07 raspberrypi kernel: [   12.807916] [<c00e3c54>] (do_init_module) from [<c00836e0>] (load_module+0x1a64/0x1fb8)
Jan 10 02:45:07 raspberrypi kernel: [   12.823747] [<c00836e0>] (load_module) from [<c0083e0c>] (SyS_finit_module+0x9c/0xac)
Jan 10 02:45:07 raspberrypi kernel: [   12.839413] [<c0083e0c>] (SyS_finit_module) from [<c000fe20>] (ret_fast_syscall+0x0/0x1c)
Jan 10 02:45:07 raspberrypi kernel: [   12.865485] ---[ end trace bfd69d325eeef2e8 ]---
Jan 10 02:45:07 raspberrypi kernel: [   16.000652] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:217084k SSFS
Jan 10 02:45:08 raspberrypi kernel: [   16.684760] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Jan 10 02:45:08 raspberrypi kernel: [   16.700181] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Jan 10 02:45:27 raspberrypi kernel: [   36.270824] random: crng init done
```
[... 26 lines cut ...]
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/action.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_data.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/rtmp_init.o
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/rtmp_init.c: In function ‘RTMPAllocAdapterBlock’:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/rtmp_init.c:106:3: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘unsigned int’ [-Wformat=]
   DBGPRINT(RT_DEBUG_OFF, ("\n=== pAd = %p, size = %lu ===\n\n",
   ^
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/rtmp_init.c:110:10: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
  #pragma GCC diagnostic ignored "-Wdate-time"
          ^
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/rtmp_init_inf.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_tkip.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_aes.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_sync.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/eeprom.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_sanity.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_info.o
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_info.c: In function ‘Set_DriverVersion_Proc’:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_info.c:34:10: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
  #pragma GCC diagnostic ignored "-Wdate-time"
          ^
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_cfg.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_wpa.o

```

MrEngman, 2017-01-16:
Any ideas anyone? I would really like to fix this but my knowledge of C is almost none.

Can anyone suggest how I could go about debugging what the issue is?

kirbyfan64, 2017-04-24:
What happens if you remove every instance of `-DRESOURCE_PRE_ALLOC` from `os/linux/config.mk` and rebuild?

MrEngman, 2017-04-25:
The driver fails to compile with a load of errors.

I see the same warnings as in my first post but then further on a number of errors shown below and the compile fails.

```
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_p2p.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_ap.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211drv.o
  CC [M]  /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.o
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:599:2: error: #error or Here?
 #error or Here?
  ^
In file included from ./include/linux/string.h:18:0,
                 from ./include/linux/bitmap.h:8,
                 from ./include/linux/nodemask.h:94,
                 from ./include/linux/mmzone.h:16,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/kmod.h:22,
                 from ./include/linux/module.h:13,
                 from /home/richard/src-4.8/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/richard/src-4.8/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/richard/src-4.8/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/richard/src-4.8/Netgear-A6210/include/rt_config.h:34,
                 from /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:17:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c: In function ‘NICInitRecv’:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:667:36: error: ‘struct _CMD_RSP_CONTEXT’ has no member named ‘TransferBuffer’
  NdisZeroMemory(pCmdRspEventContext->TransferBuffer, CMD_RSP_BULK_SIZE);
                                    ^
./arch/arm/include/asm/string.h:36:17: note: in definition of macro ‘memset’
    void *__p = (p); size_t __n = n;   \
                 ^
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:667:2: note: in expansion of macro ‘NdisZeroMemory’
  NdisZeroMemory(pCmdRspEventContext->TransferBuffer, CMD_RSP_BULK_SIZE);
  ^
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c: In function ‘NICInitTransmit’:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:765:3: warning: passing argument 1 of ‘os_alloc_mem’ makes integer from pointer without a cast [enabled by default]
   os_alloc_mem(pAd, (PUCHAR *)(&pAd->MgmtDescRing.AllocVa), pAd->MgmtDescRing.AllocSize);
   ^
In file included from /home/richard/src-4.8/Netgear-A6210/include/rt_config.h:59:0,
                 from /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:17:
/home/richard/src-4.8/Netgear-A6210/include/rt_os_util.h:33:7: note: expected ‘size_t’ but argument is of type ‘PRTMP_ADAPTER’
 void *os_alloc_mem(size_t size);
       ^
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:765:3: error: too many arguments to function ‘os_alloc_mem’
   os_alloc_mem(pAd, (PUCHAR *)(&pAd->MgmtDescRing.AllocVa), pAd->MgmtDescRing.AllocSize);
   ^
In file included from /home/richard/src-4.8/Netgear-A6210/include/rt_config.h:59:0,
                 from /home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:17:
/home/richard/src-4.8/Netgear-A6210/include/rt_os_util.h:33:7: note: declared here
 void *os_alloc_mem(size_t size);
       ^
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c: In function ‘RTMPFreeTxRxRingMemory’:
/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.c:999:35: error: ‘struct _CMD_RSP_CONTEXT’ has no member named ‘TransferBuffer’
     (PUCHAR *)&pCmdRspEventContext->TransferBuffer,
                                   ^
scripts/Makefile.build:293: recipe for target '/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.o' failed
make[2]: *** [/home/richard/src-4.8/Netgear-A6210/os/linux/../../common/cmm_mac_usb.o] Error 1
Makefile:1490: recipe for target '_module_/home/richard/src-4.8/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/richard/src-4.8/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/home/richard/src/linux'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
```

DerRM, 2017-08-08:
I got it to start with setting the kernel parameter "coherent_pool=4M". But no luck with starting an AP yet.

AnonymousDapper, 2018-01-04:
I have this issue too. Running a Raspberry Pi 3 B+, kernel version `4.9.73-v7+`

```

[  257.803709] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[  257.935348] usb 1-1.3: New USB device found, idVendor=0846, idProduct=9053
[  257.935363] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  257.935371] usb 1-1.3: Product: A6210
[  257.935378] usb 1-1.3: Manufacturer: NETGEAR
[  257.935386] usb 1-1.3: SerialNumber: 100
[  259.078714] mt7662u_sta: loading out-of-tree module taints kernel.
[  259.093707] rtusb init rt2870 --->
[  259.095367]
=== pAd = b97ae000, size = 1184160 ===

[  259.095379] driver version: 3.0.0.1 (Jan  3 2018 18:34:51) .
[  259.095421] <-- ERROR in Alloc Bulk buffer for HTTxContext!
[  259.095426] ---> RTMPFreeTxRxRingMemory
[  259.095445] <--- RTMPFreeTxRxRingMemory
[  259.095449] ERROR mt766u_sta:
[  259.095455] Failed to allocate memory - TxRxRing
[  259.095476] ------------[ cut here ]------------
[  259.100332] kernel BUG at mm/slub.c:3873!
[  259.104512] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  259.109857] Modules linked in: mt7662u_sta(O+) bnep hci_uart btbcm bluetooth ip6table_filter ip6_tables ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack iptable_filter ip_tables x_tables brcmfmac brcmutil cfg80211 spidev rfkill snd_bcm2835 snd_pcm snd_timer i2c_bcm2835 snd bcm2835_gpiomem spi_bcm2835 uio_pdrv_genirq uio fixed i2c_dev fuse ipv6
[  259.148947] CPU: 1 PID: 1601 Comm: systemd-udevd Tainted: G           O    4.9.73-v7+ #1072
[  259.157632] Hardware name: BCM2835
[  259.161167] task: b5f66740 task.stack: b3732000
[  259.165883] PC is at kfree+0x1a8/0x1e8
[  259.170137] LR is at os_free_mem+0x24/0x3c [mt7662u_sta]
[  259.175665] pc : [<8025ff90>]    lr : [<7f4e9eec>]    psr: 40000013
sp : b3733b68  ip : 0020551e  fp : b3733b94
[  259.187597] r10: b97ae000  r9 : b3733c0c  r8 : b98ce000
[  259.192959] r7 : 7f5619b4  r6 : b97ae000  r5 : 7f4e9eec  r4 : b97ae000
[  259.199664] r3 : b782b478  r2 : 00000000  r1 : 00000000  r0 : b97ae000
[  259.206369] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  259.213696] Control: 10c5383d  Table: 2a84006a  DAC: 00000055
[  259.219593] Process systemd-udevd (pid: 1601, stack limit = 0xb3732210)
[  259.226380] Stack: (0xb3733b68 to 0xb3734000)
[  259.230850] 3b60:                   801722ec 80171ac8 b97ae000 b344d200 00000003 7f5619b4
[  259.239249] 3b80: b98ce000 b3733c0c b3733bac b3733b98 7f4e9eec 8025fdf4 00000006 b98cd000
[  259.247646] 3ba0: b3733bdc b3733bb0 7f4acb8c 7f4e9ed4 00000000 b344de00 b3493070 b3493000
[  259.260430] 3bc0: 7f5619ec 7f5619b4 b3493000 b35f5c00 b3733c84 b3733be0 7f516c74 7f4ac8f8
[  259.273229] 3be0: b3733c0c b3733bf0 802eb63c 80718ef8 00000002 00000000 aaa18af0 805063f4
[  259.286058] 3c00: b34930dc 7f561a64 802eb778 00000000 60000013 60000013 00000004 b35f5c84
[  259.298871] 3c20: 60000013 60000013 00000000 b3493070 00000000 b35f5c84 b3733c5c b3733c48
[  259.311675] 3c40: 8071bd84 801e3b18 00000081 b35f5c20 b3733c84 b3733c60 80506830 ffffffed
[  259.324467] 3c60: b3493070 b3493000 7f5619ec b35f5c00 7f561a64 b35f5c20 b3733cbc b3733c88
[  259.337328] 3c80: 805611c8 7f516bf8 802ee00c 802edf0c b3733cbc 80d0e4a8 b35f5c20 00000000
[  259.350273] 3ca0: 7f5619ec 00000017 7f561b40 00000000 b3733ce4 b3733cc0 804fc1f0 805610c0
[  259.363287] 3cc0: 00000000 b35f5c20 7f5619ec b35f5c54 00000000 7f539d74 b3733d04 b3733ce8
[  259.376402] 3ce0: 804fc358 804fbfdc 804fc29c 00000000 7f5619ec 804fc29c b3733d2c b3733d08
[  259.389611] 3d00: 804fa1cc 804fc2a8 b6a7715c b62f5a34 b6a77170 7f5619ec b5d5c900 80c5c3f0
[  259.402863] 3d20: b3733d3c b3733d30 804fbb70 804fa15c b3733d64 b3733d40 804fb708 804fbb50
[  259.416126] 3d40: 7f539d1c b3733d50 7f5619ec 00000000 80c5c3f0 7f5619ec b3733d7c b3733d68
[  259.429411] 3d60: 804fca50 804fb568 7f5619b8 00000000 b3733da4 b3733d80 8055fc40 804fc9d4
[  259.442644] 3d80: 7f562d00 7f58e000 b62cd100 7f561b40 00000001 49d32e5c b3733dbc b3733da8
[  259.455868] 3da0: 7f58e09c 8055fbcc 00000000 7f561b40 b3733e3c b3733dc0 80101bf0 7f58e00c
[  259.469091] 3dc0: 00000000 00000000 b3733e14 b3733dd8 80213e88 36465000 8024f0fc b3529780
[  259.482345] 3de0: 0002930d 8025ec0c b3733e3c b3733df8 8025ec0c 80717854 8025ffcc 8025f470
[  259.495597] 3e00: 00000001 0002930d b62cd140 a0000013 b9657000 7f561b40 00000001 b62cd100
[  259.508824] 3e20: 7f561b40 00000001 49d32e5c b62cd1c8 b3733e64 b3733e40 8020a548 80101bac
[  259.522066] 3e40: b3733e64 b3733e50 8024f1f0 b3733f3c 00000001 b62cd1c0 b3733f14 b3733e68
[  259.535301] 3e60: 801a39ac 8020a4e0 7f561b4c 00007fff 7f561b40 801a1090 b3733ea4 7f561d2c
[  259.548534] 3e80: b973d384 7f5b4c7c 7f562d08 00000002 7f561c54 7f561b88 b3733ee4 b3733ea8
[  259.561765] 3ea0: 80277480 80277300 b9657000 00000000 00000000 00000000 00000000 00000000
[  259.574992] 3ec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  259.588245] 3ee0: 00000000 00000000 7fffffff 00000000 00000007 76d94004 0000017b 7fffffff
[  259.601491] 3f00: 00000000 00000000 b3733fa4 b3733f18 801a42c4 801a1fec 7fffffff 00000000
[  259.614757] 3f20: 00000003 8022d908 b3733f74 b9657000 00155ac0 00000000 00000002 b9657000
[... 1 lines cut ...]
[  259.641246] 3f60: 00000000 00026c7c 0000002d 0000002e 00000023 00000027 00000017 00000000
[  259.654506] 3f80: 0000017b 0048104c 00000000 00020000 80108244 b3732000 00000000 b3733fa8
[... cut ...]
[  259.877215] [<7f58e09c>] (init_module [mt7662u_sta]) from [<80101bf0>] (do_one_initcall+0x50/0x184)
[  259.891448] [<80101bf0>] (do_one_initcall) from [<8020a548>] (do_init_module+0x74/0x1e4)
[  259.904694] [<8020a548>] (do_init_module) from [<801a39ac>] (load_module+0x19cc/0x2108)
[  259.917848] [<801a39ac>] (load_module) from [<801a42c4>] (SyS_finit_module+0x9c/0xac)
[  259.930824] [<801a42c4>] (SyS_finit_module) from [<801080c0>] (ret_fast_syscall+0x0/0x1c)
[  259.944155] Code: 1a000003 e5932014 e3120001 1a000000 (e7f001f2)
[  259.952862] ---[ end trace f815979b8f7f43b8 ]---
```

MrEngman, 2018-01-04:
Edit file `/boot/cmdline.txt` and add `coherent_pool=4M` to the end of the file.