jurobystricky/Netgear-A6210 #58: Make - missing makefile in build folder

DANHDAN, 2017-01-12:
Hello, 
I could do with some guidance if you wouldn't mind please. I am stuck with not being able to "make". I have installed headers and just checked my build folder but there is no makefile file.

Current errors:

dan@PI:$ cd Netgear-A6210
dan@PI:/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/dan/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.4.41-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/dan/Netgear-A6210/os/linux modules
make[1]: Entering directory '/lib/modules/4.4.41-v7+/build'
make[1]: *** No rule to make target 'modules'. Stop.
make[1]: Leaving directory '/lib/modules/4.4.41-v7+/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
dan@PI:~/Netgear-A6210$

Many thanks in advance

pdktdk, 2017-01-17:
Delete the netgear folder and follow all defined steps from the beginning.

$ git clone https://github.com/jurobystricky/Netgear-A6210
$ cd Netgear-A6210
$ make
$ sudo make install

DANHDAN, 2017-01-17:
Thank you, I will try this when back from work. I'm not sure about this though as I have tried the same steps on 2 different builds, Ubuntu mate 16.04 and now on Raspbian Jessie (latest version). I have had to create the build directory manually but then it tells me that there are no instructions on how to use modules.

pdktdk, 2017-01-17:
If you delete the file and run the git command the make file will definitely be present.  I had to delete the folder for each kernel upgrade to install the driver for the same reason.  Good luck!

DANHDAN, 2017-01-17:
Great thanks. I look forward to trying this later. Should it create the build folder automatically with the makefile file inside?

pdktdk, 2017-01-17:
You should be able to run the commands listed above in sequence and it will do the rest.  

Have your adapter unplugged. Run the commands.   When you finish shut down.  Plug in the adapter.  Then boot. login.  Select network.  Let me know how it goes.

DANHDAN, 2017-01-17:
OK thanks will do. I've not tried deleting like you've said before so i'll let you know.

DANHDAN, 2017-01-17:
Same thing, i've spent a lot of time trying to get this to work :(. Any ideas would be appreciated?

pi@Pi:~/Netgear-A6210 $ sudo make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.4.41-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/pi/Netgear-A6210/os/linux modules
make[1]: Entering directory '/lib/modules/4.4.41-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.4.41-v7+/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

pdktdk, 2017-01-17:
Delete the folder again.

after you "cd"
it's just "make" not "sudo make"
then "sudo make install"

shut the computer off.  Plug in the adapter and login.

DANHDAN, 2017-01-17:
I tried just "make" first before "Sudo" make.  There is a makefile in the netgear-a6210/os/linux but it is looking into the /lib/modules folder which are headers I installed i believe. Could that be an issue?

i'll delete and retry again aswell, thanks for you help with this.

DANHDAN, 2017-01-17:
Same thing. Do you think it has something to do with my modules folder - /lib/modules/4.4.41-v7+/build ?

pi@Pi:~/Netgear-A6210 $ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.4.41-v7+/build DBGFLAGS=-DDBG SUBDIRS=/home/pi/Netgear-A6210/os/linux modules
make[1]: Entering directory '/lib/modules/4.4.41-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.4.41-v7+/build'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

pdktdk, 2017-01-17:
Try this:

sudo apt-get install linux-headers-$(uname -r)

then
prepare-kernel-sources
cd /usr/src/linux
cp -rf include/generated/* include/linux/

pdktdk, 2017-01-17:
Try this:

sudo apt-get install linux-headers-$(uname -r)

then 
Code:
prepare-kernel-sources
cd /usr/src/linux
cp -rf include/generated/* include/linux/

DANHDAN, 2017-01-17:
Thanks it looks like i've got the latest but the prepare command dosn't work, maybe i'm not doing it correctly. Should I uninstall and retry?

Thanks

pi@Pi:~ $ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-4.4.41-v7+ is already the newest version.
The following packages were automatically installed and are no longer required:
  libboost-filesystem1.55.0 libboost-program-options1.55.0
  libboost-regex1.55.0 libcwiid1 libjs-prettify libllvm3.7 libqscintilla2-11
  libqscintilla2-l10n libqt4-network libqt4-xmlpatterns libqtwebkit4
  libruby1.9.1 libruby1.9.1-dbg libtcl8.5 libtcltk-ruby1.9.1 libtk8.5 ri1.9.1
  ruby1.9.1 ruby1.9.1-dev ruby1.9.1-examples ruby1.9.1-full ruby1.9.3
  supercollider supercollider-common supercollider-ide supercollider-language
  supercollider-supernova tcl8.5 tk8.5
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 42 not upgraded.
pi@Pi:~ $ sudo prepare-kernel-sources
sudo: prepare-kernel-sources: command not found

DANHDAN, 2017-01-17:
I have been able to run the last 2 commands which created those files but Terminal is telling me that it is looking at /lib rather than /usr. I tried running the command in the /lib/modules/4.4.41-v7+/build but i get cp: cannot stat ‘include/generated/*’: No such file or directory

:(

DANHDAN, 2017-01-18:
OK so now I have managed to get it compiled by changing the Kernel address in the Makefile HOWEVER after a reboot and connecting the adapter, it still won't work.

Any more suggestions please?

1mlewis, 2017-02-25:
Just want to note that I used this to get an ASUS USB-AC55R (0b05:17eb) working on a fresh install of raspbian jesse (4.4.38-v7+) on a raspberry pi 3. Thanks for this, it worked great and I couldn't find anything to get this device up and running a few months ago.

Steps I used:

sudo apt-get install git raspberrypi-kernel-headers
git clone https://github.com/jurobystricky/Netgear-A6210.git
cd Netgear-A6210
make
sudo make install