jurobystricky/Netgear-A6210 #67: Not working with ASUS USB-AC55 - RTUSB_VendorRequest failed

itoytman, 2017-04-23:
Hi,
I am trying to get my ASUS USB-AC55 stick work under Linux Mint installed in VirtualBox (I am planning to also install the same on a standalone system later), which is essentially Ubuntu 4.4. I have the driver downloaded and compiled with no issues. However, if I plug in the stick I get this message:
`ModemManager[2870]: segfault at 0 ip 0000000000431ab3 sp 00007ffee05e6430 error 4 in ModemManager[400000+103000]`
Then I restart network manager:
`service network-manager restart`
Here is what I get in dmesg after that:
`
[ 3050.984854] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 3050.986155] -->RTUSBVendorReset
[ 3051.006227] <--RTUSBVendorReset
[ 3051.026498] .....RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x9a8,pAd->Flags=0x0
[ 3057.152658] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x6, Idx=0x9a8,pAd->Flags=0x0
[ 3057.152667] VendrCmdMultiWrite_nBytes failed!
[ 3060.242339] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=OUT, Req=0x42, Idx=0x230,pAd->Flags=0x0
[ 3060.242349] andes_usb_load_rom_patch: set fce dma descriptor fail (0x230)
[ 3060.242358] ERROR mt766u_sta:load patch failed, Status[=0x00000001]
[ 3060.243347] receive cmd msg fail(-2)
[ 3060.243488] tx_kickout_fail_count = 0
[ 3060.243494] tx_timeout_fail_count = 0
[ 3060.243498] rx_receive_fail_count = 0
[ 3060.243502] alloc_cmd_msg = 1
[ 3060.243506] free_cmd_msg = 1
[ 3060.243511] !!! rt28xx init fail !!!
[ 3063.439748] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x7, Idx=0x1000,pAd->Flags=0x0
[ 3066.481429] RTUSB_VendorRequest failed(-110),TxFlags=0x0, ReqType=IN, Req=0x47, Idx=0x148,pAd->Flags=0x0
`
Here is the output of lsusb:
`
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0b05:17eb ASUSTek Computer, Inc. 
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
Here is the output of uname -a:
`Linux ilya-VirtualBox 4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

How do I make it work?

itoytman, 2017-07-04:
Quick update - tried on recent Fedora (physically installed on a desktop PC), and not working either:
dmesg output:
```
[50690.607195] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
[50690.630444] TX0 power compensation = 0x38
[50690.631193] TX1 power compensation = 0x38
[50690.636770] ERROR mt766u_sta:
[50690.636772] RTMPSetTimer failed, Halt in Progress!

```
Fedora version: 4.11.8-200.fc25.x86_64

genodeftest, 2017-08-27:
@itoytman : Your first comment looks like the NetworkManager bug described in #73.