jurobystricky/Netgear-A6210 #70: EDUP AC1606 in Ubuntu 16.04 2 LTS

cyfex, 2017-06-07:
The driver is easily and successfully installed. But the wireless signal is really weak. It takes tens of seconds to open a Google page. The LED light is on.

cyfex, 2017-06-07:
The result of "iwlist wlan0 scanning"

`wlan0     Scan completed :
          Cell 01 - Address: F4:83:CD:89:7F:89
                    Protocol:802.11a/n
                    ESSID:"TP-LINK_5G_7F87"
                    Mode:Managed
                    Frequency:5.745 GHz
                    Quality=10/100  Signal level=-86 dBm  Noise level=-88 dBm
                    Encryption key:on
                    Bit Rates:150 Mb/s
                    IE: WPA Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
`

EserGozcu, 2017-06-10:
I have exactly the same issue. Same dongle works perfect on WIndows but signal level is very low on Ubuntu 16.04    I tried other wifi adapters and even the cheapest ones in the market can get a higher signal level. 

on Netgear for the same SSID: 

                    Frequency:2.437 GHz (Channel 6)
                    Quality=0/100  Signal level=-96 dBm  Noise level=-105 dBm

on Edimax: 

                    Frequency:2.437 GHz (Channel 6)
                    Quality=44/100  Signal level=-72 dBm  Noise level=-73 dBm