jurobystricky/Netgear-A6210 #72: Add support to Kernel 4.10 / Ubuntu 17.04

nicolasf, 2017-06-26:
After updating to Ubuntu 17.04 the driver is not behaving correctly.

network-manager or other process seems to go into loop (as seen on /var/log/syslog) and network list sometimes appear on the tray icon, but it keeps changing states.

If there are instructions on how to debug/fix this I can try to look a little, but I'm not experienced in driver development.

nicolasf, 2017-06-26:
`[  155.309316] usb 4-5: new SuperSpeed USB device number 4 using xhci_hcd
[  155.330229] usb 4-5: New USB device found, idVendor=0846, idProduct=9053
[  155.330232] usb 4-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  155.330234] usb 4-5: Product: A6210
[  155.330236] usb 4-5: Manufacturer: NETGEAR
[  155.330237] usb 4-5: SerialNumber: 100
[  155.331761] 
               === pAd = ffffa8fe473c9000, size = 1282584 ===

[  155.331763] driver version: 3.0.0.1 (Jun 25 2017 23:17:15) .
[  155.332318] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
[  155.339093] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[  155.339094] RtmpEepromGetDefault::e2p_dafault=1
[  155.339266] NVM is EFUSE mode
[  155.339268] Endpoint(8) is for In-band Command
[  155.339269] Endpoint(4) is for WMM0 AC0
[  155.339270] Endpoint(5) is for WMM0 AC1
[  155.339270] Endpoint(6) is for WMM0 AC2
[  155.339271] Endpoint(7) is for WMM0 AC3
[  155.339272] Endpoint(9) is for WMM1 AC0
[  155.339273] Endpoint(84) is for Data-In
[  155.339274] Endpoint(85) is for Command Rsp
[  155.339383] 80211> CurTxPower = 20 dBm
[  155.339385] ====> Radar Channel 52
[  155.339386] ====> Radar Channel 54
[  155.339387] ====> Radar Channel 56
[  155.339387] ====> Radar Channel 60
[  155.339388] ====> Radar Channel 62
[  155.339389] ====> Radar Channel 64
[  155.339389] ====> Radar Channel 100
[  155.339390] ====> Radar Channel 104
[  155.339391] 80211> TxStream = 0
[  155.361439] ModemManager[1201]: segfault at 0 ip 00005572c909cab7 sp 00007ffe3a706140 error 4 in ModemManager[5572c9065000+11f000]
`

genodeftest, 2017-08-27:
Looks like a duplicate of #73, but with less details.