jurobystricky/Netgear-A6210 #75: Compile issue with Kali for Pi3

davetayl, 2017-07-26:
I am having some issues with compiling the driver on Kali linux, any help would be greatly appreciated.

I have loaded up gcc, make and the available header files but keep getting an error early on in the compile.

The image is the Raspberry Pi 2/3 image from here https://www.offensive-security.com/kali-linux-arm-images/

root@Remote01:~/Downloads/Netgear-A6210# uname -a
Linux Remote01 4.4.50-v7 #1 SMP Fri Apr 21 01:18:29 CDT 2017 armv7l GNU/Linux

----------------------------------------------------------------------------------------------------------
root@Remote01:~/Downloads/Netgear-A6210# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Downloads/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.4.50-v7/build DBGFLAGS=-DDBG SUBDIRS=/root/Downloads/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/Downloads/Netgear-A6210/os/linux/../../sta/assoc.o
/bin/sh: 1: ./scripts/recordmcount: Exec format error
scripts/Makefile.build:258: recipe for target '/root/Downloads/Netgear-A6210/os/linux/../../sta/assoc.o' failed
make[2]: *** [/root/Downloads/Netgear-A6210/os/linux/../../sta/assoc.o] Error 2
Makefile:1402: recipe for target '_module_/root/Downloads/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/root/Downloads/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/kernel'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
--------------------------------------------------------------------------------------------------------------

tf0054, 2017-08-04:
Just for information, but my A6210 is working on Pi2 with the latest src https://github.com/jurobystricky/Netgear-A6210/commit/8874a06b7214fd75e70e5c8692f5f430fd185b42

```
pi@raspberrypi:~/tmp/Netgear-A6210 $ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~/tmp/Netgear-A6210 $ iwconfig wlan1
wlan1     Ralink STA  ESSID:"_test_"  Nickname:"mt7612u_sta"
          Mode:Managed  Frequency=5.18 GHz  Access Point: A0:63:91:D5:25:BA
          Bit Rate=135 Mb/s
          RTS thr:off   Fragment thr:off
          Link Quality=76/100  Signal level:-85 dBm  Noise level:-92 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

pi@raspberrypi:~/tmp/Netgear-A6210 $
```
I just followed this info.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=132089#p880942

genodeftest, 2017-08-27:
@davetayl It looks like you are compiling the kernel module as root. This is a bad (i.e. risky) idea. Can you try checking the code out again and run `make` as ordinary user (only `make install` as root)?

genodeftest, 2017-08-27:
@davetayl : Firstly, please do not compile anything as root, as you do not need to be root to compile.
Secondly, you need to install kernel headers. Try something like ` apt-get install -y linux-headers-$(uname -r)`.

Tuxprogrammer, 2017-10-24:
Do the following on the Kali raspberrypi image:
`cd /usr/src/kernel/`
`make scripts`

IDK how they packaged kernel headers, but they compiled the scripts for the wrong architecture.

genodeftest, 2017-10-24:
@Tuxprogrammer : Is there a bug report in Kali to fix this issue on their side?

Tuxprogrammer, 2017-10-24:
@genodeftest I just looked, and I can't find anything about it on the Kali Tracker. Feel free to post a report, I've never done one before so I don't know the format.