jurobystricky/Netgear-A6210 #78: Compilation problem with ArchLinux

Ayaash, 2017-08-30:
Hi,

When I execute the "make" command, I obtain an "incompatible-pointer-type error : 

```
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/ayaash/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.12.8-2-ARCH/build DBGFLAGS=-DDBG SUBDIRS=/home/ayaash/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.12.8-2-ARCH/build'
  CC [M]  /home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
/home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = CFG80211_OpsVirtualInfChg,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: note: (near initialization for ‘CFG80211_Ops.change_virtual_intf’)
/home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = CFG80211_OpsVirtualInfAdd,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: note: (near initialization for ‘CFG80211_Ops.add_virtual_intf’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:303: /home/ayaash/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
make[1]: *** [Makefile:1515: _module_/home/ayaash/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.12.8-2-ARCH/build'
make: *** [Makefile:64: debug] Error 2
```

I tried to find the script compiling cfg80211.c to remove the -Werror flag, but I didn't succeed. I also tried to compile cfg80211.c manually, but I didn't find the rightful GCC command.

I've spend quite time on it, so if you could help me !

Thanks, at least for reading,
Ayaash

genodeftest, 2017-08-31:
Hi @Ayaash, can you try the changes I suggested in #77? I haven't tested these on hardware, but at least they made the module build again. Please report back if they make your hardware work, preferably as a comment to #77.

Ayaash, 2017-08-31:
Works for me ! Thanks a lot @genodeftest !

closed 2017-08-31 by Ayaash