jurobystricky/Netgear-A6210 #81: problem with 4.13 linux

kudlatyamroth, 2017-09-29:
`make -C /lib/modules/4.13.3-1-ARCH/build DBGFLAGS=-DDBG SUBDIRS=/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.13.3-1-ARCH/build'
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/assoc.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/auth.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/auth_rsp.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/sanity.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/sync.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/rtmp_data.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/connect.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/wpa.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/sta_cfg.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../sta/sta.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/sta_ioctl.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../mgmt/mgmt_vht.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/vht.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/crypt_md5.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/crypt_sha2.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/crypt_hmac.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/crypt_aes.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../common/crypt_arc4.o
In file included from ./include/linux/bitmap.h:8:0,
                 from ./include/linux/cpumask.h:11,
                 from ./arch/x86/include/asm/cpumask.h:4,
                 from ./arch/x86/include/asm/msr.h:10,
                 from ./arch/x86/include/asm/processor.h:20,
                 from ./arch/x86/include/asm/cpufeature.h:4,
                 from ./arch/x86/include/asm/thread_info.h:52,
                 from ./include/linux/thread_info.h:37,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:80,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /var/lib/dkms/netgear-a6210/2.5.0/build/include/os/rt_linux.h:14,
                 from /var/lib/dkms/netgear-a6210/2.5.0/build/include/rtmp_os.h:30,
                 from /var/lib/dkms/netgear-a6210/2.5.0/build/include/rtmp_comm.h:64,
                 from /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:305:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:302: /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1512: _module_/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.13.3-1-ARCH/build'
make: *** [Makefile:64: debug] Error 2
`

navrit, 2017-10-01:
I also get this on 4.13.0-041300-generic

ageaenes, 2017-10-01:
+1, 4.13.3-1-ARCH

yourepitaph, 2017-10-02:
I get the exact same error on the latest kernel of Arch
Please fix!!!!!

genodeftest, 2017-10-02:
@yourepitaph wrote:
> I get the exact same error on the latest kernel of Arch
Please fix!!!!!

You could be a bit more polite and a bit less demanding. This is an open source community driven by volunteers spending spare time to develop software. Nobody's being paid for working here.

genodeftest, 2017-10-02:
This issue is a result of the new config option `CONFIG_FORTIFY_SOURCE`, which acts like glibc’s `FORTIFY_SOURCE` in trying to detect some bugs with memcpy at compile-time.
As a very dirty workaround, you could disable that config option.

See https://outflux.net/blog/archives/2017/09/05/security-things-in-linux-v4-13/ for an explanation and a few patches in different parts of the kernel source tree.

genodeftest, 2017-10-02:
#83 contains a fix for the bug reported above. Please test and report back. Note that you still have to apply the fixes in #77 to make the kernel compile or check out my [port-to-4.13 branch](https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13).
Note that you will still see a warning during compilation which is potentially serious (see #82).

You mig

kudlatyamroth, 2017-10-02:
it looks like working :)
no problem with connection after compiling it :)
thx

mattbell87, 2017-10-27:
@genodeftest thanks! Your port-to-4.13 branch works perfectly. I'm running Ubuntu 17.10.

firemaricius, 2017-10-31:
Damn i really got my hopes up when i saw the fix for 4.13 but on Arch linux i still get the error, even when using @genodeftest port-to-413 branch, any suggestions? im on 4.13.9-1-ARCH 

In function 'memcpy',
    inlined from 'rt_ioctl_iwaplist' at /home/henz/Projects/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:305:4: error: call to '__read_overflow2' declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();

genodeftest, 2017-11-01:
@firemaricius 
First suggestion: Please put code or terminal output in code blocks. See the "Styling with Markdown is supported" link below the comment box for details.

To all of you:
I guess that Arch (and maybe other distros too) use different (more pedantic) options to build the kernel and its modules, helping us to find programming mistakes. I am not sure whether this is correct, but you can try this (and please report back whether it worked):
In `os/linux/sta_ioctl.c:549`, replace `addr[0]` with `struct sockaddr` so the line looks like this
```
	memcpy(extra, addr, i*sizeof(struct sockaddr));
```
instead of
```
	memcpy(extra, addr, i*sizeof(addr[0]));
```

firemaricius, 2017-11-01:
It was actually fixed on arch as well with your solution,  I had just cloned your master branch and not the actually fix, so i just manually changed the one character and it compiled just fine.

Thanks a lot for all the work tho the wifi adapter works like a charm now

tkasozi, 2017-11-10:
@genodeftest  Thanks a lot for the work!! compiled well, problem fixed! You guys save my life everyday. I just have to know where to look. THANKS!