jurobystricky/Netgear-A6210 #84: Ubuntu 16.04.3: System Program Problem Detected.

jordanbancino, 2017-10-10:
While the driver itself works perfectly fine on Ubuntu 16.04 (Linux 4.10.0-35), I get this annoying error every time I turn on my PC: 

    System Program Problem Detected
    Do you want to report the problem now?
    Cancel                        Report Problem...

This dialog appears about 6 times right after a boot and log in, and a 7th time a few minutes later. Then, after this, it goes away until I reboot again. I used the Netgear-A6210 with these drivers, and it worked just fine, but now, I use a wired connection, so I uninstalled the drivers because I don't need them. I used this command: `sudo make uninstall` from the root of the source tree. It gave no errors, yet I still get the `system program problem detected` error every time I turn on my computer. 

To clarify, I never received this error before I installed the drivers either. This issue also effects my laptop, which is running the identical setup (Ubuntu 16.04.3 - Linux 4.10.0-35). 

Perhaps I did not uninstall the drivers correctly? Or the uninstaller leaves some stuff behind?

Really, there are 2 issues here:
1. The driver gives a system error at boot
2. The drivers do not seem to be fully removed after a `sudo make uninstall`, as the error still occurs.
    (unless the drivers are causing permanent damage to the system, where a reinstall may be required?)

I've never thought to keep system logs / reports of my system until I opened this issue, but I will try to post some when I can.

genodeftest, 2017-10-10:
Please report the bug using the "Report Problem..." link. There is much more info needed, e.g. the output of `dmesg` and probably some more debugging stuff. Most modern linux distributions provide means to report problems like these in a reasonable way. If you do so and it generates a report, that data should be linked here.

About the second issue: Did you unload the driver using `modprobe -r` or by rebooting? Kernel modules are not being removed unless one of these happens. Rebooting is the better option, because if the module has a bug, it might taint the kernel so it runs into problems somewhere else.

jordanbancino, 2017-10-11:
@genodeftest 

I used the `Report Problem...` link. All it did was crash `sudo` and `evolution-addressbook-factory`. Not sure if any of these are useful, but I was able to get screenshots of the data it showed:
[image]
[image]

This is the output of `dmesg`, right after boot: 
[dmesg.txt](https://github.com/jurobystricky/Netgear-A6210/files/1376915/dmesg.txt)

`/var/log/boot.log` contains this: 

    [  OK  ] Started LSB: Set the CPU Frequency Scaling governor to "ondemand".

And `/var/log/dmesg` contains:

    (Nothing has been logged yet.)

The output of `journalctl -b` is here:
    
[journalctl-b.txt](https://github.com/jurobystricky/Netgear-A6210/files/1376927/journalctl-b.txt)

About running `modprobe -r`, I am not sure what modules to remove. `lsmod` does not show anything that seems to be related to this driver. I've rebooted many times, and the issue only occurs right after a reboot.

Hope this is enough information?

genodeftest, 2017-10-11:
The crashes in sudo and evolution-addressbook-factory are unrelated, please report them to Ubuntu (launchpad.net).

According to the logs (both journalctl and dmesg), there is no note of this kernel module being loaded. Whatever the problem is you are seeing it is entirely unrelated to the Netgear-A6210 kernel module.

genodeftest, 2017-10-11:
Maybe if you used `sudo make`, it could have broken your system. In that case it is still not the fault of this kernel module, since one MUST NEVER compile code as root. I don't know how to resolve this issue though.

jordanbancino, 2017-10-11:
Interesting, as this issue started only started after I installed the driver, and continued after it was uninstalled. 

I know I did not compile as root, as I only used `sudo` for `make install`. I guess I'll just live with the pop-up at boot, as I've got quite a bit of data that I don't want to lose by re-installing.

closed 2017-10-11 by jordanbancino

genodeftest, 2017-10-11:
If your installation was properly configured (`/home` on a separate partition), you won't loose any data.

jordanbancino, 2017-10-11:
Unfortunately, I used the default install options, which did not set my system up this way. Everything is on the same partition.

yjlintw, 2018-03-06:
I encounter the same problem after I install this driver, too.