jurobystricky/Netgear-A6210 #85: Kali Linux Make failed

JockerZeus, 2017-10-15:
Hello, 
when I do "make" it shows me a mistake, how can I do it?

# uname -a
4.13.0-kali1-amd64 #1 SMP Debian 4.13.4-1kali1 (2017-10-03) x86_64 GNU/Linux

#make 
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.13.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.13.0-kali1-amd64/build : Aucun fichier ou dossier de ce type. Arrêt.
Makefile:59 : la recette pour la cible « debug » a échouée
make: *** [debug] Erreur 2
zsh: exit 2     make

thank you for your answers.

genodeftest, 2017-10-15:
Do you have linux kernel headers installed? Do you have the patches installed required for 4.14 (#77 and #83 or try my [port-to-4.13 branch](https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13))?

JockerZeus, 2017-10-16:
Hello,

I have not install linux kernel. I'm new to linux, I do not know how to install it. Is this tuto good?
https://korben.info/installer-noyau-linux-kernel-facilement.html 

thank you for your answers.

genodeftest, 2017-10-17:
> I have not install linux kernel. I'm new to linux, I do not know how to install it. Is this tuto good?
https://korben.info/installer-noyau-linux-kernel-facilement.html

I don't know, but it does not look good to me. You should rather use the official documentation to your linux distribution which probably covers those standard tasks. I don't use Kali linux, but maybe running `apt-get install kernel-headers-$(uname -r)` as root is enough.

JockerZeus, 2017-10-18:
hello it's fix. I have install Ubuntu. thx

genodeftest, 2017-10-18:
@JockerZeus wrote:
> hello it's fix. I have install Ubuntu. thx

Then please close this issue.

hunatae, 2017-12-19:
is this the same issue? this is what i get...

Cloning into 'Netgear-A6210'...
remote: Counting objects: 2436, done.
remote: Total 2436 (delta 0), reused 0 (delta 0), pack-reused 2436
Receiving objects: 100% (2436/2436), 3.71 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (1512/1512), done.
root@kali:~# cd Netgear-A6210
root@kali:~/Netgear-A6210# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.9.0-kali3-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.9.0-kali3-amd64/build: No such file or directory. Stop.
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
root@kali:~/Netgear-A6210# sudo make install
make -C /root/Netgear-A6210/os/linux -f Makefile.6 install
make[1]: Entering directory '/root/Netgear-A6210/os/linux'
rm -rf /etc/Wireless/RT2870STA
mkdir -pv /etc/Wireless/RT2870STA
mkdir: created directory '/etc/Wireless/RT2870STA'
cp /root/Netgear-A6210/conf/RT2870STA.dat /etc/Wireless/RT2870STA/.
install -d /lib/modules/4.9.0-kali3-amd64/kernel/drivers/net/wireless/
install -m 644 -c mt7662u_sta.ko /lib/modules/4.9.0-kali3-amd64/kernel/drivers/net/wireless/
install: cannot stat 'mt7662u_sta.ko': No such file or directory
Makefile.6:721: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/root/Netgear-A6210/os/linux'
Makefile:76: recipe for target 'install' failed
make: *** [install] Error 2

genodeftest, 2017-12-20:
@hunatae : No. The issue you are running into is caused by the fact that you ran `make` as root. Please never ever do that. You might have broken your linux installation beyond any repair. In case you see strange errors in the next days, I suggest you reinstall your linux distribution.

You need to run all the commands from Readme.md as ordinary user.

For future notice: Your error message does not have anything in common with the one posted above, so please do not hijack this issue. In case you are running into a different problem, please make sure to open a separate issue. You probably should read and consider https://www.chiark.greenend.org.uk/~sgtatham/bugs.html.

hunatae, 2017-12-28:
I'm new to linux, as you could have guessed. i'm running live with persistence. how and where would i run the commands from readme.md?

genodeftest, 2018-01-01:
>  i'm running live with persistence

I don't know what that should be.

> how and where would i run the commands from readme.md?

In a terminal, as ordinary user, **NOT AS ROOT** and **NOT WITH SUDO**

hunatae, 2018-01-09:
tried as user still didnt work.

dendo@kali:~/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/dendo/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.9.0-kali3-amd64/build DBGFLAGS=-DDBG SUBDIRS=/home/dendo/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.9.0-kali3-amd64/build: No such file or directory.  Stop.
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

genodeftest, 2018-01-13:
I guess running make as root f***ed up your filesystem. Try removing the `Netgear-A6210` folder and start over (see `README.md` for a guide how to start).
`make` really should not be trying to write anything into `/lib/modules/*` and any access to that folder must fail when running as ordinary user. Instead, it should create a build folder somewhere below your `Netgear-A6210`-folder.

In case that does not help, are you sure you have installed kernel headers?
If you have installed kernel headers too, you probably need to format and reinstall your linux because running `make` as `root` broke it hard.

hunatae, 2018-01-23:
in the source folder downloaded from git clone https://github.com/jurobystricky/Netgear-A6210, i checked the makefile.
LINUX_SRC = /lib/modules/$(shell uname -r)/build
LINUX_SRC_MODULE = /lib/modules/$(shell uname -r)/kernel/drivers/net/wireless/

so yeah....if it's not suppose to write there, then all the source code is wrong.............?

genodeftest, 2018-01-24:
From the two lines of code you quoted (use Code syntax, if you want anyone to read it, please), there is nothing that indicates a write operation to that folder. It uses these folders (of course), but only for reading.

hunatae, 2018-01-30:
is there an administrator on these forums that i can talk to please?

genodeftest, 2018-01-30:
@hunatae : The maintainer of this repository is @jurobystricky, but I haven't heard from him in a while. If you have technical questions, you should post them here. If there is something wrong, you should use the report functionality.

hunatae, 2018-02-01:
it's just  that you're not giving any answers, rather saying i'm posting wrong and i broke my linux.
i know it's something simple to get this driver installed, i just can't figure out what it is.

genodeftest, 2018-02-01:
> it's just that you're not giving any answers, rather saying i'm posting wrong and i broke my linux.

Hey, I'm trying. And I'm mostly out of ideas. And yes, you might have broke your linux. 

Have you installed kernel headers? The package might be named `kernel-headers` and the version must match your installed kernel exactly.

hunatae, 2018-02-01:
shouldn't everything be included when i installed kali? i'm currently researching how to install headers so i'll check.

hunatae, 2018-02-01:
also, thank you for at least trying. no one else no matter where i posted for help answered. So i do appreciate that

hunatae, 2018-02-01:
all i know for sure is that i installed kali live onto a usb drive using yumi. live with persistence did not work, so i figured out I had to create and mount a new partition on the usb drive which now allows persistence to work. that alone took me some time to accomplish.

hunatae, 2018-02-07:
every header command i try to run gives me this.

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-compiler-gcc-6-x86 linux-headers-4.9.0-kali3-common linux-kbuild-4.9
The following NEW packages will be installed:
  linux-compiler-gcc-6-x86 linux-headers-4.9.0-kali3-amd64
  linux-headers-4.9.0-kali3-common linux-kbuild-4.9
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 8,999 kB of archives.
After this operation, 51.3 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://http.kali.org/kali kali-rolling/main amd64 linux-compiler-gcc-6-x86 amd64 4.9.18-1kali1
  404  Not Found
Err:2 http://http.kali.org/kali kali-rolling/main amd64 linux-headers-4.9.0-kali3-common all 4.9.18-1kali1
  404  Not Found
Err:3 http://http.kali.org/kali kali-rolling/main amd64 linux-kbuild-4.9 amd64 4.9.18-1kali1
  404  Not Found
Err:4 http://http.kali.org/kali kali-rolling/main amd64 linux-headers-4.9.0-kali3-amd64 amd64 4.9.18-1kali1
  404  Not Found
Unable to correct missing packages.
E: Failed to fetch http://http.kali.org/kali/pool/main/l/linux/linux-compiler-gcc-6-x86_4.9.18-1kali1_amd64.deb  404  Not Found
E: Failed to fetch http://http.kali.org/kali/pool/main/l/linux/linux-headers-4.9.0-kali3-common_4.9.18-1kali1_all.deb  404  Not Found
E: Failed to fetch http://http.kali.org/kali/pool/main/l/linux/linux-kbuild-4.9_4.9.18-1kali1_amd64.deb  404  Not Found
E: Failed to fetch http://http.kali.org/kali/pool/main/l/linux/linux-headers-4.9.0-kali3-amd64_4.9.18-1kali1_amd64.deb  404  Not Found
E: Aborting install.

genodeftest, 2018-02-07:
Again, please use code blocks for readability. This is the least you could do.

Your dpkg sources configuration is broken. Instead of `http://http.kali.org/kali/pool/main[…]` it should be `https://http.kali.org/pool/main[…]`, i.e. without `kali/` and with `s` in `https`.

Anyway, the package pool is only listing packages in version 4.14.x for the kernel, but you still have 4.9.x installed. This indicates that you haven't updated your system for ~1 year. You should urgently update your system right now or never use it again because it probably has hundreds of known security vulnerabilities open. If updating fails, you should reinstall your linux distribution.

When done, please try installing the kernel module again.

hunatae, 2018-02-13:
this is a fresh install of linux.... 
anyways, what you just said was it's trying to fetch from an older version linux than the one i have? at this point im not surprised. can i get a link to the exact version i have and then just download it manually?
or i can try and find a whole new version; the one i used was: **kali-linux-2017.1-amd64.iso**

genodeftest, 2018-02-13:
> this is a fresh install of linux....

It may have been, about 2 years ago.

> anyways, what you just said was it's trying to fetch from an older version linux than the one i have?

No. It is trying to get the headers from the linux version you still have because either you did not try update your system for >1.5 years or because the distribution's update mechanism is broken. Or you did not update your linux for so long until the package pool changed its file structure.

>  or i can try and find a whole new version; the one i used was: kali-linux-2017.1-amd64.iso

The latest version is 2018.1, see the [official downloads page](https://www.kali.org/downloads/).

genodeftest, 2018-02-14:
> also, when you ask for code blocks, what does that mean?

Click on the link below the comment field, "Styling with Markdown is supported", which documents how the website parses your comments. Search for "Code" for details on code blocks.

austynwolfe, 2018-02-26:
Hey genodeftest. I finally managed to get the driver installed but networks still aren't appearing, at least in the GUI. I'm new to Kali and this is a fresh install running live from a flash drive. If you have any idea how to proceed with my quest for wi-fi it would be greatly appreciated. Thanks!

genodeftest, 2018-02-27:
@austynwolfe: Try having a look in the syslog using `dmesg` and `journalctl -b`. In case you want to upload them, please upload them into a [gist](https://gist.github.com/), because they are quite long. [Arch Linux Wiki](https://wiki.archlinux.org/index.php/Wireless_network_configuration) has some useful information and maybe the [Linux kernel wireless wiki](https://wireless.wiki.kernel.org/en/users/documentation) has some too.

hunatae, 2018-03-01:
im not sure if i finally got mine to work because at the very end it gave me this:

CC [M]  /home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = CFG80211_OpsVirtualInfChg,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: note: (near initialization for ‘CFG80211_Ops.change_virtual_intf’)
/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = CFG80211_OpsVirtualInfAdd,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: note: (near initialization for ‘CFG80211_Ops.add_virtual_intf’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.14.0-kali3-common/scripts/Makefile.build:319: recipe for target '/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o' failed
make[4]: *** [/home/dendo/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
/usr/src/linux-headers-4.14.0-kali3-common/Makefile:1525: recipe for target '_module_/home/dendo/Netgear-A6210/os/linux' failed
make[3]: *** [_module_/home/dendo/Netgear-A6210/os/linux] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

hunatae, 2018-03-01:
btw i looked at the page you told me to look at, and i still dont know what code blocks are so im sorry for the above if its "hard to read"