jurobystricky/Netgear-A6210 #86: Netgear WNDA3100v3 Master Mode

alexalouit, 2017-11-07:
Any chance to have a working AP/Master Mode with an Netgear_WNDA3100v3?

genodeftest, 2017-11-08:
Not unless someone steps up and implements it. I haven't seen any comment or commit from @jurobystricky since I started contributing to this code about 3 months ago, so I'm thinking about forking this project to have it open for all the pull requests here. As I don't have any hardware this driver is written for, I can't even test the code, so there will be many obstacles.

Do you have some experience in C development or even kernel drivers / low-level networking?

alexalouit, 2017-11-08:
I write rarely in C, but it is far from being my favorite language.
I'm never put into a tedious project like creating a driver..