jurobystricky/Netgear-A6210 #88: device not managed

243750496, 2017-11-18:
**install:**
```
download https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13 for NETGEAR A6210
not this one https://github.com/genodeftest/Netgear-A6210/tree/4.13-fix-only
```
```
sudo mv '/home/atc/Downloads/Netgear-A6210-port-to-4.13' /usr/src/netgear-a6210-2.5.0
sudo dkms install netgear-a6210/2.5.0
```

**device not managed :**
You should look at the contents of the file /var/lib/NetworkManager/NetworkManager.state. It should look something like this:

[main]
NetworkingEnabled=true
WirelessEnabled=true
WWANEnabled=true
Change any from 'false' to 'true' to re-enable networking. It may work better if you first stop NetworkManager:

Edit /etc/NetworkManager/NetworkManager.conf and change:

managed=false
to

managed=true
Restarted network-manager:

sudo service network-manager restart
No change. Then I tried:

touch /etc/NetworkManager/conf.d/10-globally-managed-devices.conf
sudo service network-manager restart
All three network cards came back on-line.

**Or:**
If you don't have it installed, could you install the package rfkill and post the output of "rfkill list"?
If it's software blocked enter "rfkill unblock #" where # is the number of the blocked item in the list, or "rfkill unblock name". If it reports it to be hard blocked check out the wireless switch in your laptop.

I had the same problem, I solved it by commenting the lines where the unmanaged device was mentioned, by mac address, in /etc/NetworkManager/NetworkManager.conf.

[main]
plugins=ifupdown,keyfile
dns=dnsmasq

[ifupdown]
managed=false

[keyfile]
unmanaged-devices=mac:00:26:2d:fd:5c:e0
changed to

[main]
plugins=ifupdown,keyfile
dns=dnsmasq

[ifupdown]
managed=false

\#[keyfile]

\#unmanaged-devices=mac:00:26:2d:fd:5c:e0

**device not ready:**
To fix this I had to disable secure boot in the UEFI BIOS page on startup." Yes... secure boot is useless and doesn't allow you to install some proprietary drivers such as some very common Realtek Ethernet devices and as i can see that goes for the mediatek wifi drivers-firmware too

i have done all of previous said, but still can't discover anything and show up ,if i typed 
`sudo service network-manager restart`
then it will be device not ready ,still none-functionable
[image]
[image]

[image]

could u please help me ?thx a lot!

genodeftest, 2017-11-18:
What about kernel logs? Anything in `dmesg` or `journalctl -b` about that kernel module, about NetworkManager, …? Does `lsmod` list this kernel module at all?

243750496, 2017-11-18:
`atc@ATC:~$ dmesg
[    0.000000] random: get_random_bytes called from start_kernel+0x42/0x4e1 with crng_init=0
[    0.000000] Linux version 4.13.0-16-generic (buildd@lcy01-02) (gcc version 7.2.0 (Ubuntu 7.2.0-8ubuntu2)) #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 (Ubuntu 4.13.0-16.19-generic 4.13.4)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.13.0-16-generic.efi.signed root=UUID=2516f032-1354-43b7-89d3-16358a399e3c ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009cfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009d00000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x00000000dc245fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc246000-0x00000000dc3cdfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc3ce000-0x00000000dc4cffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc4d0000-0x00000000dc8ddfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dc8de000-0x00000000dd20efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dd20f000-0x00000000dd26ffff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000dd270000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df000000-0x00000000dfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fdf00000-0x00000000fdffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000fea0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000feefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000081f37ffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.50 by American Megatrends
[    0.000000] efi:  ACPI 2.0=0xdc4d0000  ACPI=0xdc4d0000  SMBIOS=0xf05e0  SMBIOS 3.0=0xf0600  ESRT=0xdaed1658 
[    0.000000] random: fast init done
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: BIOSTAR Group X370GTN/X370GTN, BIOS 5.12 04/13/2017
[    0.000000] tsc: Fast TSC calibration failed
[    0.000000] tsc: Using PIT calibration value
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x81f380 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF write-through
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000000]   1 base 000080000000 mask FFFFC0000000 write-back
[    0.000000]   2 base 0000C0000000 mask FFFFE0000000 write-back
[    0.000000]   3 base 0000DF000000 mask FFFFFF000000 uncachable
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] TOM2: 0000000820000000 aka 33280M
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] total RAM covered: 3568M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 1G 	num_reg: 3  	lose cover RAM: 0G
[... 977 lines cut ...]
[   10.585238] ====> Radar Channel 52
[   10.585238] ====> Radar Channel 54
[   10.585238] ====> Radar Channel 56
[   10.585239] ====> Radar Channel 60
[   10.585239] ====> Radar Channel 62
[   10.585239] ====> Radar Channel 64
[   10.585239] ====> Radar Channel 100
[   10.585240] ====> Radar Channel 104
[   10.585240] 80211> TxStream = 0
[   10.585462] usbcore: registered new interface driver rt2870
[   11.041979] nvidia-modeset: Allocated GPU:0 (GPU-a431a63c-3c44-d2b6-4478-ed6a350cb274) @ PCI:0000:0c:00.0
[   16.777694] Bluetooth: RFCOMM TTY layer initialized
[   16.777699] Bluetooth: RFCOMM socket layer initialized
[   16.777703] Bluetooth: RFCOMM ver 1.11
[   23.469878] rfkill: input handler disabled
[   26.295380] r8169 0000:04:00.0 enp4s0: link up
[   26.295389] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
[   40.023172] vboxdrv: Found 16 processor cores
[   40.040207] vboxdrv: TSC mode is Invariant, tentative frequency 3593251056 Hz
[   40.040208] vboxdrv: Successfully loaded version 5.1.30_Ubuntu (interface 0x002a0000)
[   40.044981] VBoxNetFlt: Successfully started.
[   40.049159] VBoxNetAdp: Successfully started.
[   40.053401] VBoxPciLinuxInit
[   40.055514] vboxpci: IOMMU found
`

genodeftest, 2017-11-18:
Can you please put this output into code boxes or paste it into a gist? Comments are pretty bad for pasting log files.

243750496, 2017-11-18:
where the log file locate ? maybe it's easy to upload the log file

genodeftest, 2017-11-18:
> where the log file locate?

I don't understand what you mean. Can you please rephrase it?

You can find gist here: https://gist.github.com/ (that's like pastebin). It is better to put output of dmesg, journalctl or other log files into gists, because reading them here is quite hard.

243750496, 2017-11-18:
1:i wonder whether the 2 command (dmesg or journalctl -b ) is to check the 2 log files 
2:and use terminal to check the log files is really difficult to show full log , so if i know where the 2 log file locate may be i can use :sudo gedit *.log to check it , this may be easy
3:if i get the log files maybe upload it is a good choice

243750496, 2017-11-18:
hereis the output 
https://gist.github.com/243750496/e0d8212a807d0ace106a8b6b546549c2

genodeftest, 2017-11-18:
By the way, if you run `journalctl -b > filename.txt`, the logs from journald will be written to the file named "filename.txt". This way you don't have to copy so much stuff from a terminal.

243750496, 2017-11-18:
hereis all three files
[dmesg.txt](https://github.com/jurobystricky/Netgear-A6210/files/1485102/dmesg.txt)
[journalctlb.txt](https://github.com/jurobystricky/Netgear-A6210/files/1485103/journalctlb.txt)
[lsmod.txt](https://github.com/jurobystricky/Netgear-A6210/files/1485104/lsmod.txt)

genodeftest, 2017-11-18:
As you can see in these lines:

```
11月 19 05:24:30 ATC kernel: usb 4-3.1: new SuperSpeed USB device number 3 using xhci_hcd
11月 19 05:24:30 ATC kernel: usb 4-3.1: New USB device found, idVendor=0846, idProduct=9053
11月 19 05:24:30 ATC kernel: usb 4-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
11月 19 05:24:30 ATC kernel: usb 4-3.1: Product: A6210
11月 19 05:24:30 ATC kernel: usb 4-3.1: Manufacturer: NETGEAR
11月 19 05:24:30 ATC kernel: usb 4-3.1: SerialNumber: 100
```
The USB device is detected by the kernel.

According to
```

11月 19 05:24:31 ATC systemd[1]: smartd.service: Main process exited, code=exited, status=17/n/a
11月 19 05:24:31 ATC systemd[1]: smartd.service: Unit entered failed state.
11月 19 05:24:31 ATC systemd[1]: smartd.service: Failed with result 'exit-code'.
```
your smartd is broken, maybe by configuration? Anyway, that's probably unrelated to your problem with Netgear A6210.

You have this line:
```
avahi-daemon[1099]: Failed to open /etc/resolv.conf: Invalid argument
```
in the logs, which might indicate an issue with avahi. Please try whether you can connect to the internet through an ethernet cable. If yes, then this is not causing your connection problem.

As I understand this, there are some problems in NetworkManager
```
11月 19 05:24:31 ATC NetworkManager[1086]: ((src/devices/nm-device.c:1452)): assertion '<dropped>' failed
```
so initialization of a device fails. This seems to be the core issue, although I can't tell you why it fails. You probably need start the kernel and NetworkManager in debug mode to tell you more.

There is a crash in ModemManager:
```
11月 19 05:24:32 ATC systemd[1]: ModemManager.service: Main process exited, code=dumped, status=11/SEGV
11月 19 05:24:32 ATC NetworkManager[1086]: <info>  [1511040272.2988] ModemManager disappeared from bus
11月 19 05:24:32 ATC systemd[1]: ModemManager.service: Unit entered failed state.
11月 19 05:24:32 ATC systemd[1]: ModemManager.service: Failed with result 'core-dump'.
```
so please go and report it to your distribution vendor (Ubuntu, at launchpad.net) maybe they can fix it. I doubt this is related to your problem here though.

There is no crash or other bug reported in the mt7662u kernel module, so it does not fail fatally. I am out of more tips/ideas for now.

From lsmod: `mt7662u_sta` is in that list, so the kernel module gets loaded correctly.

243750496, 2017-11-18:
> You probably need start the kernel and NetworkManager in debug mode to tell you more.
how???

genodeftest, 2017-11-19:
1. Debug mode: Add the "debug" flag to your kernel command line. You can get there by presing "e" key during boot which will put you into grub's edit mode. Add " debug" to the command line and make sure to remove all other logging-related stuff like "quiet" and "rhgb".

2. I have no clue without looking at the files. If you prefer to keep this metadata private, you may want to send me an email (see my profile).

243750496, 2017-11-19:
What's next if i entered debug mode?just login and shut down then to enter normal start to copy the log file out? 
 btw: if don't set  debug flag  will  i get nomal start?
2. Whether debug kernel mode will create a log file that  contain key data? But where the log file  locate ？

243750496, 2017-11-20:
[image]
  
What should i remove?all?where to add  debug flag?

genodeftest, 2017-11-23:
In the line that reads wraps, there is a word "quiet". Replace it by "debug".

After reboot, try to use the network, then reboot again without these changes. Your linux will start normally if you don't change anything. Any changes in this grub config menu will be lost on next reboot.

genodeftest, 2017-11-24:
> What's next if i entered debug mode?just login and shut down then to enter normal start to copy the log file out?

As written above: After reboot, try to use the network, then reboot again without these changes.

> btw: if don't set debug flag will i get nomal start?

As written above: Yes. Your linux will start normally if you don't change anything. Any changes in this grub config menu will be lost on next reboot.

> Whether debug kernel mode will create a log file that contain key data?

No, it will not create a specific log file.

> But where the log file locate

After changing the command line, booting and trying to connect to a network, use `dmesg` and `journalctl -b` as before. Or if you rebooted already, you may want to use `journalctl -b -1` (or higher numbers instead of 1, see the manual for details) to select logs from previous boots.

243750496, 2017-11-25:
here is new logs:
[dmesg.new.txt](https://github.com/jurobystricky/Netgear-A6210/files/1502751/dmesg.new.txt)
[journalctlb.new.txt](https://github.com/jurobystricky/Netgear-A6210/files/1502752/journalctlb.new.txt)

genodeftest, 2017-11-26:
Have you tried connecting the USB device to a different USB port using a different USB cable?

genodeftest, 2017-11-26:
I could not find anything in the logs hinting to an issue, so I cannot help you.

243750496, 2017-11-27:
i fogot to enable wifi at the debug time , maybe this time the log may help 
btw: this time i connect the wifi adapter directly to PC without and usb cable

genodeftest, 2017-11-27:
Your journalctl output contains this line:
```
ModemManager.service: Main process exited, code=dumped, status=11/SEGV
```
Please get this bug fixed. Maybe your software is very outdated. In case you are running software without installing updates, don't expect anyone to help you.

There is a warning from nvidia graphics driver, you should take a look.
```
NVRM: Your system is not currently configured to drive a VGA console on the primary VGA device. The NVIDIA Linux graphics driver requires the use of a text-mode VGA console. Use of other console drivers including, but not limited to, vesafb, may result in corruption and stability problems, and is not supported.
```

What does `systemctl status --full NetworkManager-wait-online.service` give you? I'm asking because of
```
11月 27 09:22:50 ATC systemd[1]: NetworkManager-wait-online.service: Main process exited, code=exited, status=1/FAILURE
11月 27 09:22:50 ATC systemd[1]: Failed to start Network Manager Wait Online.
11月 27 09:22:50 ATC systemd[1]: NetworkManager-wait-online.service: Unit entered failed state.
11月 27 09:22:50 ATC systemd[1]: NetworkManager-wait-online.service: Failed with result 'exit-code'.
```
in your logs.

Have you tried connecting the device to a USB 2.0 port?

Conclusion of current state:
* The device correctly shows up on USB hub (see latest dmesg/journalctl files)
* The kernel module is loaded
* The best clue I have is that the issue is related to this log entry:
  ```
NetworkManager[1077]: ((src/devices/nm-device.c:1452)): assertion '<dropped>' failed
  ```
  Which goes wrong in nm_device.c:
  ```
NMDeviceType
nm_device_get_device_type (NMDevice *self)
{
	g_return_val_if_fail (NM_IS_DEVICE (self), NM_DEVICE_TYPE_UNKNOWN);

	return NM_DEVICE_GET_PRIVATE (self)->type;
}
```
  … and I don't have any clue about how to get this fixed. It looks like an issue "before" or within NetworkManager.

243750496, 2017-11-28:
so what to do next ? paste the bug to ubuntu lanchpad or fix by do something to fix by myself (no coding)?but what should i provide? the log file or your last posts?

genodeftest, 2017-11-28:
> so what to do next ? paste the bug to ubuntu lanchpad or fix by do something to fix by myself (no coding)?but what should i provide? the log file or your last posts?

I don't know for sure. I doubt this issue can be solved without coding, but I am not sure whether it is a bug in this kernel module or in NetworkManager. You may want to try to report a bug against NetworkManager and add a link here to all of the logs and posts.

franckgarnier21, 2017-12-05:
I have the same bug during the pppd on a VPN connection in Ubuntu 17.10 

NetworkManager[27095]: ((src/devices/nm-device.c:1452)): assertion '<dropped>' failed

Do you have more information to solve it ?

genodeftest, 2017-12-05:
> Do you have more information to solve it ?

Read the NetworkManager code (maybe after debugging it) and find out whether this is a bug in NetworkManager or in the kernel.

243750496, 2017-12-06:
https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1733138
i had post the bug to launchpad, still no official reply now