jurobystricky/Netgear-A6210 #91: read_overflow error

snguon, 2017-12-13:
Anyone know how to fix this?
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/snguon/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.13.0-19-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/snguon/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-19-generic'
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /home/snguon/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:8:0,
                 from ./include/linux/cpumask.h:11,
                 from ./arch/x86/include/asm/cpumask.h:4,
                 from ./arch/x86/include/asm/msr.h:10,
                 from ./arch/x86/include/asm/processor.h:20,
                 from ./arch/x86/include/asm/cpufeature.h:4,
                 from ./arch/x86/include/asm/thread_info.h:52,
                 from ./include/linux/thread_info.h:37,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:80,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/snguon/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/snguon/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/snguon/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/snguon/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/snguon/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:305:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
scripts/Makefile.build:302: recipe for target '/home/snguon/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/home/snguon/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1546: recipe for target '_module_/home/snguon/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/snguon/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-19-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

DerelictBoarder, 2017-12-27:
I also am having this issue, any fix yet?

dsime42, 2018-01-05:
Upgraded Fedora FC26 to 4.14.11-200.fc26.x86_64 and my old driver, that has been working, wouldn't compile (different error).
I downloaded the current driver and it gave me this error.
I tried the new driver on my old kernel 4.13.16-200.fc26.x86_64 and it still gives this error.
Recompiled old driver for 4.13.16-200.fc26.x86_64 and it still works so it doesn't seem to be an error in my software chain.
Both drivers have the same History.txt file so I can't tell you what versions I am talking about but the zip files are different sizes. I have renamed them based on when I downloaded them so don't let that cause confusion
-rw-r--r--. 1 root root 3119498 Nov  1 19:41 Netgear-A6210-master.20171101.zip
-rw-rw-r--. 1 root root 3119378 Jan  4 17:55 Netgear-A6210-master.20180104.zip

tpeterson993, 2018-04-18:
Also am having this issue. Please advise...

t0ny-fr33man, 2018-08-20:
Same issue.  Here's my latest build attempt:

export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /usr/src/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.0-32-generic/build DBGFLAGS=-DDBG SUBDIRS=/usr/src/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-32-generic'
  CC [M]  /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9:0,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /usr/src/Netgear-A6210/include/os/rt_linux.h:14,
                 from /usr/src/Netgear-A6210/include/rtmp_os.h:30,
                 from /usr/src/Netgear-A6210/include/rtmp_comm.h:64,
                 from /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:340:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^
scripts/Makefile.build:332: recipe for target '/usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1552: recipe for target '_module_/usr/src/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/usr/src/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-32-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

BLVCK971, 2019-11-02:
The update for kernel 4.15 and up is at https://github.com/kaduke/Netgear-A6210 162
```

rm -rf ~/Netgear-A6210
git clone https://github.com/kaduke/Netgear-A6210
cd Netgear-A6210
make
sudo make install
```